pbatard/rufus #893: Rufus 2.12P Annoyance or Bug?

sick13, 2017-01-27:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [x] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive
  - [ ] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>

Log
---
```
<FULL LOG>
```
Not an issue but basically an annoyance (don't know how to describe it)

Every time Rufus Starts or Exits it writes this into event log

"The Group Policy settings for the computer were processed successfully. New settings from 1 Group Policy objects were detected and applied"

And the second, Exiting Rufus makes a Notification sound in windows.

Is all this intended?

pbatard, 2017-01-27:
> Is all this intended?

Yes it is. Rufus disables Autorun while it is running, so that you don't get prompted by a Windows prompt/new Explorer window every time you plug an USB while it is running (since you are going to reformat that drive anyway), and it does so using Microsoft's official way, through the use of Local Group Policies (which ensures that, should your computer ever crash while the policy is being altered, it will remain in a known state).

Now, while the modification of Local Group Policies might be logged (as thousands of other benign events, which Administrators might want to be able to track), I can tell you that the notification sound you hear is 100% due to your own customization, and does not occur on a standard installation of Windows (else, this would be an annoyance that would be reported over and over, and you'd be able to find plenty of similar reports in this issue tracker, especially as the code that deals with Local Group Policies hasn't changed one bit in many releases).

Especially, changing Local Group Policies is not an event that requires users to be notified. So, if your computer does produce a sound every time a Local Group Policy is changed, then I would surmise that it's either because you configured that yourself, or you installed software that did that for you. If the latter, I would suggest you contact the makers of that software (probably a "security" solution or something similar), as there are plenty of legitimate reasons to for applications to want to alter Local Group Policies, such as __temporarily__ disabling AutoRun, and "security" solutions, or other type of monitoring software, shouldn't really be making their users jump every time that happens.

As I believe this should answer the question, I will close this issue.

pbatard, 2017-01-27:
Hmmm, actually, after further testing, I can replicate the issue now on a regular Windows 10 installation (and this also seems to happen with the previous version).

This only occurs when using X and not when using the _Close_ button.

I guess I'll need to take a look at it. Curious as to why nobody reported that issue until now.

pbatard, 2017-01-27:
Looks like this behaviour with X button was introduced between the 2.6 and 2.7 release...

pbatard, 2017-01-27:
Issue was introduced in [447464286315da833adecf4982292e71baf759ab] and more specifically with the [replacement of `PostQuitMessage(0)` with `SendMessage(hDlg, WM_COMMAND, (WPARAM)IDCANCEL, (LPARAM)0)`](https://github.com/pbatard/rufus/commit/447464286315da833adecf4982292e71baf759ab#diff-a3bb752083af1aa18ab57839086949d6L2649).

sick13, 2017-01-28:
yeah...now i remembered that using the close button did not produce the sound. (sorry it slipped my mind)
and i was always using the close button till for some weird reason  i used the X.

anyway hope it;s not big of a deal.

pbatard, 2017-01-28:
Nah, it shouldn't be a big deal to fix, as I have a good idea of what needs to be reverted to prevent the closing sound. And I probably should have spent a bit more time testing your initial report before dismissing it...

Anyway, it'll be fixed in the next version - thanks for the report!

commit 4373f0c4 (2017-01-30): [ui] fix notification sound when closing using X

* Closes #893
closed 2017-01-30 by pbatard