pbatard/rufus #894: Wish - Please add to the program independent function "Set partition Active"

VictorVG, 2017-01-28:
**Notes:** this is just wish fo never version.

The big request to add to the program independent feature "Make partition active" control button or checkbox that would do the following features:

_1) to mark the selected partition active when formatting the media;
2) marked as an active partition exists on the disk regardless of whether the files on it similar to the diskpart;
3) to perform the reverse operation for removing the selected partition or disk partition group status "Active" if necessary._

**Possible applications of this function** - loading multiple operating systems management with common carrier in the event that for some reason (such as diagnostic equipment packages) using boot managers is not desirable, but you need to be able to quickly select the boot disk partition without reformatting and wear the work area to which much sensitive devices such SunDisk iSSD (SATA) made in the form of chips with BGA replacement contacts which is impossible without special equipment.

pbatard, 2017-01-28:
Too custom and too disruptive for the existing USB creation flow, and likely to only help a __very__ limited number of people &rarr; not gonna happen, sorry.

Besides, as I explain on the FAQ, I have zero interest in making Rufus more friendly for multiboot. If you need multiboot, I will kindly request that you use something else than Rufus.

closed 2017-01-28 by pbatard

VictorVG, 2017-01-28:
Ok, sorry...