pbatard/rufus #895: Request

bbs-addict, 2017-01-29:
This is a request.

It is possible that there could be a submenu for Rufus to be the ultimate converter?

ISO
 +---IMG
 +---CDI
 +---NRG
 +---etc

Then vice versa with those files to ISO...so any file to any file. 
If this was requested before - sorry - thank you for your time and effort - we use this everyday.

pbatard, 2017-01-29:
You are misunderstanding what Rufus is for (it's not an image converter) and are asking for [Feature Creep](https://en.wikipedia.org/wiki/Feature_creep). There are dedicated applications that can perform these kind of conversions and, even if you might see it differently, not enough Rufus users for whom it'll be useful in the first place.

Rejected.

closed 2017-01-29 by pbatard

bbs-addict, 2017-01-29:
Ok, thank you. I was just asking. Thank you for replying. So many authors
do not do this anymore.

---

David Snyder

sendittodavid@gmail.com

I can do all things in him who strengths me.

Philippians 4:13