pbatard/rufus #896: Can't bring windows back while checksuming

oneohthree, 2017-02-02:
Checklist
---------
- [x ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x ] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [ ] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive
  - [ ] Plugging the USB into a different port
  - [x ] Running Rufus on a different computer
- [x ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
If I minimize all windows using Windows+D key combination while Rufus is checking the ISO hashes I can't bring either Rufus' windows when the checksum is finished. I can't do it even using Windows Task Manager.

Log
---
```
<FULL LOG>
```

pbatard, 2017-02-02:
I'm not seeing that issue at all on Windows 10 x64.

Of course, since you chose not to provide a log, I have no idea of the platform you are using or the ISO you ran the checksum against...

oneohthree, 2017-02-02:
Sorry for not providing a log. The platform I'm using is **Windows 7 x64 Ultimate**, ISO **es_windows_8.1_with_update_x64_dvd_6051503.iso**.

To reproduce the problem:

1. Run Rufus.
2. Select ISO file.
3. Click on # button.
4. Minimize all windows using Windows+D combination.
5. Wait for Rufus to finish the process.

pbatard, 2017-02-02:
Please provide a log rather than piecemeal information. Even if, for you, it's just gonna display the checksum, I still want to see a full log. For one thing, I still don't even know the version of Rufus you are using (which I would have gotten from the log).

pbatard, 2017-02-03:
Just in case that wasn't clear, I'm not asking you to produce the log of when you can't access Rufus after using <kbd>Windows</kbd>-<kbd>D</kbd>, but the one you get when you do the exact same thing, without using <kbd>Windows</kbd>-<kbd>D</kbd>.

Having the log is always helpful in getting details that users don't feel are relevant, but which can make a world of difference, and this is why I ask to __ALWAYS__ provide a log.

oneohthree, 2017-02-03:
All right, here's the log, sorry for the annoyances.

```
Rufus version: 2.12.1054 (Portable)
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0409
Will use default UI locale 0x0409
Found USB 2.0 device 'JetFlash Transcend 16GB USB Device' (8564:1000)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1899, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x05AAC3B6
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.6 GB (15626928128 bytes)
  Start Sector: 2048, Boot: No
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'IR5_CCSA_X64FRE_ES-ES_DV9'
  Size: 4.1 GB
  Uses: EFI
  Uses: Bootmgr
  Uses: Install.wim (version 0.13.1)
Using image: es_windows_8.1_with_update_x64_dvd_6051503.iso

Computing checksum for 'C:\Users\Foo\Desktop\es_windows_8.1_with_update_x64_dvd_6051503.iso'...
  MD5:	 dcbbe7d872b99aef149553c5b95c6055
  SHA1:	 b2c02e02c70569eb0cc7a912f72b1c3417c67e01
  SHA256: 0f3f3b8eda1e4e0eae6bdc2efa7fabe3185cdf2732ee51197166a5f6f74bc95a
```

pbatard, 2017-02-03:
Thanks. I know having to provide a log can be annoying, but it can avoid a lot of guesses.
I only ran a quick test yesterday so I'll look into trying to replicate your issue again when I have a chance.

pbatard, 2017-02-06:
Tested further, and I am now able to replicate the issue (using <kbd>Windows</kbd>-<kbd>D</kbd> or just the _reduce_ button). I'll see what I can do to fix this.

pbatard, 2017-02-06:
This will be fixed in the next version

commit f7b839c5 (2017-02-06): [ui] fix restore when a dialogbox is displayed while the main window is minimized

* Closes #896
closed 2017-02-06 by pbatard