pbatard/rufus #897: Error: Undetermined error while formatting

daninich, 2017-02-04:
Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [ ] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive
  - [ ] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I am trying to use Rufus to install ESXi6.5 on a USB drive.  Not working:

Rufus version: 2.12.1054 (Portable)
Windows version: Windows 10 64-bit (Build 14393)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0409
Will use default UI locale 0x0409
Found non-USB removable device 'VMware, VMware Virtual S SCSI Disk Device' => Eliminated
Found USB 2.0 device 'SanDisk Cruzer Glide USB Device' (0781:5575)
1 device found
No volume information for drive 0x81
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3809, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x009C84C9
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 29.2 GB (31330402304 bytes)
  Start Sector: 2048, Boot: Yes
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux.cfg' for Syslinux
  Detected Syslinux version: 3.86 (from '/isolinux.bin')
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'ESXI-6.5.0-4564106-STANDARD'
  Size: 328.3 MB
  Uses: Syslinux/Isolinux v3.86
    With an old menu.c32
  Uses: EFI
Using image: VMware-VMvisor-Installer-6.5.0-4564106.x86_64.iso
Will replace obsolete 'menu.c32' from ISO with the one found in './rufus_files/syslinux-4.07/menu.c32'

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
No drive letter was assigned...
Will use 'D:' as volume mountpoint
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Volume does not have an x86 Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (FAT32)...
Using cluster size: 16384 bytes
Slow format was selected
Error while formatting
Format error: Undetermined error while formatting.
Skipping GUID volume for '\Device\Floppy0'

Found non-USB removable device 'VMware, VMware Virtual S SCSI Disk Device' => Eliminated
0 devices found---
```
<FULL LOG>
```

daninich, 2017-02-04:
[image]
this is the error I get

pbatard, 2017-02-04:
Please follow the checklist. Have you tried with another USB drive? On a different computr? Bad blocks check??

The checklist is there for a reason, so don't ignore it.

daninich, 2017-02-04:
Hi - 
Thanks for the reply.  This is a brand new USB 3.0 drive.  I do not have another computer to check it on.  I am running the Bad blocks check now but it is taking a very long time...

pbatard, 2017-02-06:
Any update on the bad blocks check? What was the outcome.
Also, please be aware that `Undetermined error while formatting.` comes from Windows rather than Rufus, and I'm pretty sure that you will have the same error if you try to format your drive using the default Windows tool (right click on your drive, then select _Format..._). So there's probably not much I'll be able to do in Rufus...

daninich, 2017-02-06:
I gave up and downloaded unetbootin, which worked the first time.  I will update the case with this info.

closed 2017-02-06 by daninich