pbatard/rufus #900: Iso image extraction failure

deafly, 2017-02-09:
I am trying to extract windows 8.1 onto a USB drive and did it with the help of Microsoft customer service. However, after they finally got off the phone with me, Rufus is telling me that there is an Error: iso image extraction failure.

These are the logs:
Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE2 for write access
Will use 'F:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Rufus Master Boot Record
Volume has an unknown Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Writing master boot record...
Drive has a Zeroed Master Boot Record
Set bootable USB partition as 0x80
Using Rufus MBR
Found volume GUID \\?\Volume{837d1574-ee93-11e6-8256-24fd52576d54}\
Opened drive \\?\Volume{837d1574-ee93-11e6-8256-24fd52576d54} for write access
Writing partition boot record...
Using Standard NTFS partition boot record
Confirmed new volume has an NTFS boot sector
Successfully remounted Volume{837d1574-ee93-11e6-8256-24fd52576d54}\ on F:\
Copying ISO files...
Extracting files...
Image is an UDF image
Extracting: F:\autorun.inf (128 bytes)
Extracting: F:\boot\bcd (256 KB)
Extracting: F:\boot\boot.sdi (3 MB)
Extracting: F:\boot\bootfix.bin (1 KB)
Extracting: F:\boot\bootsect.exe (106.8 KB)
Extracting: F:\boot\en-us\bootsect.exe.mui (15.5 KB)
Extracting: F:\boot\etfsboot.com (4 KB)
Extracting: F:\boot\fonts\chs_boot.ttf (3.5 MB)
Extracting: F:\boot\fonts\cht_boot.ttf (3.7 MB)
Extracting: F:\boot\fonts\jpn_boot.ttf (1.9 MB)
Extracting: F:\boot\fonts\kor_boot.ttf (2.3 MB)
Extracting: F:\boot\fonts\malgun_boot.ttf (164.3 KB)
Extracting: F:\boot\fonts\malgunn_boot.ttf (161.9 KB)
Extracting: F:\boot\fonts\meiryo_boot.ttf (131.4 KB)
Extracting: F:\boot\fonts\meiryon_boot.ttf (129.8 KB)
Extracting: F:\boot\fonts\msjh_boot.ttf (151.3 KB)
Extracting: F:\boot\fonts\msjhn_boot.ttf (149.3 KB)
Extracting: F:\boot\fonts\msyh_boot.ttf (142.8 KB)
Extracting: F:\boot\fonts\msyhn_boot.ttf (138.8 KB)
Extracting: F:\boot\fonts\segmono_boot.ttf (35.2 KB)
Extracting: F:\boot\fonts\segoe_slboot.ttf (75.6 KB)
Extracting: F:\boot\fonts\segoen_slboot.ttf (75.3 KB)
Extracting: F:\boot\fonts\wgl4_boot.ttf (46.3 KB)
Extracting: F:\boot\memtest.exe (1.1 MB)
Extracting: F:\boot\resources\bootres.dll (17.8 KB)
Extracting: F:\bootmgr (394.8 KB)
Extracting: F:\bootmgr.efi (1.5 MB)
Extracting: F:\efi\boot\bootx64.efi (1.5 MB)
Extracting: F:\efi\microsoft\boot\bcd (256 KB)
Extracting: F:\efi\microsoft\boot\cdboot.efi (1.3 MB)
Extracting: F:\efi\microsoft\boot\cdboot_noprompt.efi (1.3 MB)
Extracting: F:\efi\microsoft\boot\efisys.bin (1.4 MB)
Extracting: F:\efi\microsoft\boot\efisys_noprompt.bin (1.4 MB)
Extracting: F:\efi\microsoft\boot\fonts\chs_boot.ttf (3.5 MB)
Extracting: F:\efi\microsoft\boot\fonts\cht_boot.ttf (3.7 MB)
[... 466 lines cut ...]
Extracting: F:\sources\iiscomp.dll (13.5 KB)
Extracting: F:\sources\imagingprovider.dll (184.5 KB)
Extracting: F:\sources\inf\setup.cfg (4.2 KB)
Extracting: F:\sources\input.dll (253.5 KB)
Extracting: F:\sources\install.exe (4.8 MB)
Extracting: F:\sources\install.wim (3.4 GB)
libcdio: fread (): Permission denied
  Error reading UDF file /sources/install.wim

Found USB 2.0 device 'Generic Flash Disk USB Device' (058F:6387)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1988, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x003A94F2
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 15.2 GB (16356737024 bytes)
  Start Sector: 2048, Boot: Yes

I think that this part is the problem: "libcdio: fread (): Permission denied
  Error reading UDF file /sources/install.wim"

Sorry for being such a newbie. Any help would be much appreciated!

pbatard, 2017-02-09:
There is a checklist for a __reason__!!

Unfortunately, you did not provide the full log, so I have no idea what ISO you are using (or what version of Rufus you are using, or what versions of Windows). There are plenty of ISOs of Windows 8.1, and I need to know the precise one

But more importantly, you didn't perform any of __these__ steps:

- [ ] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Now, I can pretty much guarantee that your error is either because:
- Your ISO is corrupted, which you should have been able to confirm with Microsoft support by computing the SHA1 and comparing it with the official one
- Your USB is corrupted, which you should also have been able to check by performing the first step.

The checklist is there to help __YOU__!!! If you choose to ignore what it says, please don't expect the people who develop the application to provide you with much help.

pbatard, 2017-02-12:
Since there hasn't been any update, I'm going to consider that OP realized that the issue was with their ISO, and close the issue.

closed 2017-02-12 by pbatard