pbatard/rufus #901: When Set as Mbr partition scheme for uefi, USB's MBR is unknown

apple3000, 2017-02-09:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [x] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive
  - [x] Plugging the USB into a different port
  - [x] Running Rufus on a different computer
- [x] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>
First , Sorry I am . I wrote at smartphone. So I did not read carefully your attention. 
I am poor in english, Understand me.

I want to make dual booting USB for MBR and UEFI.
So USB's MBR must NT60 or USBHDD+ with Active and must have EFI partition.

So I used rufus like below setting.

partition Scheme : MBR partition scheme for UEFI
File system : NTFS
Format options : Create a bootable disk Using - UEFI:NTFS (or iso image)

After that, Automatically non-activated "Use Rufus MBR with BIOS" at your advanced Options.
(I guess that my USB's MBR was unknown from this menu.)

and After Finishing, I confirmed my USB.
one partition NTFS and second partition FAT12 for EFI, was good.
But USB's mbr was Unknown.

So i wish you will set usb's MBR to NT60 or USBHDD+ although "MBR Partition Scheme for EFI"

Thanks very much.

Log
---
```
![2017-02-09_17-38-16](https://cloud.githubusercontent.com/assets/14814595/22775339/95d5739c-eeee-11e6-8abb-4a29ceba9ddb.png)
![2017-02-09_17-38-16](https://cloud.githubusercontent.com/assets/14814595/22775345/9f95e01a-eeee-11e6-9b30-abeb824cec20.png)
![2017-02-09_17-27-50](https://cloud.githubusercontent.com/assets/14814595/22775346/9f96e14a-eeee-11e6-8026-be90d0d0ecb4.png)
<FULL LOG>
Rufus version: 2.12.1054
Windows version: Windows 10 64-bit (Build 14393)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0412
Will use default UI locale 0x0412
NOTE: The Korean (한국어) translation requires an update, but the current translator hasn't submitted one. Because of this, some messages will only be displayed in English.
If you think you can help update this translation, please e-mail the author of this application
Found USB 3.0 device 'SanDisk Ultra USB Device' (0781:5581)
Using autorun.inf label for drive G: 'Winstall_201702HC'
Found USB 3.0 device 'ORGE GZX202 USB Device' (152D:1562)
Device eliminated because it was detected as a Hard Drive (score 13 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3781, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0xB7A86377
Drive has an unknown Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 29.0 GB (31102815744 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2 (UEFI:NTFS):
  Type: EFI FAT (0xef)
  Size: 256 KB (262144 bytes)
  Start Sector: 60749735, Boot: No
Found USB 3.0 device 'SanDisk Ultra USB Device' (0781:5581)
Using autorun.inf label for drive G: 'Winstall_201702HC'
Found USB 3.0 device 'ORGE GZX202 USB Device' (152D:1562)
Device eliminated because it was detected as a Hard Drive (score 13 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
Found USB 3.0 device 'SanDisk Ultra USB Device' (0781:5581)
2 devices found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3781, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0xB7A86377
Drive has an unknown Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 29.0 GB (31102815744 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2 (UEFI:NTFS):
  Type: EFI FAT (0xef)
  Size: 256 KB (262144 bytes)
  Start Sector: 60749735, Boot: No
Found USB 3.0 device 'ORGE GZX202 USB Device' (152D:1562)
Device eliminated because it was detected as a Hard Drive (score 13 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
Found USB 3.0 device 'SanDisk Ultra USB Device' (0781:5581)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3781, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0xBEA7B55E
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 25.1 GB (26910654464 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2:
  Type: FAT32 LBA (0x0c)
  Size: 3.9 GB (4193254912 bytes)
  Start Sector: 52561920, Boot: No

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE5 for write access
Will use 'M:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Windows 7 Master Boot Record
Volume has an unknown Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (MBR)...
Adding extra partition
Reserved 33 tracks (1 MB) for extra partition
Writing UEFI:NTFS partition...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Writing master boot record...
Drive has a Zeroed Master Boot Record
Using zeroed MBR
Found volume GUID \\?\Volume{6ef03bf9-ee66-11e6-ac40-50b7c3aa76aa}\
Successfully remounted Volume{6ef03bf9-ee66-11e6-ac40-50b7c3aa76aa}\ on M:\
Finalizing, please wait...
Created: M:\autorun.inf
Created: M:\autorun.ico
Successfully remounted Volume{6ef03bf9-ee66-11e6-ac40-50b7c3aa76aa}\ on M:\

Found USB 3.0 device 'ORGE GZX202 USB Device' (152D:1562)
Device eliminated because it was detected as a Hard Drive (score 13 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
Found USB 3.0 device 'SanDisk Ultra USB Device' (0781:5581)
Using autorun.inf label for drive M: 'USB_NTFS'
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3781, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x00000000
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 29.0 GB (31102815744 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2 (UEFI:NTFS):
  Type: EFI FAT (0xef)
  Size: 256 KB (262144 bytes)
  Start Sector: 60749735, Boot: No
Found USB 3.0 device 'ORGE GZX202 USB Device' (152D:1562)
Device eliminated because it was detected as a Hard Drive (score 13 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
Found USB 3.0 device 'SanDisk Ultra USB Device' (0781:5581)
Using autorun.inf label for drive M: 'USB_NTFS'
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3781, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x00000000
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 29.0 GB (31102815744 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2 (UEFI:NTFS):
  Type: EFI FAT (0xef)
  Size: 256 KB (262144 bytes)
  Start Sector: 60749735, Boot: No

![2017-02-09_17-27-50](https://cloud.githubusercontent.com/assets/14814595/22775272/438419ae-eeee-11e6-9eec-5a6316e86dbe.png)

pbatard, 2017-02-09:
> I want to make dual booting USB for MBR and UEFI.

If you want to do so for a Windows ISO, then please read [this](https://github.com/pbatard/rufus/wiki/FAQ#Why_doesnt_Rufus_create_a_Windows_installation_USB_that_can_be_booted_in_both_BIOS_and_UEFI_mode) and also read about the <kbd>Alt</kbd>-<kbd>E</kbd> cheat mode at the end of the FAQ. You will need to use <kbd>Alt</kbd>-<kbd>E</kbd>.

> So USB's MBR must NT60 or USBHDD+ with Active

`USBHDD+` is UltraISO specific stuff. Rufus is a generic tool - it does not try to copy what every other tool does, if it's only going to help a handful of people, which is the case here &rarr; I have no plans to ever add USBHDD+ support in Rufus (or any of the other MBRs displayed in your screenshot apart from the regular Windows ones).

>  and must have EFI partition.

If you're talking about a separate EFI System partition (ESP), that is incorrect. An USB can boot in "pure" UEFI mode without an ESP. 

> But USB's mbr was Unknown.

Rufus tends to use its own MBRs. It's not because a specific tool does is not able to recognize it that it means it won't work.

However, if you used "MBR partition scheme for UEFI" (and __not__ "MBR partition scheme for BIOS or UEFI") then your USB will only be bootable in pure UEFI mode and not in BIOS mode.

If you want to have dual BIOS + UEFI boot with Windows you need to:
- Use <kbd>Alt</kbd>-<kbd>E</kbd>
- Use "MBR partition scheme for BIOS or UEFI" in the options, after selecting your ISO (and make sure that it does say "MBR partition scheme for BIOS or UEFI" and not "MBR partition scheme for BIOS or UEFI-CSM". If it says "CSM" then you didn't use <kbd>Alt</kbd>-<kbd>E</kbd>)

apple3000, 2017-02-10:
Thanks , I learned ALT+E (Possible Dual booting), very thanks. 
I tested. 
I compared  MEthod what I did and Method of ALT+E, 
Same Second Partition for EFI - driver, 
Different thing is "Unknown MBR" and NT6.X

Yes, it is that I want. 
normally I set USB's MBR to NT6 . Just I had set to USBHDD+ (UltraISO write) before.

with ALT+E and Use "MBR partition scheme for BIOS or UEFI" , rufus set MBR to NT6 ,
**So I meet.
Thanks very much.**

IF you do not mind, I question One more,, 
Some board,, ex,, As HP Board, 
When Second partition is for EFI or NTFS-Driver for EFI(as rufus), HP efi system did not detect USB'EFI
Just When First partition is for EFI(ESP or FAT32..), HP EFI system detected USB's EFI. 

What do you think the problem?

pbatard, 2017-02-10:
> When Second partition is for EFI or NTFS-Driver for EFI(as rufus), HP efi system did not detect USB'EFI

Do you have Secure Boot enabled? You __MUST__ disable Secure Boot to be able to boot from NTFS in UEFI mode.

pbatard, 2017-02-12:
Since there hasn't been further updates, I'm going to close this issue.

closed 2017-02-12 by pbatard