pbatard/rufus #904: Can't re-access disk after Syslinux/NTFS installation

pbatard, 2017-02-16:
As per process explorer, it appears that Rufus leaves a handle open on the NTFS volume after a Syslinux/NTFS installation, instead of releasing all handles, which prevent it from re-acquiring access during the next format operation.

Note that this only occurs when NTFS is being used, not FAT32 (which makes me think that this could be a bug with the Syslinux code itself rather than Rufus), and that usage of Syslinux with NTFS drives was also clearly flagged as __EXPERIMENTAL__ with the 2.12 release.

pSatishC, 2017-02-17:
I see the same **[0x00000005] Access is denied** for FAT32 as well. The log is further below.
The sha256sum utility 'OK's the ISO so there is no issue with the image file.
This is on an Win7 Enterprise machine that has an encryption service (EMS/Dell Data Protection) running which prompts whenever a new USB disk is connected. Not sure if that is the cause though.

I copied the Rufus utility, ISO to a non-work laptop and the boot image burn went through without any issue.
```
Rufus version: 2.12.1054
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0409
Will use default UI locale 0x0409
Found USB 2.0 device 'JetFlash Transcend 8GB USB Device' (8564:1000)
Using autorun.inf label for drive F: 'Ubuntu 16.10 amd64'
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 978, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x088DBD4D
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 7.5 GB (8048869376 bytes)
  Start Sector: 2048, Boot: Yes
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.03/20160618 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Ubuntu 16.10 amd64'
  Size: 1.5 GB
  Has a >64 chars filename
  Uses: Syslinux/Isolinux v6.03
  Uses: EFI
  Note: This ISO uses symbolic links, which will not be replicated due to file system limitations.
  Because of this, some features from this image may not work...
Using image: ubuntu-16.10-desktop-amd64.iso
Will reuse 'ldlinux.sys' and 'ldlinux.bss' from './rufus_files/rufus_files/syslinux-6.03/20160618/' for Syslinux installation

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
Will use 'F:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Syslinux Master Boot Record
Volume has an unknown Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (FAT32)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Writing master boot record...
Drive has a Zeroed Master Boot Record
Set bootable USB partition as 0x80
Using Syslinux MBR
Found volume GUID \\?\Volume{842fbf14-2437-11e4-82df-ecf4bb17be59}\
Installing Syslinux 6.03...
Opened drive \\?\Volume{842fbf14-2437-11e4-82df-ecf4bb17be59} for write access
Using existing './rufus_files/syslinux-6.03/20160618/ldlinux.sys' ✓
Using existing './rufus_files/syslinux-6.03/20160618/ldlinux.bss' ✓
Successfully wrote 'ldlinux.sys'
Successfully wrote Syslinux boot record
Successfully remounted Volume{842fbf14-2437-11e4-82df-ecf4bb17be59}\ on F:\
Copying ISO files...
Extracting files...
Image is an ISO9660 image
This image will be extracted using Rock Ridge extensions (if present)
Extracting: F:\.disk\base_installable (0 bytes)
Extracting: F:\.disk\casper-uuid-generic (37 bytes)
Extracting: F:\.disk\cd_type (15 bytes)
Extracting: F:\.disk\info (55 bytes)
Extracting: F:\.disk\release_notes_url (78 bytes)
Extracting: F:\boot\grub\efi.img (2.3 MB)
  Unable to create file: [0x00000005] Access is denied.
Unable to create F:\syslinux.cfg - booting from USB will not work

Found USB 2.0 device 'JetFlash Transcend 8GB USB Device' (8564:1000)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 978, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x15AB2E42
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 7.5 GB (8048869376 bytes)
  Start Sector: 2048, Boot: Yes
```
I remember reading somewhere about automount or the drive letter auto-assign having something to do with such errors but my memory is vague at best.

pbatard, 2017-02-17:
That's __NOT__ the same issue. This issue is specifically for Syslinux installation on NTFS, so please don't hijack an existing issue because you are seeing something that looks remotely similar.

Especially, there is a [dedicated FAQ entry](https://github.com/pbatard/rufus/wiki/FAQ#Error_0x00000005_Access_is_denied) for regular `[0x00000005] Access is denied.` and I'm afraid that, yes, if it works on a non-work laptop then whatever is installed on your work laptop is the source of the problem, and there is nothing for me to fix in Rufus.

closed 2017-02-17 by pbatard

pSatishC, 2017-02-18:
My intention was only to provide more information should anyone land on this page like I did after a google search. That was also the reason I mentioned that the issue is only seen in my work machine. But clearly it wasn't pertinent and I apologize if it came across that I was trying to fault Rufus. It's a great utility that I relied on more times than I can count - thanks for the good work.

commit 136fd7a3 (2017-02-17): [syslinux] fix non release of the volume handle with NTFS

* Closes #904