pbatard/rufus #908: Undetermined Error while installing Os in Rufus

Kenosis16, 2017-03-02:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [x] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [ ] I also tried one or more of the following:
  - [x] Using a different USB drive
  - [x] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

undetermined error
-----------------
<Please describe your issue here>

Log
---
Rufus version: 2.11.995
Windows version: Windows 8.1 64-bit (Build 9600)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0409
Will use default UI locale 0x0409
Found USB 2.0 device 'USB2.0 Flash Disk USB Device' (1221:3234)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 509, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x08163525
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 (0x0b)
  Size: 3.9 GB (4193255424 bytes)
  Start Sector: 2048, Boot: Yes
Found USB 2.0 device 'USB2.0 Flash Disk USB Device' (1221:3234)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 509, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x08163525
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 (0x0b)
  Size: 3.9 GB (4193255424 bytes)
  Start Sector: 2048, Boot: Yes
Found UAS (USB 3.0) device 'Seagate BUP Slim BK SCSI Disk Device' (0BC2:AB24)
Device eliminated because it was detected as a Hard Drive (score 33 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
Found USB 2.0 device 'USB2.0 Flash Disk USB Device' (1221:3234)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 509, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x08163525
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 (0x0b)
  Size: 3.9 GB (4193255424 bytes)
  Start Sector: 2048, Boot: Yes
Found UAS (USB 3.0) device 'Seagate BUP Slim BK SCSI Disk Device' (0BC2:AB24)
Device eliminated because it was detected as a Hard Drive (score 33 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
Found USB 2.0 device 'USB2.0 Flash Disk USB Device' (1221:3234)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 509, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x08163525
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 (0x0b)
  Size: 3.9 GB (4193255424 bytes)
  Start Sector: 2048, Boot: Yes
Found USB 2.0 device 'USB2.0 Flash Disk USB Device' (1221:3234)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 509, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x08163525
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 (0x0b)
  Size: 3.9 GB (4193255424 bytes)
  Start Sector: 2048, Boot: Yes
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'GRMCPRXFRER_EN_DVD'
  Size: 3224686592 bytes
  Uses: EFI (win7_x64)
  Uses: Bootmgr
  Uses: Install.wim (version 0.13.1)
Using image: en_windows_7_professional_x64_dvd_x15-65805.iso

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
Will use 'D:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Volume has an unknown Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Error while formatting
Format error: Undetermined error while formatting.
Re-mounted volume as 'D:' after error

Found USB 2.0 device 'USB2.0 Flash Disk USB Device' (1221:3234)
1 device found
No volume information for drive 0x81
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 509, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x082048CB
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS/exFAT/UDF (0x07)
  Size: 3.9 GB (4193255424 bytes)
  Start Sector: 2048, Boot: Yes
Found USB 2.0 device 'USB2.0 Flash Disk USB Device' (1221:3234)
1 device found
No volume information for drive 0x81
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 509, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x082048CB
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS/exFAT/UDF (0x07)
  Size: 3.9 GB (4193255424 bytes)
  Start Sector: 2048, Boot: Yes
Found USB 2.0 device 'USB2.0 Flash Disk USB Device' (1221:3234)
1 device found
No volume information for drive 0x81
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 509, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x082048CB
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS/exFAT/UDF (0x07)
  Size: 3.9 GB (4193255424 bytes)
  Start Sector: 2048, Boot: Yes

<FULL LOG>
```

pbatard, 2017-03-02:
Why didn't you run the bad blocks check?
Also what happens if you try to reformat the same drive in Windows?

Please be aware that the error you are getting comes from the native Windows formatting tool, so there's not much I can do in Rufus when Windows reports it can't format the drive.

Kenosis16, 2017-03-02:
I think its not the rufus or the Image that has problem..ni think my two
usb is broken.. cause after i try it to a new brought usb. It works
properly. Anyways. Im sorry for the inconvenience.

pbatard, 2017-03-02:
That's why the checklist asks to perform a bad blocks check. I will close the issue then.

closed 2017-03-02 by pbatard