pbatard/rufus #910: New to UEFI + Rufus | Windows 10 x64 + Samsung 960 EVO PCIe SSD

BlackBox-, 2017-03-03:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [x] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive
  - [x] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
First time trying UEFI, as a result of buying a Samsung 960 EVO PCIe x4 drive (and Startech adapter).  Samsung's firmware update docs direct one to use Netbootin, which failed. 

Searched the topic extensively and found threads that recommended your product. Download your product, tried using Rufus on my VLSC/MSDN Win10x64 Enterprise ISO, which was used to install OS on several x64 workstations under Legacy mode. 

Used Rufus in each partition scheme, and File system, before finding your recommendation for GPT for UEFI + NTFS here: https://github.com/pbatard/rufus/issues/506 - didn't work.

More specifically, I reset BIOS to UEFI and disable SATA/RAID (I do not have Secure Boot option), restart, boot to UEFI USB (thanks to Rufus!), Windows setup runs without issue in +/- 4 mins, then at first prompt to restart, I remove the USB before Dell BIOS splash, then either left alone, or manually specifying 'Windows boot manager',the result is always 'No bootable device' message.

I've read the short answer. I would still greatly appreciate your help. 

Rufus creates the USB, the USB boots, Windows setup runs, but somehow the Samsung 960 EVO PCIe SSD does not contain the necessary bootmgfw.efi (or whatever the real file is called to allow it to properly boot under UEFI)?  _THAT_ is what's irking me, as it makes no sense (to me).

Finally, thought it was time to contact you, and made sure to include the Log. However, I did not attempt those final steps outlined above, as this is the first I am seeing them.

My existing Windows 10 OS (Legacy) recognized the Samsung 960 EVO on first boot. It works. It's fast. Just does not seem to boot.

Dell Precision T3600, Hex-Core Xeon E5, 32 GB RAM, 3 GB Nvidia Quadro K4000, Samsung 960 EVO

I appreciate your time, and consideration.

Cheers

Log
---
```
Rufus version: 2.12.1054 (Portable)
Windows version: Windows 10 64-bit (Build 10586)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0409
Will use default UI locale 0x0409
Found USB 3.0 device 'BUFFALO External HDD USB Device' (0411:01F0)
Device eliminated because it was detected as a Hard Drive (score 18 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
Found USB 2.0 device 'PNY USB 2.0 FD USB Device' (154B:007A)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 992, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {6F18CFE7-704D-4505-90D2-7F9B84D89577}
Max parts: 128, Start Offset: 17408, Usable = 8166668800 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {2D94AD3E-BA1B-4C0B-AAAB-8F2485B1E46A}
  Size: 7.6 GB (8165637632 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
[... 1087 lines cut ...]
Found USB 3.0 device 'BUFFALO External HDD USB Device' (0411:01F0)
Device eliminated because it was detected as a Hard Drive (score 18 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
Found USB 2.0 device 'PNY USB 2.0 FD USB Device' (154B:007A)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 992, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {F0981909-2680-4EE1-A988-D8EDEE49C48C}
Max parts: 128, Start Offset: 17408, Usable = 8166668800 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {C6E52DCB-9DE4-4E10-A16A-736990F2605A}
  Size: 7.6 GB (8164541952 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'UEFI:NTFS'
  ID: {C72E7961-1C84-455D-AD84-B61337421BC8}
  Size: 256 KB (262144 bytes)
  Start Sector: 15948419, Attributes: 0x0000000000000000

```

pbatard, 2017-03-03:
Okay. First of all, this looks more like a Windows installer issue than a Rufus issue. If Windows was able to start the installation process, then Rufus' job was properly done (it booted the installer), and unless you got errors during the file copy process (which you didn't), the Windows installer had access to every single file it needs for Windows to be installed properly.

Which means that your best option is to tweak the installation parameters from the Windows installer. The first think you should try is make sure you:
1. Have only the disk you want to install Windows on connected, with every other disk disconnected
2. The disk you want to install Windows on has been cleaned (i.e. it is partitioned).

If you do that, then Windows should be happy enough to create the partitions it needs, including the EFI System Partition (ESP) which is usually used to boot into the Windows system partition. Please be mindful that the Windows system partition does not need to contain a `bootmgfw.efi` as this is usually meant to be copied onto the ESP (which is not accessible from Windows).

The other thing I would advise you to try, since your `install.wim` is less than 4GB, is to use FAT32 instead of NTFS when creating the drive in Rufus, as I have gotten reports that lead me to think that, in some circumstances, Windows does not seem to like to see 2 NTFS partitions on different devices containing installation (the Windows installer is a mini Windows in itself).

BlackBox-, 2017-03-03:
Greetings pbatard, and thanks for the prompt reply.

That is my concern as well, but having never tried Rufus before this week, wanted to confirm that I had not overlooked anything.

1. Yes, in addition to setting BIOS for UEFI + SATA/RAID == off, I physically disconnected all SATA drives. Samsung 960 EVO is the only internal drive connected.
2. Correct; I used the Windows Setup to remove all partitions from the Samsung 960 EVO, and installed to the unallocated space, allowing the OS to format + partition as needed.

Everything seems to be there (on the PCIe SSD), it just doesn't ever actually boot, which I find odd, and have never seen before, in any workstation, nor server (I am an MTA Server).

I will try this procedure again after recreating the USB as FAT32, per your suggestion... If that doesn't work, I'll contact MSDN Support.

Cheers

pbatard, 2017-03-19:
No update &rarr; Closing.

closed 2017-03-19 by pbatard