pbatard/rufus #911: New Feature: Download all rufus_files at once to allow offline usage

bobafetthotmail, 2017-03-03:
I find mildly annoying that I cannot download all secondary files (bootloaders like syslinux or grub) at once and have them always with me so I can use rufus even when offline.

I have checked the FAQ and googled and I did not find any way other than using web crawlers to download stuff recursively from your site's /files folder and place that in the rufus_files folder.

I thought you could add a cheat code for this.

pbatard, 2017-03-04:
I don't want to add such a feature. Very few people would need and it would be a poor return on investment for me. I'd have to add __and maintain__ a web crawler a la HTTrack in Rufus just so that a few people, who aren't part of the user group Rufus is designed for, can do sysadmin stuff. First of all, this sounds a lot like feature creap, and as I have to repeat a lot more than I'd like but Rufus is __NOT__ designed to help sysadmin and power users, or even make their lives easier (which, for instance, is why I will never implement multiboot). It is targeted at the single usage regular crowd, who have an internet connection on the machine they create the USB on. As such, going out of my way to facilitate offline usage is not something I have any plans for.

Considering that you have found your own workaround, I will close this request.

closed 2017-03-04 by pbatard

bobafetthotmail, 2017-03-04:
> I'd have to add and maintain a web crawler a la HTTrack in Rufus 

That's beyond overkill, I wasn't asking for that.

You can just bundle all files (rufus executable and these additional files already in the folder) in a single zip that is downloaded together by the user himself. This can be done with a 2-line bash server-side script, and will require 0 mainteneance on your side.

> It is targeted at the single usage regular crowd, who have an internet connection on the machine they create the USB on. 

FYI, most power users and sysadmins are able to do tethering with their phones so it's not like this is a major issue for them. For the more normal users it can be an issue. That's why I thought it could be good to report this.

> Considering that you have found your own workaround, I will close this request.

Could you add a line for this in the FAQ at least?

pbatard, 2017-03-04:
> and will require 0 mainteneance on your side.

Famous last words, always proffered by the very person who wouldn't be involved with said maintenance.

Let's be honest here: You are rather selfishly asking me to spend time, which I have in limited supply, just so you can avoid having to use httrack or any other directory slurper. That's not gonna happen. So I will respectfully ask you to please do your monitoring and slurping of the files in your own time.

> Could you add a line for this in the FAQ at least?

F stands for frequently. I genuinely don't expect a question, with an obvious answer, about the best way to slurp the files to be asked that frequently. So, I'm afraid this wouldn't qualify for the FAQ (though I do have a line about telling people more precisely where they should copy the downloaded files, if Rufus can't do it for them, as part of #505 - but this is a different matter).

bobafetthotmail, 2017-03-04:
> Let's be honest here: You are rather selfishly asking me to spend time, which I have in limited supply, just so you can avoid having to use httrack or any other directory slurper.

Kinda, my main motives are that I'd rather not have to workaround client-side when I could have it fixed server-side, so to speak. I hate having to workaround bad design. And "needs internet to work" when it really does not, is bad design in my book.

But I also raised some other points (lack of internet might be an issue for people you target), and after you voiced your sensible concerns I offered a simple 0-mainteneance idea to get that (bundle rufus + files in a zip with a bash script running regularly in the server itself). 

And now I offer you to make this script myself (as I have downloaded your site so I know its folder structure, I also have some experience in scripting and linux server administration, although it's a very minor thing any decent linux user could do), and post it here for you to load it in the server on a cron job.

Does this make me a worthy contributor and not just a selfish asshole asking you to work for free?

>F stands for frequently. 

So tell me, you were getting asked frequently about the various stuff you have in cheat codes?
Most look very very exotic to me and I'm a power user with a job in IT. Only one I use is the UEFI+BIOS boot one.

>I genuinely don't expect a question, with an obvious answer, about the best way to slurp the files to be asked that frequently.

The most needed thing imho is a link to the webpage to download the files, as that isn't easy to reach without looking at the source (**rufus.h** has the links), playing guessing games (the local folder with rufus files is "rufus_files", so maybe the folder in the server is /files) or just downloading the whole site hoping you keep all stuff in there.

pbatard, 2017-03-04:
>And "needs internet to work" when it really does not, is bad design in my book.

Are seriously positing that the application should be quite a few MB larger so that it includes all the files one might possibly need? IMO, bad design would be to give in to request that will __penalize__ everybody by forcing them to go through an unneeded large file download, when in 85% of case, they wouldn't even need a single file from that extra download, and in the remaining 14.9%, not use more than a few 100KB. So that's way to suddenly make the application a lot less user friendly, when usage without an internet connection is the __exception__ rather than the norm. Also, even if I was happy to suddenly have everyone who downloads Rufus spend 10 times as long waiting for their download to complete (a small footprint, which is one of my actual design goals, tends to make for good user esperience) who's gonna pay for the extra bandwidth costs?

Once again, your power user/sysadmin design goals are very different from the goals I actually have when it comes to designing the app. So, I'm going to have to repeat that the app is not designed for your specific usage pattern, and trying to make it closer to what you want would end up making it worse for a vast majority of users. Besides, if you really want to make Rufus more suitable for your usage, the source is very much inviting you to do it, so it's not like you don't have any options there.

> lack of internet might be an issue for people you target

That's not the case. The users I do target must have an internet connection on the machine they create the USB on (be it only to be notified from a critical update, in case there is a bug or incompatibility in the app, which is usually a _good thing_. For instance, I half expect the next Windows refresh to break Rufus' DD image mode, so I sure hope Rufus users do have a valid internet connection and chose to enable the update check). But, as I pointed out earlier, I'm still planning to add a line on how to manually obtain the files in case you can't download them in the FAQ, as part of #505.

> bash script running regularly in the server itself

Not gonna add a cron job on the server just to keep a single person happy. And since you are mistakenly asserting that this might be the case, it's not the technical difficulty that makes me reject that proposal. I don't need someone to provide me with a ready made script. I just don't want to customize the server for single person's usage. And that's the same answer I gave to people who wanted me to upload new version release notes, on the server, in a format that they could parse with run of the mill tools, as well as countless other very specific requests I had over the years. if it's not gonna help the vast majority of users, it's not gonna happen.

> Does this make me a worthy contributor and not just a selfish asshole asking you to work for free?

Besides translations for the homepage, I don't accept contributions to modify the server's content, even if it's a very simple script.

> So tell me, you were getting asked frequently about the various stuff you have in cheat codes?

The FAQ entry is there to describe __what__ the cheat codes are. You are misinterpreting seeing a list of things being mentioned in the FAQ as if those were individual FAQs, whereas it is the list itself that constitutes the FAQ. In other words, the __frequent__ thing I am being asked is "Do you have a cheat mode list?", __NOT__ something like, "Hey, I want to disable Rock Ridge - How can I do that?". The vast majority of the cheat modes is actually stuff that, I, the developer, thought might come handy, mostly for troubleshooting, but not stuff that I was explicitly requested to add. So I'm afraid that your premise is wrong there: the list is what I was frequently requested, NOT the individual cheats.

> The most needed thing imho is a link to the webpage to download the files

Which I'm planning to do as part of #505 when i get around to it. It will detail to users who have issues getting Rufus to download the files where they can find them on the server, and where to copy them. And FYI, the log does tell where the files are being downloaded from (even if you don't have internet access). For instance, you'll see a line such as:
```
Downloading 'ldlinux.sys' from http://rufus.akeo.ie/files/syslinux-6.03/20150819/ldlinux.sys
```
As a power user, and even if the log is intended for internal uses, I'd expect you to have taken a closer look at it, as Rufus is always very explicit about what it does.

bobafetthotmail, 2017-03-04:
> Are seriously positing that the application should be quite a few MB larger so that it includes all the files one might possibly need?

I just need a "run offline" zip offered as an option, same as you offer the portable version as an option for power users (as most people don't know why they should use it anyway).
It will waste a grand total of 15-ish MB on your server, will require a 2-line script to be made, and to add a line with its link in the homepage.

I'm amazed at how you manage to read what I write always in the worst possible way. Do I really need to hold your hand like this?

> so I sure hope Rufus users do have a valid internet connection and chose to enable the update check)

We both know that most people leave settings default, so if it's not on by default it's likely to not check for updates. So heh, you claim you want internet access because it might check for critical updates but then leave autoupdate off, this kinda defies the point.

> and since you are mistakenly asserting that this might be the case

No, I'm just pulling your leg a bit with that, as your previous answers implied that what I ask MUST require a grandiose rewrite of rufus and that this major change will make things harder for kittens or something.

> Besides, if you really want to make Rufus more suitable for your usage, the source is very much inviting you to do it, so it's not like you don't have any options there.

One of the reasons I'm bugging you instead of just forking it or fixing it my way (with a .bat script controlling a wget-for-windows that pulls down stuff in crawler mode) is that I thought it might be useful for others too.

> And FYI, the log does tell where the files are being downloaded from (even if you don't have internet access). 

Lol that's too late. If I get that error it means I don't have internet access.

Even if I now know where these files are I need to get internet access if I want to download them, and if I can't I'm screwed.

pbatard, 2017-03-04:
Wow, talk about misreading things here:

1. There's no default for checking updates. There's a very prominent prompt, with a Yes/No choice the first time you run Rufus for which I expect most sensible to say Yes (you __MUST__ click on one of the 2). So, as opposed to your alternate narrative, the expectation is that most people do run Rufus with update checks, because they will have been presented with a choice... which is also very much confirmed through my web server logs. But Rufus is nice enough to make it opt-in, as any respectable application should.

2. The grandiose rewrite of Rufus is what you implied with your bad design statement (which clearly wasn't about putting a zip on the server). If you don't like the reply, then don't start criticizing design choices that you clearly haven't considered through the lens of the majority, instead of your own limited view.

3. As I tried to explain, I don't believe it will actually be useful for enough users to have me try to remember what cron scripts I need to carry over when switching servers. I'm also definitely not taking for granted that I may always have a dedicated server, with cron access, to run extra stuff on (especially if bandwidth costs become an issue). So the less dedicated server actions I have to implement when restoring content, to satisfy just a couple of people who can achieve the same on their own, the better. IMO, that's just good planning.

4. Your complaint was about having to dig through `rufus.h` to find the location of the downloadable, to which I replied that, you simply had to look at the log to get the same information. I have no idea what you're rambling about if you're complaining that you're screwed if you don't have internet access while at same time asking for a .zip to be placed on a remote server...

bobafetthotmail, 2017-03-04:
> The grandiose rewrite of Rufus is what you implied with your bad design statement (which clearly wasn't about putting a zip on the server).

But it wasn't either about adding a full-blown website scraper like you said in your answers. That's only how I worked around it.

I proposed the very-easy-way after your feedback.

> As I tried to explain, I don't believe it will actually be useful for enough users to have me try to remember what cron scripts I need to carry over when switching servers.

I don't believe you when you say you find hard to write down a line in the same place you write down other such technical and rarely used stuff already.
Anyway, if this happens and someone is still using it, you will get reminded by him, don't worry about that.

> I'm also definitely not taking for granted that I may always have a dedicated server, with cron access

The same can be done by a windows script you run either manually or after you recompile stuff or on a schedule on your build PC.
http://superuser.com/a/111266

> I have no idea what you're rambling about if you're complaining that you're screwed if you don't have internet access while at same time asking for a .zip to be placed on a remote server...

Because it's not apparent that you need internet access while running rufus too, while if there is a "offline rufus" option with a line explaining why there is such option, then people that need to run it offline will download that.

-------

Really, I've long since understood that you have no intention to yeld, I kept posting as I was curious to know what you were going to claim as "good reason to not do it".
So far, it's been entertaining.

pbatard, 2017-03-04:
> it wasn't either about adding a full-blown website scraper like you said in your answers.

That actually depends on whether you want to offer a cheat mode that enables Rufus to download all files at once, to fill up the `rufus_files` directory, for later offline usage, which is one of many ways to try to satisfy users who want an offline file repository that doesn't involve touching the server, which was always one of my concerns. But judging by your dismissive comments each time I bring one of the many factors that contributed to the original rejection of your request, I feel that trying to discuss with you what __actually__ goes into a design decision is a waste of time, especially when, apparently, the only person who gets to decide how a proposal should be implemented or how easily maintainable it might be is NOT the person who will be tasked to do these things in the long run.

There are also aspects of Rufus that you are unlikely to be aware of, which contributed to my initial decision, such as:
1. You conveniently ignore that I'd need to test how the fallback mechanism works on local files, and confirm that it is fit for any offline solution (and no, I would not take your word for it that "I tested it, and it works!"). There's a relatively complex lookup and fallback routine in Rufus when looking for Syslinux and GRUB files, which might very well work as is for offline, but which would need to be confirmed "fit for consumption". I certainly did not design the fallback for offline (at least not for using external file, as per 2.), and if your idea of adding the feature is simply "the files just need to be there and it __should__ work", then I'm afraid that you have a very poor idea of what software developments entails. There's no such thing as __untested__ "should work". Which means that, contrary to what you'd like to pretend, there's a lot more work than just adding a script on the server.
2. There already exists an "offline fallback" in Rufus and, because of the nature of the Syslinux business, if you are using anything that is based on Syslinux 6.03 (even if 6.03 with patches), which is the case of the vast majority of current Linux distros, then the internet download is actually pointless, because the `ldlinux.sys` you will download is exactly the same as the one embedded in Rufus, which Rufus will actually use if it detects that it can't download anything from the server (there's already an offline feature). So, in the vast majority of circumstances, for current Syslinux based distros, the download effected by Rufus is completely unnecessary but is conducted "just in case". Now, if you use GRUB or Grub4DOS based ISOs, or install a bare Syslinux (advanced mode), that's another story, because in that case Rufus really needs to fetch a file that it cannot replace from embedded, but in that case, all you ever really need are about 4 files, and not a whole repository that is mostly compose of Syslinux pre-release files, and there are hardly any ISOs out there that are GRUB or Grub4DOS based.

If you are lacking entertainment, may I suggest that you actually try to test the behaviour of your ISOs with Rufus in an offline fashion, and see for yourself how many files from the server you actually need?

harakiru, 2018-03-17:
I think a way to satisfy everyone by compromising a bit would be to add the necessary files separately on git and exempt it from any kind of compilation or releases. That way package maintainers or wiki writers can add a small note on their respective descriptions to download them and add it to the directory for offline usage. I don't think that would be very hard. Users who know what they are doing will find a way to do this anyway but it would be nice to not leave newcomers in the dark.

bobafetthotmail, 2018-03-17:
Since I know he will shoot down your proposal too for some other nonsense reason (it's harder to just have a script running in his server that packs stuff up into a single zip file, and he didn't want to do that either), I'll just point other people needing this to crawl his website with wget or curl or whatever, and machine-download stuff.

It has to restrict the download speed and add delays between each download so it does not trigger the server's IP ban protection that of course blocks connection from your IP for a while if you try to download them all together. This is common way to protect the server from DoS attacks, I don't think it is set up to screw with honest people that just want a local copy of all the files to use Rufus offline and have to crawl the website because the dev decided to screw them for no good reason.

This commandline works fine (takes a hour or so but this isn't something you need to do very often). It waits 10 seconds after each download and limits download rate to 20Kbit/s for the sake of not triggering a temporary IP ban. I didn't take much time to optimize it, so it may also work fine with less stringent limitations on time and bandwith.

**wget --wait=10 --limit-rate=20K -r -p http://rufus.akeo.ie/**

and then you get all the files you need in the rufus.akeo.ie/files/ folder it generates in your pc.

This can be done with a windows build of **wget**. http://gnuwin32.sourceforge.net/packages/wget.htm

It will actually pull down also the scripts he uses to ddownload/update the bootloader binaries or readme files telling where is the upstream repository with these files too, although they won't work on Windows.

pbatard, 2018-05-14:
> Rufus only functions while connected to the internet.

As I explained that is incorrect. For Syslinux, the fallback to embedded Syslinux 6.03 when no internet connection exist will work with most recent Syslinux based distros. I also explained that I don't expect a simple extract a zip file and forget about it to satisfy your usage, because the current code will not try to fall back locally if it doesn't find the requested data.

For instance, most distros tend to append their own version string when they add their bootloader (e.g. `syslinux-6.03/PCLinuxOS`), which, as you will see does not exist in the [current server files](https://rufus.ie/files/syslinux-6.03/). So what happens is:
1. If an online connection is available, Rufus looks first for `files/syslinux-6.03/PCLinuxOS/...` on the server
2. If that is not found (which will be the case), it falls back to picking up the files from `files/syslinux-6.03/` while copying them into a __local__ `rufus_files\syslinux-6.03\PCLinuxOS\` directory on your machine.
3. Next time Rufus runs, it sees `rufus_files\syslinux-6.03\PCLinuxOS\` and uses that.

If no online connection is available, Rufus simply uses the Syslinux 6.03 files that are __already__ embedded in the application, which, unless the distro maintainers went crazy, __will currently actually work!__. In other words, for most recent Syslinux bases distros, the Syslinux 6.03 downloads we retrieve from the server are unneeded (embedded will do just fine) but just a safety measure just in case.

Now, the problem you'll face with your fine "There's nothing to creating a zip file, why don't you do it!?!" is, you won't get the `rufus_files\syslinux-6.03\PCLinuxOS\` when you extract the zip. And I certainly don't remember designing anything in the code to fall back locally (i.e. go up a directory).

So the next thing I know is you're going to ask me to add local fallback code. And now I've got a new feature to test and maintain, with a bunch of people that I know will be fairly vocal if I start to provide something and it doesn't work as they want. Now, you may understand why I'd prefer to pass on that, and also why I have a [whole point in the FAQ](https://github.com/pbatard/rufus/wiki/FAQ#Why_cant_you_just_add_the_feature_I_want) that reminds people that that "5 minutes job, no work at all" feature you are sure would be peanuts to do usually only painless to add when you're NOT the person having to do it.

At this stage, and since people clearly dismiss the points I make as mere irrational rejection, rather than something that has been logically considered, I think this charade has been been going long enough so I'm going to set this topic as private.

If you really need that feature, you can create your own version of Rufus that points to an URL where you host and maintain that zip file of yours, and then see whether it does work as you hoped or not. The application is Open Source for crying out loud. If you are unhappy that the main developer(s) are not adding the feature you want, and you believe that feature would be easy for them to do, then just create your own version and stop expecting others to bend over backwards for you.