pbatard/rufus #912: Update feature: want that we download 2.12 but it is malware?

survivor303, 2017-03-13:
When trying to downlad file: http://rufus.akeo.ie/downloads/rufus-2.12.exe
I get error window: "The Requested header could not be located".

And when i trying download it with browser (firefox) i get this scary SonicWall message:

"Gateway Anti-Virus Alert

This request is blocked by the SonicWALL Gateway Anti-Virus Service. Name: (Cloud Id: 51406602) Upx-27 (Trojan)"

Because of this, Google Chrome not allowing any downloads of 2.12 files.

pbatard, 2017-03-13:
> When i trying to download file: http://rufus.akeo.ie/downloads/rufus-2.12.exe
> I get error window: "The Requested header could not be located".

Works for me (using the latest Chrome or Firefox).
And the real-time server logs confirm that this also work for other people. Therefore, the issue with the download is on your side.

> And when i trying download it with browser (firefox)

Wait a minute... how were you trying to download http://rufus.akeo.ie/downloads/rufus-2.12.exe if not with a browser? In case you were using Chrome and then tried with Firefox, please be aware that both these applications are "browsers", which is why your sentence is a bit confusing.

> i get this scary SonicWall message

Then the issue is a __false positive__ from SonicWALL. Please contact SonicWALL, as I have grown a bit tired of overzealous "security" solution that produce false positives every other day. See also [this FAQ entry](https://github.com/pbatard/rufus/wiki/FAQ#Antivirus_X_reports_that_Rufus_contains_malware).

As there is nothing for me to do, I will close this issue.

closed 2017-03-13 by pbatard

NSbuilder, 2017-03-13:
Rufus is clean, and the fact that it is open source lets everyone browse the source and see exactly what it is doing behind the scenes.

In older versions Avast antivirus would tell me that it contains a malware but it was fixed in later versions of Rufus - so I guess they figured out it is clean or maybe Rufus did something in the past that looked suspicious to the antivirus.

survivor303, 2017-03-13:
I was trying with rufus own update tool and when it not worked i try use
firefox because any other dw tool hasnt available for me then :)

But yes, we have internal problem with unauthorized sonicwall.

13.3.2017 12.28 ip.

downloads/rufus-2.12.exe

downloads/rufus-2.12.exe if not with a browser? In case you were using

<https://github.com/pbatard/rufus/issues/912#issuecomment-286069300>, or mute
the thread