pbatard/rufus #913: Fails to detect XenServer 7 as EFI capable ISO

kode54, 2017-03-15:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [x] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive
  - [ ] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [x] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Rufus does not detect the XenServer ISO as EFI capable. It contains a syslinux setup for BIOS booting, but it also contains EFI Grub, as:

```
.\EFI\xenserver\grubx64.efi
.\EFI\xenserver\grub.cfg
.\EFI\xenserver\grub-usb.cfg
```

Attempting to create boot media with unetbootin (ugh) results in a confusing mess, as my firmware lists the memory stick twice, once for each partition it creates. I end up having to use the UEFI shell, locate the correct partition, cd to the EFI\xenserver directory, then execute grubx64.efi manually.

Log
---
```
Rufus version: 2.12.1054
Windows version: Windows 10 64-bit (Build 14393)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0409
Will use default UI locale 0x0409
0 devices found
Found USB 2.0 device 'Kingston DataTraveler 2.0 USB Device' (0930:6544)
Using autorun.inf label for drive G: 'ESXI-6.5.0-20170104001-STANDARD'
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 486, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {E5056DAF-94B6-4E90-B7B1-C9547FD6E654}
Max parts: 128, Start Offset: 17408, Usable = 4005493248 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {C6F3DA69-7D9D-4095-9AC2-BAD72B3692B3}
  Size: 3.7 GB (4004462080 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Scanning image...
ISO analysis:
  Image is an ISO9660 image
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
  Will use '/boot/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 4.06/0x58a37aa4 (from '/boot/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'XenServer-7.1.0 Base Pack'
  Size: 674 MB
  Has a >64 chars filename
  Uses: Syslinux/Isolinux v4.06
Using image: XenServer-7.1.0-install-cd.iso
Found USB 2.0 device 'Kingston DataTraveler 2.0 USB Device' (0930:6544)
Using autorun.inf label for drive G: 'ESXI-6.5.0-20170104001-STANDARD'
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 486, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {E5056DAF-94B6-4E90-B7B1-C9547FD6E654}
Max parts: 128, Start Offset: 17408, Usable = 4005493248 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {C6F3DA69-7D9D-4095-9AC2-BAD72B3692B3}
  Size: 3.7 GB (4004462080 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Found USB 2.0 device 'Kingston DataTraveler 2.0 USB Device' (0930:6544)
Using autorun.inf label for drive G: 'ESXI-6.5.0-20170104001-STANDARD'
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 486, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {E5056DAF-94B6-4E90-B7B1-C9547FD6E654}
Max parts: 128, Start Offset: 17408, Usable = 4005493248 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {C6F3DA69-7D9D-4095-9AC2-BAD72B3692B3}
  Size: 3.7 GB (4004462080 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Saved C:\Users\kode5\Downloads\rufus.log

Computing checksum for '\\VBOXSVR\Downloads\XenServer-7.1.0-install-cd.iso'...
  MD5:	 eda05809a44141188ff32a35012bd958
  SHA1:	 84aadfca8a9014b7812c84e636f9c0444d24df0b
  SHA256: b6632fcdacfad4f2053df9b496a198a29d8672f664b87258425fb27b3d0df880
```

pbatard, 2017-03-15:
That's because, for BIOS boot, Rufus only supports GRUB that uses a standard config dir (`/boot/grub/i386-pc/`).

And for UEFI boot, Rufus only supports ISOs that use the __UEFI boot standard__ of placing the bootloader as `/efi/boot/bootx64.efi`. The Xen people seem to have decided to use `efi/xenserver/grubx64.efi` for their UEFI bootloader, which is nonconformant to any UEFI standard I know of, and therefore unsupported by Rufus.

And just in case you want to request that I add a lookup in Rufus, so that it detects the presence of a `efi/xenserver/grubx64.efi` and converts it to `efi/boot/bootx64.efi`, which would already be very custom in itself, please be mindful that this would be unlikely to work, without some additional very custom directives, as the GRUB bootloader tends to look for the cfg file in the same directory, and may call on additional config files using a relative path, which would likely get broken if moving the `.cfg` to `efi/boot/`. So any fixes added for Xen would be very custom and very brittle, and therefore something to avoid.

Better contact the Xen people and ask them why they didn't create their ISO with an `efi/boot/bootx64.efi` bootloader, as it would make it very easy to convert to UEFI bootable USB, even without using Rufus...

I will close this issue now, as the root of the problem is a non UEFI compliant ISO.

closed 2017-03-15 by pbatard

reinderien, 2018-02-03:
I ran into this, as well. It's fine that you're not going to write in support in Rufus, but is there some other workaround? Currently my Dell doesn't detect that the USB bootstick is EFI-bootable.

reinderien, 2018-02-03:
So I discovered that with XenServer 7.3, running Rufus in "dd" mode succeeds in producing a UEFI-bootable stick even though it doesn't allow the explicit UEFI mode. If you do not select dd, the stick is not UEFI-bootable but is still BIOS-bootable.

andyhhp, 2018-05-25:
The XenServer ISO is created with ISOHybrid, so is already MBR and EFI bootable.

I can't comment about the standards complientness of the result, but I believe it does pull some particularly ugly tricks for compatibility with various implementations

pbatard, 2018-05-25:
> so is already MBR and EFI bootable.

That's not at all what ISOHybrid is about. ISOHybrid has __nothing__ to do with EFI or BIOS (which I assume is what you meant when you said _MBR_, which is only a partition type).

All ISOHybrid does is apply a hack that allows the image to be written byte by byte in DD mode, in a manner that will create a partition table (MBR or GPT, it can be anything) as well as the relevant file systems and file content in these partitions. Then, if the people who created that image did things properly, it may contain the necessary files and boot records for booting. But that is not something that ISOHybrid mandates, and it certainly makes no obligation that the resulting drive will be bootable under UEFI. Also, because ISOHybrid is really about presenting __completely different__ content depending on whether you are looking at the ISO as a DD image or as an ISO9660/UDF image, it's not because something can boot in a certain manner when copied in DD mode that it contains the files necessary to make it boot in the same manner in non DD mode.

So, once again, it's not because XenServer is an ISOHybrid that it can be booted as UEFI media in ISO mode, and since the Xen people seem to have decided to do things their own way, in a very non de-facto mode, I have zero interest in adding super custom code to Rufus to try to look at the ISO in a DD manner, in order to extract whatever UEFI boot files the Xen people use their, and attempt to make it work in ISO mode.

If you are unhappy about this situation, __PLEASE__ contact the Xen people and tell them that:
1. Not everybody wants to use DD mode when copying images
2. As such, they should ensure that their ISO is also UEFI bootable when its content is copied onto a FAT32 formatted flash drive.

That's all.