pbatard/rufus #917: Chinese file name  Become gibberish

bowknot, 2017-03-25:
[image]

[image]

pbatard, 2017-03-25:
If you don't follow the checklist, and choose __not__ to provide the full log as requested, I will simply close this issue without looking at it.

bowknot, 2017-03-25:
log:
[rufus.zip](https://github.com/pbatard/rufus/files/870121/rufus.zip)

pbatard, 2017-03-25:
The ISO you are trying to use was mastered by relying on the computer it is being used on having the same codepage as the one it was created on, __instead of relying on Unicode_ (through properly formatted Joliet extensions).

The issue of codepage conversion __HAS BEEN SOLVED ONCE AND FOR ALL YEARS AGO, WITH UNICODE__. If someone, in 2017, such as the creators of `Ylmf_Ghost_Win7_SP1_x64_2017_0113.iso` (which by the way is unlikely to be an __official__ Windows 7 ISO, and therefore not something I would support in the first place) chose not to use Unicode names when mastering their image, then that issue is entirely theirs. In other words, I have no plans to make Rufus handle anything but ISOs that use Unicode. Other ISOs, such as yours, are unsupported.

pbatard, 2017-03-25:
Oh and by the way, every single site that appears to be serving the ISO you are using is being flagged as a malware site by Google, so I'd rather avoid having anything to do with that.

closed 2017-03-25 by pbatard