pbatard/rufus #918: Feature request: Allow running multiple instances of Rufus

xates, 2017-03-25:
Can you please allow to run multiple Rufus instances at once thus allowing multiple ISOs to be written to different USB drives contemporary?

pbatard, 2017-03-25:
No. It would be WAY too much work for very few people. I'd rather use my limited development time on more important features.

closed 2017-03-25 by pbatard

xates, 2017-03-25:
Ok, good job though! :)

beadon-SAT, 2017-07-06:
Could the check for another rufus application be re-used to make this easier to implement ?  Windows is already tracking the open file handles, a call to the OS would allow you to see the device that's currently being written by another process, filtering those devices from the list of possible options at application start time - or greying them out could work  ( try this tool for example to get the list of open handles)  https://technet.microsoft.com/en-us/sysinternals/bb896655.aspx  ).  I know I could use the feature too!

pbatard, 2017-07-07:
>  I know I could use the feature too!

Yes, a few admins and power users could use the feature. But Rufus is __not__ targeted at helping admin/powerusers if doing so requires a lot of effort for very little return on benefits.

If you really need that feature, I will kindly ask that you modify the code yourself or use a different application.

ziesemer, 2017-08-22:
I could also benefit from this - as writing to multiple USB drives now needs to be done sequentially, whereas I could otherwise do so in parallel.

The mutex checking to prevent multiple instances was effectively additional work that is already completed.  The request here is effectively to simply disable this existing functionality, not create additional functionality (from one perspective) - correct?

@pbatard - are there any issues or specific concerns you'd have with running multiple instances concurrently, even in the current version / as-is?  Beyond someone trying to write to the same media from multiple instances?  (Which wouldn't be any worse than trying to do so from Rufus and another alternative program?)  Or what was so significant about the "WAY too much work" that you had in mind?

I already have a local rebuild working that skips the multiple instance check (commented-out one line).  If I cleaned this up to require a command-line switch to enable this as an advanced feature, etc. - would you consider accepting a pull request for this?

pbatard, 2017-08-22:
> The request here is effectively to simply disable this existing functionality, not create additional functionality (from one perspective) - correct?

No, that is incorrect. As opposed to what many believe, the vast majority of time spent developing is not spent coding, but validating that the functionality an application provides works properly. And this needs to be done on regular basis, especially if any code that may affect the functionality may be modified. So this is not a matter of "the feature is there, why not just enable it?" but "if I am to provide the feature, it's better have been thoroughly validated, else the quality of the application and its perception will greatly suffer". I just can't go around saying "Here you go, you should be able to run multiple instances of Rufus", if I need to add "knowing that you should not not attempt to do X or do Y if you use multiple instances, because there are no guards against this in Rufus". For instance, a proper multiple instance implementation would have guards against 2 instances trying to use the same drive. And it should also have logic with regards to what happens if the saved settings are modified from one instance. Isn't there going to be a concurrency problem? For instance, if someone uses the portable version, knowing that I haven't really taken any specific precautions to allow dual access to the `.ini` file, what if that file is write locked by the first instance, and the user attempt to modify the settings with the second one. And there are all kind of potential concurrency issues, that may not be apparent to the non programmer, that I would have to validate before I can say "Yes, I'm happy to remove the lock that prevents multiple instances to run, because, I have __tested__ that multiple instances of Rufus should be able to function in parallel, without interfering with one another".

Now, in parallel to that, I consider that this is one of the requests that I would qualify as "selfish", as per point 2 of [this FAQ entry](https://github.com/pbatard/rufus/wiki/FAQ#Why_cant_you_just_add_the_feature_I_want). All things considered it's not going to be beneficial for a large enough number of user, and I will venture to say that the people who really need it should already be tech savvy enough not to have to require that an application that is designed to be used by the gerenal public in a one-of fashion (and against a single drive) should be sysadmin or power user friendly. I just don't have the scope to make Rufus power-user/sysadmin friendly, because if I started accepting these kind of requests (which are A LOT more work than people realize), I would not only have to spend a lot of time maintaining these advanced features from version to version, but also, since this is advanced stuff that is going to thread on terrain that I cannot anticipate, and this is where I expect most of my time to eventually be spent, helping sysadmin/power-user troubleshoot their super-specific scenario that should work but doesn't. I just don't have the time for this, especially as I still have loads of features lined up, which I haven't gotten started on yet, that are instantly going to help millions of users, rather then 10 or 100 at most. The cost/benefit of adding and supporting multiple instances of Rufus is just too high.

> I already have a local rebuild working that skips the multiple instance check (commented-out one line).

That's the way to go. The code is Open Source for a reason &mdash; so that, if you __really__ want a feature, you are never stuck, but can add it yourself (and then release your version, and then provide support for it yourself).

> would you consider accepting a pull request for this?

Absolutely not.

But feel free to release your modified version and advertise it, so that people who think they need this feature can find it. The one small request I have is that, if you release a binary, you please, please, PLEASE, digitally sign it (which means you'll need to get your hands on a code signing certificate that works with Windows executable), as the application needs to run with Administrative Rights, and you should never ever imply to your users that it's okay to run an unsigned elevated application on Windows (as it'll keep promoting security behaviours that are straight-down suicidal in this day and age).

snicholakis, 2018-08-09:
Try this.  It images multiple drives at once.  Worked great.  Rufus Rocks also!!!!