pbatard/rufus #919: A device was eliminated because it didn't report itself as a non fixed USB disk

jbburks, 2017-03-26:
Checklist
---------
- [x ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x ] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [x ] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive
  - [ ] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
A device was eliminated because it didn't report itself as a non fixed USB disk

Log
---
```
Rufus version 1.3.4.270
Windows version: Windows 8 or later 64-bit
Found device 'Generic USB Disk USB Device'
A device was eliminated because it didn't report itself as a non fixed USB disk
0 devices found.
Saved 'C:\Users\jbburks\Desktop\rufus.log'

```
System is Windows 10. USB is USB 2.0 spinning disk hard drive. Currently formatted and I can read the data on it.

pbatard, 2017-03-26:
Have you tried [this](https://github.com/pbatard/rufus/wiki/FAQ#Help_my_USB_drive_is_not_detected)???

pbatard, 2017-03-26:
Also please try <kbd>Alt</kbd>-<kbd>.</kbd> to enable debug enumeration, and post the new log you get. It should contain a lot more information about how Rufus detects your devices, which may help identify the issue.

pbatard, 2017-04-01:
No update &rarr; Closed.

closed 2017-04-01 by pbatard