pbatard/rufus #921: Windows 10 ISO Image Extraction Failed

1Ghostly1, 2017-04-01:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [x] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [x] I also tried one or more of the following:
  - [ ] Using a different USB drive
  - [x] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue Description
-----------------
I'm trying to install Windows 10 on the USB for another PC but, I keep getting this error at some point during the operation.

Extracting: G:\sources\dlmanifests\microsoft-windows-rasconnectionmanager\cmmigr.dll (63.5 KB)
Extracting: G:\sources\dlmanifests\microsoft-windows-rasserver-migplugin\rasmigplugin.dll (471.8 KB)
  Unable to create file: [0x00000003] The system cannot find the path specified

I have tried plugging the USB in a different port but, I continued to get the same error.

Log
---
```
<Rufus version: 2.12.1054
Windows version: Windows 7 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0409
Will use default UI locale 0x0409
0 devices found
Found USB 2.0 device 'Lexar USB Flash Drive USB Device' (05DC:C75C)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 974, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x011A9081
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 7.5 GB (8016363520 bytes)
  Start Sector: 2048, Boot: Yes
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'ESD-ISO'
  Size: 4.4 GB
  Uses: EFI
  Uses: Bootmgr
Using image: Windows.iso

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
Will use 'G:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Rufus Master Boot Record
Volume has an unknown Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (MBR)...
Closing existing volume...
[... 572 lines cut ...]
Extracting: G:\sources\replacementmanifests\international-core-replacement.man (9.3 KB)
Extracting: G:\sources\replacementmanifests\internet-naming-service-runtime-rep.man (4.6 KB)
Extracting: G:\sources\replacementmanifests\internet-naming-tools-replacement.man (640 bytes)
Extracting: G:\sources\replacementmanifests\ipv4ipv6coexistencemigration-net-replacement.man (3.4 KB)
Extracting: G:\sources\replacementmanifests\kernel-pnp-repl.man (1.3 KB)
Extracting: G:\sources\replacementmanifests\legacy-sapi-repl.man (4.8 KB)
Extracting: G:\sources\replacementmanifests\mdm_migration-replacement.man (25.0 KB)
Extracting: G:\sources\replacementmanifests\mfmpeg2srcsnk-migration-replacement.man (1 KB)
Extracting: G:\sources\replacementmanifests\mfplat-migration-replacement.man (1.1 KB)
Extracting: G:\sources\replacementmanifests\mfsrcsnk-migration-replacement.man (1.1 KB)
  Unable to create file: [0x00000003] The system cannot find the path specified.

Found USB 2.0 device 'Lexar USB Flash Drive USB Device' (05DC:C75C)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 974, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0127870F
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 7.5 GB (8016363520 bytes)
  Start Sector: 2048, Boot: Yes
>
```

pbatard, 2017-04-01:
I see that you are using an ESD ISO. Chances are likely that your ISO was corrupted during the download or creation. Unfortunately, since all ESD are different, there's no way to conclusively say that your ISO isn't corrupted.

Can you try to download the retails Windows 10 ISO from https://www.microsoft.com/en-us/software-download/windows10ISO/ and try again, after verifying the SHA-1 to confirm that the ISO is valid?

pbatard, 2017-04-01:
By the way, validating that your ISO is not corrupted is precisely what this part of the checklist is for:

- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

1Ghostly1, 2017-04-01:
I'm sorry, here they are.

MD5: d9653526b55c5d3c8b55c3af1929411a

SHA1: 3e7aa282931d4ba3adfc0e165276106a6d1a1757

SHA256: 93613331cfe19a2a0ea909237627f1644a1bf5f0d04a14342f92c822deb3db89

From what I've checked, the values don't match the official image. I will try again after I'm finished downloading the Windows 10 ISO.

1Ghostly1, 2017-04-02:
I have just finished downloading the ISO and have tried again but, it still shows the same error.

pbatard, 2017-04-02:
What's the SHA-1 of the new ISO?

1Ghostly1, 2017-04-02:
The SHA-1 is:

SHA-1: 529669ba774799b39488e439722180f09a05c262

pbatard, 2017-04-02:
hmmm, I do see any [hits for `529669ba774799b39488e439722180f09a05c262`](https://www.google.com/search?q=529669ba774799b39488e439722180f09a05c262). That means that you are still not using a retail ISO.

Again, can you please download the retail ISO from https://www.microsoft.com/en-us/software-download/windows10ISO/ so that we can confirm that the ISO is okay. I can't really help you otherwise.

Please note however that you must use a non Windows 10 machine to properly access the downloads from https://www.microsoft.com/en-us/software-download/windows10ISO/, as Microsoft will redirect you to https://www.microsoft.com/en-us/software-download/windows10 otherwise which does not offer the retail downloads that can be validated for corruption.

1Ghostly1, 2017-04-03:
I've downloaded the ISO again and, it gave me another corrupted one. I will try downloading again but, on a friend's PC. Both of our PCs are running Windows 7, by the way.

pbatard, 2017-04-11:
No update &rarr; Closed.

closed 2017-04-11 by pbatard