pbatard/rufus #922: Error: Undetermined error while formatting using rufus 2.12

lukemac28, 2017-04-02:
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'ESD-ISO'
  Size: 4.4 GB
  Uses: EFI
  Uses: Bootmgr
Using image: windows 10 pro.iso
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 2039, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x022F9AA4
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 15.6 GB (16776167424 bytes)
  Start Sector: 2048, Boot: Yes

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE2 for write access
Will use 'F:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Volume has an unknown Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Error while formatting
Format error: Undetermined error while formatting.
Re-mounted volume as 'F:' after error

Found USB device 'Generic Flash Udisk USB Device' (1000:1320) [GP]
1 device found
No volume information for drive 0x82
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 2039, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x02377884
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS/exFAT/UDF (0x07)
  Size: 15.6 GB (16776167424 bytes)
  Start Sector: 2048, Boot: Yes
Closed Windows format prompt

pbatard, 2017-04-02:
The elements from the checklist are __NOT__ optional. Please run the bad blocks test.

pbatard, 2017-04-02:
Also you didn't perform a search and the log is not the full log. Too many elements were ignored &rarr; Closing.

closed 2017-04-02 by pbatard