pbatard/rufus #925: PC crashes mid write to USB. Afterwards USB sticks cause Windows PC to crash

Jorazon, 2017-04-07:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [x] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive
  - [x] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Almost every time while writing a .img to a USB stick my PC crashes and won't boot until the USB stick is removed. Plugging the stick back in crashes the PC within a few seconds. The sticks work normally on Linux after a format. This has happened to three sticks (Kingston) so far with different files.

Log
---
```
Rufus version: 2.13.1081 (Portable)
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~rc2
System locale ID: 0x0409
Will use default UI locale 0x0409
Found USB 2.0 device 'Generic STORAGE DEVICE USB Device' (05E3:0745)
Device eliminated because it appears to contain no media
0 devices found

```

pbatard, 2017-04-07:
It does look like your USB drives are defective. Unfortunately, flash memory only has a number of write cycles before it fails, and when it does, you would typically experience what you are describing. Depending on the quality of your flash drives, you can very much experience similar failures with drives from the same batch.

Also please be mindful that formatting (be it on Linux or Windows) writes __very little data__ to the drive, so it's not a good way to tell if your drive works or not. On the other hand, as you write data to the drive, such as when writing an image.

Anyway, you should really run the bad blocks check (which you somehow chose to ignore from the checklist). If the bad blocks check fails, then this is an issue with your drive(s).

pbatard, 2017-04-07:
Could you also copy/paste the log from when you are actually creating the image from Rufus? I have no idea what image you are writing, which makes it VERY DIFFICULT to try to help.

Jorazon, 2017-04-08:
I was writing the latest images for Ubuntu, OPNsense (both 32bit) and Raspbian. I think you are right about the drives are defective as I've had all sorts of problems with them recently. I didn't do the bad blocks check at the time of first posting because it was very late. I did it today and two of the three drives failed.

pbatard, 2017-04-08:
Okay. Then I will close this issue on account that this is most likely a hardware failure. Thanks for the update.

closed 2017-04-08 by pbatard