pbatard/rufus #926: [INFO] How use a USB hard drive partition with RUFUS

MinionAttack, 2017-04-08:
Hi, I've checked the FAQ section but I couldn't find the answer. I need make a USB image disk of Windows 10 but I forgot all my USB on my house. I have an external USB HDD and I've created a FAT32 partition.

The external USB HDD it's (K:) and I've created a partition inside of 8GB (I:), when I select "List USB Hard Drives" I can select it but says "Multiple partitions (I:) (K:) [512 GB]". If I select "Start" the message says:

"All data on Multiple partitions (I:) (K:) [512 GB] will be deleted.".......

It means on partition (I:) or on the all hard drive (K:)? It's a bit confuse the message for me.

The name showed it's how Rufus detect the partition (I:) or only detects the whole disk (K:). I'm a bit confused :/

Regards.

pbatard, 2017-04-08:
It means __the whole disk, INCLUDING ALL THE DATA ON BOTH PARTITIONS I: and K:__ will be deleted.

Your disk is not K:, but instead is a container for both partitions K: __AND I:__. If you use a tool like Windows Disk Manager, you should see that the disk contains these 2 partitions.

See also [this entry from the FAQ](https://github.com/pbatard/rufus/wiki/FAQ#Does_Rufus_delete_all_partitionsvolumes_when_it_formats_a_drive).

closed 2017-04-08 by pbatard

MinionAttack, 2017-04-08:
Ok, thank you so much for your acclaration!

hellosputnik, 2018-08-27:
Is it possible to install to only one partition? I'm trying to set up persistence, but I can't find a way to shrink the partition that rufus created.