pbatard/rufus #928: unable to browse to select iso file

LeCube98, 2017-04-08:
Checklist
---------
- [x ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x ] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [x ] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [x ] I also tried one or more of the following:
  - [x ] Using a different USB drive
  - [x ] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Hello
i'm trying to use rufus to burn latest win10 retail release (1607) to secure the Os on a drive
i have many times do so without any issue but today i try this and was unable to do so because when i click on the browse icon to select the iso file  nothing happen !!! 
i just keep clicking but no browse window open and i can not chose the file i wanted 
the log file doesn't show any errors because the process haven't even start ! 
i never had this issue before. 
i have try to change usb port or usb drive even version of rufus... nothing i can't manage to make it work 
i think it's a windows problem... with a change in the code to how rufus browse windows file 
to be fully explicit i'm on the very new insider release of w10x64 (160170.rs_prerelease.170331-1532)
it could be an issue of compatibility between both build (rufus and w10)
hope it helps you and me in return. 
cheers.

LeCube

Log
---
```
0 devices found
Found USB 3.0 device 'Kingston DataTraveler 3.0 USB Device' (0951:1666)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3762, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0F187E6C
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 28.8 GB (30942947328 bytes)
  Start Sector: 2048, Boot: Yes
Found USB 3.0 device 'Kingston DataTraveler 3.0 USB Device' (0951:1666)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3762, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0F187E6C
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 28.8 GB (30942947328 bytes)
  Start Sector: 2048, Boot: Yes
0 devices found
Found USB 3.0 device 'PNY USB 3.0 FD USB Device' (154B:008A)
Waiting for access...
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3773, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {CF1E8204-2F50-47FE-9387-67DAD2A312D4}
Max parts: 128, Start Offset: 17408, Usable = 31042009600 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {46418EAB-0A4E-401A-97EB-AA8B528CFAA9}
  Size: 28.9 GB (31040978432 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000

```

pbatard, 2017-04-09:
Gaaah!! Why do people truncate the log even though I explicitly say NOT to do so.

Please update your log so that it complies with the following checklist item, which explicitly says _"The log I am copying is the FULL log, starting with the line Rufus version: x.y.z - I have NOT removed any part of it."_, because even though you checked the box, you did NOT follow the instructions.

LeCube98, 2017-04-09:
Sorry bro...
here they are 

Rufus version: 2.13.1081
Windows version: Windows 10 64-bit (Build 16170)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~rc2
System locale ID: 0x040C
Will use default UI locale 0x040C
0 devices found
Found USB 3.0 device 'PNY USB 3.0 FD USB Device' (154B:008A)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3773, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {CF1E8204-2F50-47FE-9387-67DAD2A312D4}
Max parts: 128, Start Offset: 17408, Usable = 31042009600 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {46418EAB-0A4E-401A-97EB-AA8B528CFAA9}
  Size: 28.9 GB (31040978432 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
0 devices found
Found USB 3.0 device 'Kingston DataTraveler 3.0 USB Device' (0951:1666)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3762, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0F187E6C
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 28.8 GB (30942947328 bytes)
  Start Sector: 2048, Boot: Yes

pbatard, 2017-04-09:
Thanks.  I think I'm going to sit and wait on this one, as this is an issue with insider build 16170, which Microsoft __explicitly states__ is [expected to break major stuff](https://blogs.windows.com/windowsexperience/2017/04/04/time-check-windows-insider-program-settings/):

>Now comes our standard warning that these new builds from our Development Branch may include more bugs and other issues that could be slightly more painful for some people to live with.

I have seen multiple cases of Microsoft breaking major functionality that Rufus relies on in insider builds, so unless an issue persists until close to the next expected release of Windows, I'm simply going to dismiss it as a Windows bug.

closed 2017-04-09 by pbatard

LeCube98, 2017-04-09:
i think you are right, i have trouble with other software ...