pbatard/rufus #935: Rufus 2.14.1086 and Win 10 1703 Build 16170.1000 issue

steve6375, 2017-04-12:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [x] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [ ] I also tried one or more of the following:
  - [x] Using a different USB drive
  - [x] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
New Windows 10 (pre-release) causes issue. I tested Removable USB Flash drives only.
Also prevents RMPrepUSB (NTFS) from working too
EaseUS Home Partition Master does seem to work however.
Using Large FAT32 ALT+L works in Rufus - so issue with Windows API?

This is just a heads-up as it may not affect current Win10 release.

Log
---
```
<FULL LOG>

Format operation started
Requesting disk access...
Opened \\.\PHYSICALDRIVE4 [\Device\Harddisk4\DR8] for write access
Will use 'F:' as volume mountpoint
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Volume does not have an x86 Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (FAT32)...
Using cluster size: 16384 bytes
Quick format was selected
Error while formatting
Format error: Undetermined error while formatting.
Re-mounted volume as 'F:' after error

Found USB 3.0 device 'Corsair Voyager GT 3.0 USB Device' (1B1C:1A09)
1 device found
No volume information for drive 0x84
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3897, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00309996
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 29.9 GB (32053919744 bytes)
  Start Sector: 2048, Boot: No
Closed Windows format prompt
Found USB 3.0 device 'Corsair Voyager GT 3.0 USB Device' (1B1C:1A09)
1 device found
No volume information for drive 0x84
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3897, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00309996
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 29.9 GB (32053919744 bytes)
  Start Sector: 2048, Boot: No

Format operation started
Requesting disk access...
Opened \\.\PHYSICALDRIVE4 [\Device\Harddisk4\DR8] for write access
Will use 'F:' as volume mountpoint
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Volume does not have an x86 Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 2048 sectors
Deleting partitions...
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (Large FAT32)...
Opened \\?\Volume{91fb284f-bb64-11e5-830c-bcee7b876d72} [\Device\HarddiskVolume18] for write access
Size : 29.9 GB 62605312 sectors
Cluster size 16384 bytes, 512 Bytes Per Sector
Volume ID is 12ea:154b
32 Reserved Sectors, 15278 Sectors per FAT, 2 FATs
1955460 Total clusters
1955459 Free Clusters
Clearing out 30620 sectors for reserved sectors, FATs and root cluster...
Initializing reserved sectors and FATs...
FAT #0 sector at address: 32
FAT #1 sector at address: 15310
Writing partition boot record...
Using Standard FAT32 partition boot record
Confirmed new volume has a primary FAT32 boot sector
Setting primary FAT32 boot sector for boot...
Confirmed new volume has a secondary FAT32 boot sector
Setting secondary FAT32 boot sector for boot...
Setting Label (This may take while)...
Format completed.
Writing master boot record...
Drive has a Zeroed Master Boot Record
Partition is already FAT32 LBA...
Using Windows 7 MBR
Found volume GUID \\?\Volume{91fb284f-bb64-11e5-830c-bcee7b876d72}\
Successfully remounted Volume{91fb284f-bb64-11e5-830c-bcee7b876d72}\ on F:\

Found USB 3.0 device 'Corsair Voyager GT 3.0 USB Device' (1B1C:1A09)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3897, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x003321F7
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 29.9 GB (32053919744 bytes)
  Start Sector: 2048, Boot: No

```

pbatard, 2017-04-12:
Hi Steve.

Yeah, the latest insider releases are known to have broken various fundamental behaviours of Windows for various apps (for instance see #928).

As per [Microsoft's own words](https://blogs.windows.com/windowsexperience/2017/04/04/time-check-windows-insider-program-settings/):

>Now comes our standard warning that these new builds from our Development Branch may include more bugs and other issues that could be slightly more painful for some people to live with.

So I think I'm gonna wait for a while and let it fizzle. Early insiders are really ALPHA releases, so people using them should expect major breakage.

steve6375, 2017-04-16:
P.S. I think you can also have a flat-file WinToGo Win 10 now.
I downloaded the current Win10 ISO from MS website and used WinToUSB to create a flat-file (non-VHD) USB Removable flash drive and (much to my surprise) it booted fine (no everlasting spinning circle of dots).
I haven't tried it with Rufus but I suspect it is the same.

pbatard, 2017-04-16:
Aha! Thanks for pointing that out. I think I'll give it a try to confirm.

I had been wondering if the automounting of multiple partitions from REMOVABLE drives had some other motive, and it looks like that might be it.

pbatard, 2017-04-16:
Wow, it actually works! And it's fast enough to be usable (from a Sandisk Extreme 16GB).
But now I got to update Rufus, because I only allowed WinToGo on REMOVABLE in MBR mode (dual BIOS + UEFI), and prevented it on GPT unless FIXED... That's great news for Windows users though!

NSbuilder, 2017-04-17:
Great news for windows to go users!

pbatard, 2017-04-19:
I'm not seeing much of an issue with the latest insider. Or at least, some of the problems I have been seeing with 16170 seem to have been reduced. I am therefore going to close this issue.

closed 2017-04-19 by pbatard