pbatard/rufus #938: error undetermined error while formating

albusgandulf, 2017-04-16:
Checklist
---------
- [x ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x ] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [x ] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ x] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive
  - [ ] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>

Log
---
```
<FULL LOG>
Rufus version: 2.14.1086
Windows version: Windows 10 64-bit (Build 16176)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~rc2
System locale ID: 0x0809
Will use default UI locale 0x0809
Found USB 2.0 device 'TOSHIBA TransMemory USB Device' (0930:6544)
Found USB 2.0 device 'Multiple Card  Reader USB Device' (058F:6366)
Device eliminated because it appears to contain no media
1 device found
No volume information for drive 0x81
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 942, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: SFD (Super Floppy Disk) or Unpartitioned
Partition 0:
  Type: Small FAT16 (0x00)
  Size: 7.2 GB (7756087296 bytes)
  Start Sector: 0, Boot: No
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 4.04/0x4e86cf20 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'Android-x86 LiveCD'
  Size: 460.3 MB
  Uses: Syslinux/Isolinux v4.04
  Uses: EFI
Using image: naugatx8620160902.iso

Format operation started
Requesting disk access...
Opened \\.\PHYSICALDRIVE1 [\Device\Harddisk1\DR1] for write access
Will use 'D:' as volume mountpoint
Analyzing existing boot records...
Drive does not have an x86 Master Boot Record
Volume does not have an x86 Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (GPT)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (FAT32)...
Using cluster size: 4096 bytes
Quick format was selected
Error while formatting
Format error: Undetermined error while formatting.
Re-mounted volume as 'D:' after error

Found USB 2.0 device 'TOSHIBA TransMemory USB Device' (0930:6544)
Found USB 2.0 device 'Multiple Card  Reader USB Device' (058F:6366)
Device eliminated because it appears to contain no media
1 device found
No volume information for drive 0x81
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 942, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {0D2A9DAA-7789-4828-A7BA-CD1CEDEF2896}
Max parts: 128, Start Offset: 17408, Usable = 7756052992 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {36754C6F-9348-47BF-AD88-1EF8590C28B6}
  Size: 7.2 GB (7755021824 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Closed Windows format prompt
```

pbatard, 2017-04-16:
Have you tried formatting your drive from Windows? If so what happens?

pbatard, 2017-04-16:
Hold on, you're using the latest Windows Insider. Duplicate of #935. This is a Windows issue with Insider builds, so please report it to Microsoft so they can fix it. Be very mindful that, as indicated by Microsoft, the current Insider builds are ALPHA software and will break applications. It is up to Microsoft to fix that.

closed 2017-04-16 by pbatard