pbatard/rufus #939: Can't access external USB disks after upgrading to Windows Creators Update (1703)

tsynik, 2017-04-16:
- [*] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [*] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [*] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [*] I also tried one or more of the following:
  - [*] Using a different USB drive
  - [*] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
After upgrading Windows 10 from Annivesary Update (1607) to Creators Update (1703) I can't access disks in exteranal USB enclosures (blank list). Tried 2 different enclosures with various disks on various ports. They worked fine with the same Rufus version (2.14 1086) and same hardware on this PC on 1607 build.

Log
---
```
Found UAS (USB 2.0) device 'ASMT 2105 SCSI Disk Device' (174C:5106)
NOTE: This device is an USB 3.0 device operating at lower speed...
Waiting for access...
Could not open \\.\PHYSICALDRIVE2 [\Device\Harddisk2\DR3]: [0x00000020]  The process cannot access the file because it is being used by another process.
Device eliminated because it appears to contain no media
0 devices found
```

tsynik, 2017-04-16:
Hm, found the case of this issue. After upgrade, I probably also updated Paragon ExtFS for windows to current release, but doesn't mentioned it, because Rufus worked on same machine just few hours ago before upgrade with old Paragon ExtFS version. I also have Paragon HFS+ for windows, but it doesn't affect Rufus functionality in drives detection thou listed as incompatibe.

closed 2017-04-16 by tsynik

pbatard, 2017-04-16:
FYI, Paragon HFS+ is mentioned in the [List of software that is incompatible with Rufus](https://github.com/pbatard/rufus/wiki/FAQ#Software).