pbatard/rufus #940: Win Vista Migration to Win 10- [0x00000020] The process cannot access the file

guaitiao, 2017-04-19:
Checklist
---------
- [ X] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [X ] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [X ] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [X ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive
  - [X ] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>
I am trying to create am Windows 10 USB boot from Windows Vista 64bits...Same error as previous issues I read with Win 7, 8 and 10 before ver 2.14. I ran Process Explorer and find that the HarddiskVolume4 is being used by searchindex.exe from Microsoft Search Indexer. I tried to disable the locking using Alt-, however same error came. I killed the Search Indexer and start over, same thing happened. I did tried previous versions 2.13...and your tests version T1, T2 and they fail. I could not find ver 2.11.

At the beginning Rufus 2.14 was formatting ok my USB. However, I did a mistake, and cancel the process. After that this error started to appear.. I formatted the USB using Format command from File explorer. This did not solved the problem. And by the way, I am also experimenting an AutoPlay popup window when the Rufus re-mounts the USB drive after aborting the process.
 
What should I do?
Log
---
```
<FULL LOG>
Rufus version: 2.14.1086
Windows version: Windows Vista SP2 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~rc2
System locale ID: 0x0409
Will use default UI locale 0x0409
Found USB 2.0 device 'SanDisk Cruzer USB Device' (0781:5530)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 973, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x019FB4F5
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 7.5 GB (8003256320 bytes)
  Start Sector: 2048, Boot: Yes
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'CCSA_X64FRE_EN-US_DV5'
  Size: 4.1 GB
  Uses: EFI
  Uses: Bootmgr
  Uses: Install.wim (version 0.13.1)
Using image: Win10 64x.iso

Format operation started
Requesting disk access...
Opened \\.\PHYSICALDRIVE1 [\Device\Harddisk1\DR2] for write access
Will use 'F:' as volume mountpoint
Waiting for access...
Could not open \\?\Volume{740a9ecd-99a8-11e5-b3e7-001e687ec550} [\Device\HarddiskVolume4]: [0x00000020] The process cannot access the file because it is being used by another process.
Could not lock volume
Re-mounted volume as 'F:' after error

Found USB 2.0 device 'SanDisk Cruzer USB Device' (0781:5530)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 973, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x019FB4F5
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 7.5 GB (8003256320 bytes)
  Start Sector: 2048, Boot: Yes
Saved C:\Users\user1\Downloads\rufus.Log
```

pbatard, 2017-04-19:
First of all, please be mindful that, since Microsoft stopped supporting Windows Vista, I too am no longer supporting this platform any longer.

The ability to run Rufus on Windows Vista is being kept in Rufus __for the time being__, but that does not mean it is still supported. In other words, if you find that Rufus works on other platforms, but not Vista (or XP), I'm afraid that you're pretty much on your own...

> I ran Process Explorer and find that the HarddiskVolume4 is being used by searchindex.exe from Microsoft Search Indexer.

Then I guess this is your issue. You will need to disable search altogether in Windows Vista.

guaitiao, 2017-04-19:
I just start up in Safe Mode. An Rufus 2.14 did without any trouble. I need to test it. That is why I am trying to move from Vista to Win 10 since support has ended in April 11, 2017.  Most of the web pages and apps does not support it since IE and many of its core can not be kept updated. The only way I can move it to Win 10 is by creating the USB Win 10 installation disk since I do not have any other machine available running Win 7, 8 or 10....

I am very happy to find Rufus. It worked fine just with the ones needed on Safe Mode....

Thank you for your effort in keeping Rufus,  it is a great tool. 

Thanks.

closed 2017-04-19 by guaitiao