pbatard/rufus #943: Rufus 2.14 needs a long time to start and does not find my USB drive

JustMyGithub, 2017-04-20:
Checklist
---------
- [x ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x ] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [x ] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [x] I also tried one or more of the following:
  - [ ] Using a different USB drive
  - [x] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Rufus 2.12 does not start writing because of an access error. Rufus 2.14 does not list the thumbdrive, probably because of the same error. Rufus 2.14 needs 12 seconds to show the GUI.
Unetbootin can create a bootable media on that very thumbdrive and computer without any problem, so there is probably no other application accessing the drive.

Log
---
```
<FULL LOG>
Rufus version: 2.14.1086 (Portable)
Windows version: Windows 10 64-bit (Build 14393)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~rc2
System locale ID: 0x0407
Will use default UI locale 0x0407
Found USB 2.0 device 'SMI USB DISK USB Device' (090C:1000)
Waiting for access...
Could not open \\.\PHYSICALDRIVE4 [\Device\Harddisk4\DR14]: [0x00000020] Der Prozess kann nicht auf die Datei zugreifen, da sie von einem anderen Prozess verwendet wird.
Device eliminated because it appears to contain no media
0 devices found

pbatard, 2017-04-20:
> so there is probably no other application accessing the drive.

That's very unlikely. If you get error `0x00000020`, then it very much means that there actually is another process accessing the drive, and you will need to figure out what this is.

Considering that you are the only person to report this issue with Rufus 2.14, it is logical to conclude that the issue is with your environment, and not Rufus (otherwise, many people would report the same error as you do, which they don't).

DX37, 2017-04-21:
I have this bug too. When I start portable Rufus 2.14, it's starting too long. When I plugging in my USB or selecting `List USB Hard Drives` (I have one, actually), then rufus is frozes for a time, which is matches with time of starting.
Rufus 2.12 works perfectly on same PC and work scenario.
There is a log: [rufus.log](https://github.com/pbatard/rufus/files/946644/rufus.txt)

pbatard, 2017-04-21:
> Then it's not listing any USBs or HDDs

Rufus never lists HDDs by default.

>  then Rufus is frozes for a time, which is matches with time of starting.

I don't understand what you are saying here. If your drive needs a long spin up time, then yes, Rufus will probably freeze until your drive has completed its spin up.

Can you try <kbd>Alt</kbd>-<kbd>.</kbd> with your drive plugged (this will turn USB debug on), and then copy/paste the log so that I get a better insight into what's happening?

I still have to point out that, even though Rufus 2.14 has already been downloaded about 1 million times, nobody else appears to reports this problem. So, even if Rufus 2.12 appeared to work, there __definitely__ is something specific to your environment that is the cause of this issue.

For your information, Rufus 2.13 and later are a lot less forgiving about applications that prevent exclusive access to a drive, because if other applications are accessing the drive while Rufus is partitioning/formatting/copying files, then data may get corrupted (and the user will blame Rufus, when it's really the other application that is at fault). So, starting with version 2.13, Rufus has become more strict about getting exclusive access, which means that you may run into the issue you describe if you have an ill-behaved application on your system, that is trying to keep a lock on your drive(s) when it shouldn't. It could be an anti-virus solution, or anything that accesses a disk.

The first thing you may want to do then is make sure you're not running any of the applications mentioned in the [List of software that is incompatible with Rufus](https://github.com/pbatard/rufus/wiki/FAQ#Software).

Then, if you still can't find it, you will have to:

1. Download and install [Process Explorer](https://technet.microsoft.com/en-us/sysinternals/processexplorer.aspx)
2. Run Process Explorer __as administrator__
3. In Process Explorer type <kbd>Ctrl</kbd>-<kbd>L</kbd> until you see the lower pane, and also type <kbd>Ctrl</kbd>-<kbd>H</kbd>, so that the lower pane will display Handles.
4. Go through each process one by one and look at the handles, until you find the one that is keeping a handle to `\Device\Harddisk2\DR3` (which is your SanDisk Extreme USB Device).

DX37, 2017-04-21:
> If your drive needs a long spin up time, then yes, Rufus will probably freeze until your drive has completed its spin up.

My HDD can't spin up for ~28 seconds >__<

> The first thing you may want to do then is make sure you're not running any of the applications mentioned in the List of software that is incompatible with Rufus.

Ah, hell no...
I'm running `Paragon ExtFS of Windows`. And of course right after uninstalling it, Rufus 2.14 now working very good. (But it was working before, WTF?)
I'm taking my words back. Thank you.

pbatard, 2017-04-21:
> I'm running Paragon ExtFS for Windows.

There you go.

> But Rufus worked with it before 2.14

If it did, then you were very lucky, because I've had tons of reports from people using Rufus 2.12 or earlier, who had an issue with Paragon software. As a matter of fact, the notice about Paragon was added to the FAQ long before the 2.12 release.

Anyway, glad to hear you sorted out. If you have support with Paragon, maybe you should report to them that keeping a lock on all the drives is a bad idea...

pbatard, 2017-04-29:
Since @JustMyGithub never followed up and @DX37 sorted their problem, I'm going to close this issue.

closed 2017-04-29 by pbatard

JustMyGithub, 2017-05-08:
Actually I'm using Paragon as well. I wonder why other Tools similar to Rufus can work with Paragon ExtFS Driver, but Rufus cannot....

pbatard, 2017-05-08:
Other tools consider that an application that has the drive open for write access is no big deal (i.e. they don't care about __exclusive locking__), which, as far as I'm concerned, is a mistake. And, if you look at what's really going on with paragon software with a tool like [Process Hacker](https://wj32.org/processhacker/nightly.php) you'll see that it has a service that keeps all drives opened for read and write __all the time__ (which, IMO, is also another mistake that Paragon should fix, as there is no reason for any application to keep a permanent write handle to a drive unless it is poorly designed: You should only get a write handle when you need to write, and close it as soon as you are done).

The problem is that when that, when an external process does that, then Rufus can offer ZERO GUARANTEE that the data will be written to the drive as it should. That's because, the other application, with a write handle open, has the ability to write at any time, regardless of what Rufus is doing. So it might very well be writing data over the one that Rufus has just written, which will corrupt the drive in strange ways... and then I get Rufus users who complain that Rufus is doing a bad job, when it's really another process that is at fault.

Other tools similar to Rufus may decide that this is no big deal, but my first goal with Rufus is to ensure that your drive will boot, and leaving any uncertainty as to whether that will really be the case is therefore not a risk I am willing to take.

Now, if you decide for yourself that you wanna run that risk, you may be able to get Rufus to work with Paragon software using <kbd>Alt</kbd>-<kbd>,</kbd> [cheat mode](https://github.com/pbatard/rufus/wiki/FAQ#Power_keysCheat_modes). But then you're on your own, as this is unsupported.