pbatard/rufus #945: Where is Syslinux version: 6.03/20151222 ?

LinuxUser9537127, 2017-04-24:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [x] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive
  - [x] Plugging the USB into a different port
  - [x] Running Rufus on a different computer
- [x] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Multiple Linux ISO files that I have gotten from official websites (For Example: Linux Mint Cinnamon and MATE, Xubuntu, Ubuntu MATE) are requiring a Syslinux version that is not listed under https://rufus.akeo.ie/files/syslinux-6.03/. The version is 6.03/20151222. I have looked online, including through your website but cannot find this version.

When I try to write the ISO to a USB, a "Download Required" box pops up saying "This image uses Syslinux 6.03/20151222 but this application only includes the installation files for Syslinux 6.03/2014-10-06." Then asks me to click yes to download files, but when I do click yes, nothing is downloaded as the files cannot be located. The program could not download file and will try to use embedded 6.03 version instead. This is show in the log for Example #3 below.

Log
---
```
Rufus version: 2.14.1086
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~rc2
System locale ID: 0x0409
Will use default UI locale 0x0409
Found USB 2.0 device 'SanDisk Cruzer Glide USB Device' (0781:5575)
Using autorun.inf label for drive E: 'Lubuntu 17.04 amd64'
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1887, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x004EA267
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.5 GB (15521021952 bytes)
  Start Sector: 2048, Boot: Yes

Scanning image... (Example #1)
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.03/20151222 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Lubuntu 16.04.2 LTS amd64'
  Size: 880 MB
  Uses: Syslinux/Isolinux v6.03
  Uses: EFI
  Note: This ISO uses symbolic links, which will not be replicated due to file system limitations.
  Because of this, some features from this image may not work...
Using image: lubuntu-16.04.2-desktop-amd64.iso

Scanning image... (Example #2)
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.03/20151222 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Linux Mint 18.1 MATE 64-bit'
  Size: 1.7 GB
  Uses: Syslinux/Isolinux v6.03
  Uses: EFI
Using image: linuxmint-18.1-mate-64bit.iso

Scanning image... (Example #3)
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.03/20151222 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Ubuntu-MATE 16.04.2 LTS amd64'
  Size: 1.6 GB
  Uses: Syslinux/Isolinux v6.03
  Uses: EFI
  Note: This ISO uses symbolic links, which will not be replicated due to file system limitations.
  Because of this, some features from this image may not work...
Using image: ubuntu-mate-16.04.2-desktop-amd64.iso
Downloading 'ldlinux.sys' from https://rufus.akeo.ie/files/syslinux-6.03/20151222/ldlinux.sys
Unable to send request: The server name could not be resolved.
Could not download the file - will try to use embedded 6.03 version instead
Downloading 'ldlinux.bss' from https://rufus.akeo.ie/files/syslinux-6.03/20151222/ldlinux.bss
Unable to send request: The server name could not be resolved.
Could not download the file - will try to use embedded 6.03 version instead

```

pbatard, 2017-04-24:
> re requiring a Syslinux version that is not listed under https://rufus.akeo.ie/files/syslinux-6.03/

That's because the default version of Syslinux 6.03 will work.

Please understand that, because distros tend to recompile their own version, but most of the times with minimal changes, Rufus uses a fallback mechanism when it looks for Syslinux and there will only be a specific `Syslinux-x.y_#####` directory __IF__ it happens that the version from the distro __is  ACTUALLY incompatible__ with the official `Syslinux-x.y` binary. If not, then Rufus will fall back to using the official binary.

In this case, there is __NO NEED__ for a special `syslinux-6.03_20151222` version because the binaries that can already be found in `syslinux-6.03` will work just as well, therefore there is no `20151222\` subdirectory on the server, as Rufus will be smart enough to fall back to the parent directory __provided you don't have network issues, which you do__.

Moreover, even if you have a network issue, the version of Syslinux that would have been downloaded in your case is exactly the same as the one found on the server, because the 6.03 embedded in Rufus is the same as the one at `https://rufus.akeo.ie/files/syslinux-6.03/`, so this is a non issue.

Unless you __actually__ experience an issue with Syslinux during boot, you shouldn't assume that there is an issue with the Syslinux downloads (though I have to state that getting a `The server name could not be resolved.` message indicates a network error on your side, which has nothing to do with Rufus). The fact is that:
- The directories you mention don't exist because they aren't needed
- The fallback mechanism worked exactly as expected, even with your network issue, and you did end up with a drive that boots as expected.

Therefore I will close this (non) issue.

closed 2017-04-24 by pbatard

LinuxUser9537127, 2017-04-25:
Thank you for the taking time to respond. Will proceed with using the embedded Syslinux 6.03 that comes with Rufus 2.14.