pbatard/rufus #946: Rufus 2.14

belgariad, 2017-04-29:
Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [ ] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive
  - [ ] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Rufus 2.14 keeps not finding my usb. When it does it says failed to open media.

Log
---
```
<FULL LOG>
```
Rufus version: 2.14.1086
Windows version: Windows 10 64-bit (Build 15063)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~rc2
System locale ID: 0x0809
Will use default UI locale 0x0809
Found UAS (USB 3.0) device 'Seagate Expansion Desk SCSI Disk Device' (0BC2:3320)
Device eliminated because it was detected as a Hard Drive (score 33 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
Found USB 2.0 device 'Kingston DataTraveler 2.0 USB Device' (0930:6544)
Removing J: from the list: This is the disk from which Rufus is running!
Found USB 2.0 device 'Lexar USB Flash Drive USB Device' (05DC:C75C)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 973, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x005A771E
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 7.5 GB (8007974912 bytes)
  Start Sector: 2048, Boot: Yes

pbatard, 2017-04-29:
Well, there are 2 issues there:

1. You are running Rufus from an USB drive (`J:`), so of course Rufus will always eliminate that drive, since it would be pointless to attempt to format it. You must ensure that neither Rufus not your ISO reside on the drive you want to format.
2. In case your issue is with the other drive, i.e. the `'Lexar USB Flash Drive USB Device' (05DC:C75C)`, the fact that it's only listed on an intermittent basis, and, when listed, Rufus has trouble accessing it is indicative of a __hardware__ issue. Either you have issues with your cables/connectors or the flash memory on your drive is failing. I notice that you skipped the checklist item that says _Running Rufus on a different computer_. Please try that. You should also see what happens when you try to format that drive in Windows.

poloznsk, 2017-04-30:
The same problem with Rufus 2.14. Trying on Windows 10 and two other laptops with Windows 7, cannot find usb drive. Rufus 2.6 works fine at all my laptops.

pbatard, 2017-05-02:
Try [rufus-2.15_TEST2.exe](https://rufus.akeo.ie/testing/rufus-2.15_TEST2.exe).

pbatard, 2017-05-07:
Since there was no update, I will close this call, as this should be fixed in the latest BETA.

closed 2017-05-07 by pbatard