pbatard/rufus #947: i cant fix this by myself anymore. please help

craneofbloom, 2017-04-30:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [x] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive
  - [x] Plugging the USB into a different port
  - [x] Running Rufus on a different computer
- [x] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>
im creating a bootable windows 8.1 usb drive using rufus. i bought a new usb just for this. and when i started to copy the iso into my usb, an error message always shows up when its about to be finished. i copied the whole log.,im noob when it comes to these, and im trying solve it for days by myself, but i cant figure it out, please help.
Log
---
```
<FULL LOG>
```
Rufus version: 2.14.1086
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~rc2
System locale ID: 0x0409
Will use default UI locale 0x0409
0 devices found
Found USB 2.0 device 'Generic Flash Disk USB Device' (058F:6387)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 979, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0150793B
Drive has a Rufus Master Boot Record
Partition 1:
  Type: FAT32 (0x0b)
  Size: 7.5 GB (8050966528 bytes)
  Start Sector: 2048, Boot: No
Found USB 2.0 device 'Generic Flash Disk USB Device' (058F:6387)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 979, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0150793B
Drive has a Rufus Master Boot Record
Partition 1:
  Type: FAT32 (0x0b)
  Size: 7.5 GB (8050966528 bytes)
  Start Sector: 2048, Boot: No
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'IR1_CPRA_X64FREV_EN-US_DV5'
  Size: 3.6 GB
  Uses: EFI
  Uses: Bootmgr
  Uses: Install.wim (version 0.13.1)
Using image: -Getintopc.com-en_windows_8_1_pro_vl_x64_dvd_2971948.iso

Format operation started
Requesting disk access...
Opened \\.\PHYSICALDRIVE1 [\Device\Harddisk1\DR3] for write access
Will use 'E:' as volume mountpoint
I/O boundary checks disabled
[... 1764 lines cut ...]
Extracting: E:\sources\sxs\x86_netfx-aspnet_webadmin_secur_res_b03f5f7f11d50a3a_6.3.9600.16384_none_de86328903f0aec1\setupauthentication.aspx.resx (1.7 KB)
Extracting: E:\sources\sxs\x86_netfx-aspnet_webadmin_security_b03f5f7f11d50a3a_6.3.9600.16384_none_7d993bccd0f7a820\security.aspx (9.9 KB)
Extracting: E:\sources\sxs\x86_netfx-aspnet_webadmin_security_b03f5f7f11d50a3a_6.3.9600.16384_none_7d993bccd0f7a820\security0.aspx (1.4 KB)
Extracting: E:\sources\sxs\x86_netfx-aspnet_webadmin_security_b03f5f7f11d50a3a_6.3.9600.16384_none_7d993bccd0f7a820\setupauthentication.aspx (2.5 KB)
Extracting: E:\sources\sxs\x86_netfx-aspnet_webadmin_users_b03f5f7f11d50a3a_6.3.9600.16384_none_4cc4ad18b1f53a26\adduser.aspx (5.2 KB)
Extracting: E:\sources\sxs\x86_netfx-aspnet_webadmin_users_b03f5f7f11d50a3a_6.3.9600.16384_none_4cc4ad18b1f53a26\edituser.aspx (11.7 KB)
Extracting: E:\sources\sxs\x86_netfx-aspnet_webadmin_users_b03f5f7f11d50a3a_6.3.9600.16384_none_4cc4ad18b1f53a26\findusers.aspx (7.8 KB)
Extracting: E:\sources\sxs\x86_netfx-aspnet_webadmin_users_b03f5f7f11d50a3a_6.3.9600.16384_none_4cc4ad18b1f53a26\manageusers.aspx (14.8 KB)
Extracting: E:\sources\sxs\x86_netfx-aspnet_webadmin_users_res_b03f5f7f11d50a3a_6.3.9600.16384_none_cbeb2f8f23cbeb63\adduser.aspx.resx (1.5 KB)
Extracting: E:\sources\sxs\x86_netfx-aspnet_webadmin_users_res_b03f5f7f11d50a3a_6.3.9600.16384_none_cbeb2f8f23cbeb63\edituser.aspx.resx (3.2 KB)
Extracting: E:\sources\sxs\x86_netfx-aspnet_webadmin_users_res_b03f5f7f11d50a3a_6.3.9600.16384_none_cbeb2f8f23cbeb63\findusers.aspx.resx (1.9 KB)
Extracting: E:\sources\sxs\x86_netfx-aspnet_webadmin_users_res_b03f5f7f11d50a3a_6.3.9600.16384_none_cbeb2f8f23cbeb63\manageusers.aspx.resx (3.5 KB)
Extracting: E:\sources\sxs\x86_netfx-aspnet_webadmin_wizard_b03f5f7f11d50a3a_6.3.9600.16384_none_81de9108507fb9bd\confirmation.ascx (2.3 KB)
Extracting: E:\sources\sxs\x86_netfx-aspnet_webadmin_wizard_b03f5f7f11d50a3a_6.3.9600.16384_none_81de9108507fb9bd\wizard.aspx (10.7 KB)
Extracting: E:\sources\sxs\x86_netfx-aspnet_webadmin_wizard_b03f5f7f11d50a3a_6.3.9600.16384_none_81de9108507fb9bd\wizardadduser.ascx (1.6 KB)
Extracting: E:\sources\sxs\x86_netfx-aspnet_webadmin_wizard_b03f5f7f11d50a3a_6.3.9600.16384_none_81de9108507fb9bd\wizardauthentication.ascx (2.6 KB)
Extracting: E:\sources\sxs\x86_netfx-aspnet_webadmin_wizard_b03f5f7f11d50a3a_6.3.9600.16384_none_81de9108507fb9bd\wizardcreateroles.ascx (7.5 KB)
Extracting: E:\sources\sxs\x86_netfx-aspnet_webadmin_wizard_b03f5f7f11d50a3a_6.3.9600.16384_none_81de9108507fb9bd\wizardfinish.ascx (271 bytes)
Extracting: E:\sources\sxs\x86_netfx-aspnet_webadmin_wizard_b03f5f7f11d50a3a_6.3.9600.16384_none_81de9108507fb9bd\wizardinit.ascx (491 bytes)
Extracting: E:\sources\sxs\x86_netfx-aspnet_webadmin_wizard_b03f5f7f11d50a3a_6.3.9600.16384_none_81de9108507fb9bd\wizardpermission.ascx (24.5 KB)
Extracting: E:\sources\sxs\x86_netfx-aspnet_webadmin_wizard_b03f5f7f11d50a3a_6.3.9600.16384_none_81de9108507fb9bd\wizardproviderinfo.ascx (1.7 KB)
Extracting: E:\sources\sxs\x86_netfx-aspnet_webadmin_wizard_res_b03f5f7f11d50a3a_6.3.9600.16384_none_c08fbe6ba6d0e852\confirmation.ascx.resx (775 bytes)
Extracting: E:\sources\sxs\x86_netfx-aspnet_webadmin_wizard_res_b03f5f7f11d50a3a_6.3.9600.16384_none_c08fbe6ba6d0e852\wizard.aspx.resx (2 KB)
Extracting: E:\sources\sxs\x86_netfx-aspnet_webadmin_wizard_res_b03f5f7f11d50a3a_6.3.9600.16384_none_c08fbe6ba6d0e852\wizardadduser.ascx.resx (1.1 KB)
E

pbatard, 2017-04-30:
Duplicate of #379.

Unless your USB doesn't boot, you can ignore this error. Have you actually tried booting your USB?

craneofbloom, 2017-04-30:
not yet, so do you mean i can ignore this error message and use it as it is?

pbatard, 2017-04-30:
Yup.

craneofbloom, 2017-04-30:
thank you for your quick reply. i will try to boot my usb. and here is the picture of the error screen
[image]

craneofbloom, 2017-04-30:
sorry i pressed the close issue

craneofbloom, 2017-04-30:
I configured my computer to boot using that usb, but it did not work,the windows 8.1 logo did not show up,it started normally.what am i gonna do?is it because of the error message while copying the files into the usb?

pbatard, 2017-05-01:
Nah, it should work, but you need to be mindful that you created a BIOS bootable USB. If your computer is UEFI (which is likely if you bought it in the past 5 to 10 years) then you either need to enable Legacy Boot/CSM in your UEFI fimware, or recreate the USB and make sure you select "GPT partition scheme for UEFI".

craneofbloom, 2017-05-01:
I did it., even if a undetermined error message still poped up.i booted the usb and i used the older version of rufus 1.4.6 as stated in this video. https://youtu.be/VmtRE9LTCOE. And i downloaded a different windows 8.1 iso from a different site., but for further knowledge,and for other people, can you teach me how to do the setup you stated above?thanks.

pbatard, 2017-05-01:
> can you teach me how to do the setup you stated above?

I already explained as far as I can go. If you need more of a tutorial, the google is your friend, because this issue tracker is not the place to teach people basic concepts like BIOS, UEFI, MBR or GPT.

Considering that you solved your problem, I will close your issue.

Also, be mindful that I can only ever support __official__ retail ISOs from Microsoft. If you aren't using n official Windows ISO, it's up to you to solve boot issues.

closed 2017-05-01 by pbatard