pbatard/rufus #951: [Enhanchement] Store Checksum (To avoid check same image)

SmileYzn, 2017-05-09:
Hi, have a possibility to store checksums of a image that already have been checked?

I clicked one time on #, and closed the window. But when i click # again the program check image again.

I hope dev can understand my suggestion (OH, add a button to force check again if my suggestion is applicabe here)

Thanks for read

pbatard, 2017-05-09:
I don't think this is a valid suggestion as it would defeat the whole purpose.
Imagine this scenario: Someone downloads an image, and finds out that the cheksum is invalid, which means that their image is corrupted. So they download it again and press `#`. If I was doing what you ask for, they would get the same invalid checksum the second time, even if the image is valid. 

Also, and this is the most important point, a checksum validator should not leave any doubt that the computed checksum match the content of the data as it exists when the computation was launched. Again, if we go with your suggestion, then it is possible for someone or something to change one byte of the data, and still produce a valid checksum, which is NOT acceptable under any circumstances, even if, in 99.99% of cases, the data will not have changed. The point of computing a checksum is to detect these 0.01% cases where alteration occurs.

This being said, the solution to your issue is actually quite simple: As long as you didn't close Rufus, the computed checksums will still be available __in the log__. So all you have to do is press the `Log` button, and you'll be able to copy/paste all the checksums you already computed.

I will now close this issue as I believe I have answered your question.

closed 2017-05-09 by pbatard

pbatard, 2017-05-11:
> That do not make any sense if you do not closed rufus or not load other image.

Yes it actually does, and it is you who are not understanding what I previously explained. Have you __actually__ looked into the Log. As I explained previously, the Log will __STILL__ have the previous checksum __even if you closed the checksum window__.

__PLEASE__ take the time to carefully read what I posted earlier. It contains both a complete explanation on why your request is not receivable, and why what you want (getting instant access to previous computed checksums) is already available.

SmileYzn, 2017-05-11:
No, i not need to check previous checksum, but save some time to not make some check in the same loaded image.

Thanks

pbatard, 2017-05-11:
Sorry, but if you understand how checksums work, there is NO WAY to _"save some time to not make some check in the same loaded image"_ unless you __completely defeat the whole point of using a checksum in the first place__. It just doesn't make any sense to want to reuse old data when the purpose of computing a checksum is to get a unique fingerprint of the data as it exists on the disk __at the time the user selected to launch checskum computation__.

Please heed these words very carefully: _There is no such thing as a shortcut when it comes to security_.

Now, if you haven't closed Rufus and if you have reasons to __BELIEVE__ (buy by doing so, you are actually putting yourself at risk) that your image has not been modified, then you can get the checksum(s) you computed in the Rufus Log, which you can always access by clicking the `Log` button, __even AFTER you closed the checksum window__. On the other hand, if your idea is to have Rufus somehow maintain a database of the images it has previously seen, and store the checksums it computed for them, so that it doesn't have to go through a lengthy computation again, you are __UTTERLY__ missing the whole point of what checksums are for, and you will need to understand that this is a completely ridiculous request.

Please be mindful that I will not be trying to explain this for a 4th time, so it is up to you to take the time to process and understand what am I saying. Again, despite what you may __believe__, I do understand what you are requesting, and it is __YOU__ who still doesn't realize that it makes no sense.

NSbuilder, 2017-05-11:
@SmileYzn I don't think you understand the point of a checksum...