pbatard/rufus #952: Rufus Undertermined error while formatting

antfuze, 2017-05-11:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [x] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive
  - [x] Plugging the USB into a different port
  - [x] Running Rufus on a different computer
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I put in the USB, I select the ISO I want, it automatically selects NTFS & the Partition is MBR partition scheme for BIOS or UEFI-CSM.

I have selected Quick format (as of this moment, currently letting it go in slow format mode). And it is on Standard Windows Installation & the Create extended label and icon files is selected.

I have included the current LOG & if the slow format doesn't work, I will post that one too.

I have tried it on 2 different computers now with 2 different USBs. One is an 8GB & the other is 249GB.

I will also include the # info as soon as the slow format is done.

I have looked, but my Rufus does not include the Rufus version or Windows version or anything like that. If you can let me know where to find that, that would be freat.

But I do have Rufus 2.14.1086
Windows 10 Home  Ver. 1607 OS/Build: 14393.1066
Log
---
```

Format operation started
Requesting disk access...
Opened \\.\PHYSICALDRIVE2 [\Device\Harddisk2\DR19] for write access
Will use 'F:' as volume mountpoint
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Volume does not have an x86 Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Error while formatting
Format error: Undetermined error while formatting.
Re-mounted volume as 'F:' after error

Found USB 2.0 device 'Generic Flash Disk USB Device' (0011:7788)
Removing E: from the list: This is the disk from which Rufus is running!
Found USB 1.1 device 'Lexar USB Flash Drive USB Device' (05DC:C75C)
1 device found
No volume information for drive 0x82
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 974, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x49FCE17A
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS/exFAT/UDF (0x07)
  Size: 7.5 GB (8016363520 bytes)
  Start Sector: 2048, Boot: Yes
Closed Windows format prompt
Closed Windows format prompt

```

antfuze, 2017-05-11:
Whenever gets just past creating file system. A pop up from Windows appears telling me the USB needs to be formatted & then Rufus fails. Is that maybe it? How do I stop Windows automatically trying to format the USB?

pbatard, 2017-05-11:
Sigh...

> I have looked, but my Rufus does not include the Rufus version or Windows version or anything like that.

It does. It's in the log. If you scroll or press `Save Log` and then open the saved file, you __WILL__ have it, period. Unless you mistakenly pressed `Clear Log` (which noone advised you to do), your log will start with the Rufus version.

Unfortunately, because your log is truncated, I can't see any info about the devices you use, and other data, so it's hard to help you...

The only thing I will advise however is:

1. __PLEASE__ run the bad blocks check (don't bother with "slow" format). It's in the checklist __for a reason__!

2. What happens if you try to format your drive to NTFS using the Windows formatting tool (right clock on your drive and select format). Given the nature of your error, I am pretty sure you'll get an error, which means that this is not a Rufus issue.

pbatard, 2017-05-16:
No update &rarr; Closed.

closed 2017-05-16 by pbatard