pbatard/rufus #953: Feature suggestion: allow to make a full SD backup\clones with Rufus

centur, 2017-05-15:
Checklist
---------
- [x ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [n/a] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [n/a] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [n/a ] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [ n/a] I also tried one or more of the following:
  - [n/a] Using a different USB drive
  - [n/a] Plugging the USB into a different port
  - [n/a] Running Rufus on a different computer
- [n/a] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Feature suggestion 
-----------------
Hi, 
thanks for the great tool and the time you spend on it. I found that every time when I need to write images to sd cards and googling for tools - I ended up using Rufus (now it's in my Portable apps stash). 

But there is one feature that is currently missing and seems logical for the SD card tool - the ability to make a full copy of an existing SD card for backup or cloning purposes. Any format that is suitable for rufus would work - I think there is no point of chasing multiple compatible formats for other tools - make a backup with Rufus, restore a backup with the same tool. Given that Rufus is blazingly fast - I don't think there will be anyone competing in the speed of restoration.

Is it possible to add this feature to Rufus, so users can use a single tool for backing their bootable SD cards or cloning them for multiple devices i.e. set up a proper dashboard for Raspberry PI and then clone it and distribute your work for others to use?

pbatard, 2017-05-15:
Feature's already there. You can use <kbd>Alt</kbd>-<kbd>V</kbd> in Rufus to back up the current drive to (uncompressed) VHD.

I will therefore close this issue.

closed 2017-05-15 by pbatard

centur, 2017-09-24:
I was returning back to this issue 7-10 times to find that hotkey over and over again. Any chance that Rufus will get a screen which list all hotkeys and what they do ?

pbatard, 2017-09-25:
They are listed at the [end of the FAQ](https://github.com/pbatard/rufus/wiki/FAQ#Power_keysCheat_modes).

Because they are cheat modes, and therefore __unsupported__, they will never be listed in the application.

centur, 2017-09-25:
Great. Sad that they are unsupported :( . Still want to praise Alt+V - it literally saved our dashboards raspberryPI setups like 4 times from being lost. My personal most used feature  :)

Thanks for the great tool.

NSbuilder, 2017-09-25:
I think Rufus need a screen / place to show all cheat modes and what they do.

pbatard, 2017-09-25:
I __strongly__ disagree. This would lead to users believe that these cheat modes are supported, whereas they aren't (which is why they are present as __hidden__ cheat modes).