pbatard/rufus #954: GUID Partition Table (GPT) header contains invalid CRC-32 checksum of GPT entries array

JoseJimeniz, 2017-05-15:
Checklist
---------
- [X] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [X] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [X] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [X] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [X] I also tried one or more of the following:
  - [X] Using a different USB drive
  - [X] Plugging the USB into a different port
  - [X] Running Rufus on a different computer
- [X] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------

I am attempting to write a [DD image](http://mirror.wdc1.us.leaseweb.net/opnsense/releases/mirror/OPNsense-17.1.4-OpenSSL-vga-amd64.img.bz2). The image contains a GUID Partition Table (GPT) disk image, as well as a protective MBR in front of it.

GPT partitions contain CRC-32 checksums of the header and partition table entries to check for corruption; as well as backup copies of the GPT header and GPT entries placed at the end of the disk.

MBR & GPT partitioning
------------------------

In traditional MBR disks, **Sector 0** contains both:

- boot code
- partition table

The BIOS reads the contents of the **Sector 0** into memory and `JMPs` to the first byte - which is executable code. With GPT partitioning, the partition table actually exists in **Sector 1**.  But they also create a dummy, or **Protective MBR** in sector 0.

- Partition Type: `EE` *(protective MBR)*
- Size: *the size of the stuff on the disk*

This is done so that any disk tools that don't understand GPT will see that there is some sort of partition present, and leave it alone.

Sector 1 - The GPT partition table
--------------

The Guid Partition Table on this image, as well as the resulting USB stick, is easily decoded:

**Sector 1 - Main GPT**

    0200: 45 46 49 20 50 41 52 54                         ;Signature: "EFI PART"
    0208: 00 00 01 00                                     ;EFI Version: 0x0001 0x0000 ==> 1.0
    020C: 5C 00 00 00                                     ;Header size: 0x0000005C ==> 92 bytes
    0210: C2 9E 77 DC                                     ;CRC-32 checksum of all 92 bytes (correct)
    0214: 00 00 00 00                                     ;Reserved. Must be zero.
    0218  01 00 00 00 00 00 00 00                         ;LBA of main   GPT Header. 0x0000000000000001 ==> Sector 1
    0220: 22 57 1C 00 00 00 00 00                         ;LBA of backup GPT Header. 0x00000000001C5722 ==> Sector 1,857,314
    0228: 03 00 00 00 00 00 00 00                         ;First usable LBA.         0x0000000000000003 ==> Sector 3
    0230: 20 57 1C 00 00 00 00 00                         ;Last  usable LBA.         0x00000000001C5720 ==> Sector 1,857,312
    0238: 89 48 F5 57 41 14 E7 11 A9 9F 00 0C 29 54 E1 33 ;DiskGUID {8948F557-4114-E711-A99F-000C2954E133}
    0248: 02 00 00 00 00 00 00 00                         ;Partition entry LBA       0x0000000000000002 ==> Sector 2
    0250: 03 00 00 00                                     ;Number of Partition Entries. 0x00000003 ==> 3 entries
    0254: 80 00 00 00                                     ;Size of partition entries:   0x00000080 ==> 128 bytes
    0258: EF 8E E1 B6                                     ;CRC-32 of partition entries array

The interesting part comes from the two CRC-32 checksums

    0210: C2 9E 77 DC      ;CRC-32 checksum of all 92 bytes (correct)

In order to calculate the checksum of the header, you calculate the CRC-32 of the first **`HeaderSize`** bytes, while setting the four checksum bytes to `00 00 00 00`:

> 0200: 45 46 49 20 50 41 52 54 00 00 01 00 5C 00 00 00   
> 0210: **00 00 00 00** 00 00 00 00 01 00 00 00 00 00 00 00   
> 0220: 22 57 1C 00 00 00 00 00 03 00 00 00 00 00 00 00   
> 0230: 20 57 1C 00 00 00 00 00 89 48 F5 57 41 14 E7 11   
> 0240: A9 9F 00 0C 29 54 E1 33 02 00 00 00 00 00 00 00   
[... 205 lines cut ...]
Opened \\.\PHYSICALDRIVE2 [\Device\Harddisk2\DR2] for write access
No drive letter was assigned...
Will use 'G:' as volume mountpoint
Analyzing existing boot records...
Drive has an unknown Master Boot Record
Volume does not have an x86 Partition Boot Record
Writing Image...
Drive has an unknown Master Boot Record

Found non-USB removable device 'KINGSTON SHFS37A240G' => Eliminated
Found USB 2.0 device 'Samsung Flash Drive USB Device' (090C:1000)
Found non-USB removable device 'WDC WD1003FZEX-00MK2A0' => Eliminated
1 device found
No volume information for drive 0x82
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3899, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has an unknown Master Boot Record
Partition 1:
  Type: GPT Protective MBR (0xee)
  Size: 906.9 MB (950945280 bytes)
  Start Sector: 1, Boot: No

```

pbatard, 2017-05-16:
> I am attempting to write a DD image.

If you do that in Rufus then I can tell you that no other operation is performed by the application besides writing the data to disk, exactly as it is on the image. Especially Rufus does not reprocess __any__ of the content it just wrote (to calculate CRC or even rewrite the partition table).

As you probably understand, there's really nothing much that needs to be done when writing a DD image, and all the code basically stands in the [`WriteImage()`](https://github.com/pbatard/rufus/blob/master/src/format.c#L1510-L1608) call, which, despite some small Windows quirks is really as simple as `Read a set of blocks from the source image, Write the same set of blocks to the drive starting at offset 0, Repeat until the source data is depleted`. Once that operation is complete, Rufus declares success and asks Windows to re-enumerate the drives (which is a read-only operation as far as Rufus is concerned).

Therefore, if any CRC32 alteration occurs after the image has been written, it's pretty clear that it comes from Windows (which is already known for [polluting data on DD-written drives](https://github.com/QubesOS/qubes-issues/issues/2051#issuecomment-224249753)). Especially, the `IOCTL_DISK_SET_DRIVE_LAYOUT_EX` call is __NOT__ used at all when writing a DD image (because it's clearly not needed).

In other words, whatever you have picked up __is__ a Windows bug, and you ought to report it to Microsoft, even more so if, as I gather, you have picked the same issue with Win32DiskImager and Etcher. Else, you're going to spend your time asking for developers of DD-writing application to work around a Windows bug, which might not be as straightforward to work around as you assume.

For the record, the Windows Insider program from Microsoft does welcome feedback, so if you can demonstrate the issue with multiple applications, you should very much report it.

> Rufus can still work around it. Rufus can calculate the perform a sanity check on the generated GPT header (and backup header) and optionally fix it before completing.

That might not be that simple, because you are assuming that Windows changes the CRC immediately, whereas this is most likely a delayed operation (just like the creation of the `System Volume Information\` directory on a mountable partition from a new drive is delayed), even more so as Rufus does keep an exclusive lock on the drive while it writes it (which should keep even Windows away).

So what would most likely be needed a as workaround is a monitoring service, that runs after completion, that checks the CRC32 of GPT drives to find out if Windows has altered them.

Furthermore, there is no telling if Windows and that monitoring service would not be fighting. For instance, have you considered that the following is the most likely scenario?

1. Rufus writes an image in DD mode. When it is done, because it has an exclusive lock, the CRC32 is correct (or at least, it is the same one as the image).
2. As soon as Rufus releases the lock, Windows, which detected that the drive has been altered, alters the CRC32
3. Shortly after that (and hopefully before the user gets a chance to eject the drive, which is a big 'IF' - and don't get me started on the issue this will likely create if the user doesn't select to safely eject the drive), the new Rufus monitoring service checks the CRC32, finds that it was modified, and writes a corrected version
4. However, since Windows controls and monitors the write channels, and can therefore find out whether the content of a partition table might have been altered (which the OS probably want to be altered about, even if it's just to change the CRC32), we may find out that Windows will reparse the partition table, find that its CRC32 is not to its liking, and alter it back again.
5. And now we have a potential endless loop with Rufus and Windows fighting to write what they see as the "correct" CRC32... Not good.

All this to say that, before I want to commit to try to do anything about this in Rufus, I would very much like to see a response from Microsoft with regards to this issue.  Bugs that appear to originate from a specific software are better addressed to the makers of that software, even if the software happens to be a proprietary OS...

JoseJimeniz, 2017-05-16:
For people with this problem, i wanted to documented how you can repair the written flash disk. There is nothing wrong with your downloaded original disk image; it's the version written to a flash stick (using Windows APIs rather than written directly) that cause it contain invalid GPT checksums.

You can use the free hex editor **HxD** to open the physical USB disk for writing. This lets you directly edit the data on the USB disk.

1. Open the physical USB disk for writing (for the love of all that is holy; make sure you're opening the right disk - you don't want to destroy your hard drive partition):
   [image]

2. Go to **Sector 2**, and select the first **384 bytes** (180h bytes):

    [image]

3. Calculate the **CRC-32 Checksum** of the 384 selected bytes by going to **Analysis** -> **Checksums...**:

    [image]

4. HxD will show you the calculated 4-byte checksum. In the case of my 2GB USB stick, the checksum is `0xB6E18EEF`:

[image]

5. Return to **Sector 1** and overwrite the 4-byte checksum at offset 0x0258

    [image]

6. Overwrite the GPT Partition Entries checksum with your calculated checksum. Remember to enter it in *little-endian* order. In my case that means entering `0xB6E18EEF` as `EF 8E E1 B6`:

    [image]

7. Now that we've modified the contents of the Sector 1 GPT Header, we need to recalculate the checksum for **Sector 1** stored at offset 0x0210:

    [image]

8. The first step in calculating the checksum requires setting the existing four checksum bytes to all zero (`00 00 00 00`):

    [image]

9. Now we can select the first 92 bytes (0x5C bytes) of **Sector 1**:

   [image]

10. Calculate the CRC-32 checksum of the selected 92 bytes by going to **Analysis** -> **Checksums...**, and again calculating the checksum of the selected bytes. In my case the checksum is `0xAE3C7980`:
    [image]

11. Overwrite the GPT header checksum at offset 0x0210 with your computed checksum. Remember to enter it in little endian order. In my case the means to enter it as `80 79 3C AE`:

    [image]

And then click **Save** (Ctrl+S).

When you try to boot off the stick, OpenBSDs boot-loader will warn you that the checksums in the **backup** GPT entries doesn't match; but it will continue with the boot cycle. If you wish you can repair the checksums of the backup GPT header.

------------------------

The fact that Rufus relies on the Windows API to write data to a physical disk is an implementation detail.  A USB writing tool is free to write the image directly; bypassing the buggy Windows code.

JoseJimeniz, 2017-05-16:
What i'm left with is trying to find minimally reproducible code. Rufus *does* crawl the GPT data structures in the image in order to find out how many partitions and what sizes to create? 

And that information is populated into:

- `DRIVE_LAYOUT_INFORMATION_EX`
- `DRIVE_LAYOUT_INFORMATION_GPT`
- `PARTITION_INFORMATION_EX`

Correct?

pbatard, 2017-05-16:
> Rufus does crawl the GPT data structures in the image in order to find out how many partitions and what sizes to create?

Incorrect.

Again, when writing an image in DD mode, there is no explicit creation of a partition layout by Rufus, because the partition layout is __already part__ of the image data that's going to be written.

The only place where Rufus will use the drive layout information structures, in DD mode, is after the operation has been __fully completed__, when Rufus re-enumerates the drive so that it can let the user know how many partitions exist on that drive, their sizes, etc. But that's a __READ ONLY__ operation.

I want to make sure that you understand that a DD image is for the whole disk. There is absolutely no need to fiddle with partition creation when writing in DD mode. You just write the whole data, and that's it.

pbatard, 2017-05-16:
> so Rufus doesn't fix the image written to a USB stick to account for the fact that the partition information would be invalid.

Nope, because that's the OS job.

> Which means that __someone__ sees the USB disk contains an invalid partition and attempts to repair it

That someone __is__ the OS.

> Someone then attempts to repair the invalid partition, but it calculates one of the checksums incorrectly.

And that is why you need to fill a Windows bug with Microsoft, because, as I explained, even if Rufus were to create the backup GPT, if Windows decides it doesn't like its checksum (and, as we've established, Windows does monitor the partition tables to see if they need fixing), you'll be back to square one.

Please fill a bug with Microsoft already, okay?

> i don't know if the Unix dd tool handles GPT images correctly

Trust me, it will behave just like the Windows dd tools. By essence, a dd operation is designed to be dumb and not attempt to interpret/alter the data it writes. 

Please consider this, what if someone created an image with an invalid backup copy of the GPT, precisely to test the behaviour of various systems when presented with invalid or non-existent backup data. If `dd` were to alter the data written (even if that data falls outside of the size of the original image), I can guarantee that you'd get a whole bunch of UNIX people rightfully in arms because a tool is doing something that it was never asked to do. As such, I'm not sure I'd want to do anything in Rufus either.

It's called a __backup__ table for a reason. It's the job of the OS to create or alter it if it finds it is different from the prime table. And if, during that update, the OS does compute a wrong checksum, then it's clearly an OS bug that needs to be fixed in the OS.

pbatard, 2017-11-22:
Closing this. This was kept open in case I had a chance to take a second look at it, but it's pretty clear that it's not going to happen.

closed 2017-11-22 by pbatard

JackTrapper, 2017-11-22:
The guys over at OpnSense worked around the bug by manually created a dummy 4th partition. Yes, ideally something that manually creates partitions on drives would create them without invalid checksums; but at least we know how to manually repair them using HxD