pbatard/rufus #957: Windows XP x64 Error

KidProPlus, 2017-05-19:
Rufus support Windows XP perfect, But no Support Windows XP x64, Please Support Windows XP x64.
I download Windows XP x64 at [https://msdn.microsoft.com/en-us/subscriptions/downloads/#searchTerm=Windows&ProductFamilyId=0&Languages=en&Architectures=x64&ProductFamilyIds=140&FileExtensions=.iso&PageSize=10&PageIndex=0&FileId=0](url)
Rufus call this ISO no bootable, but I can using this ISO file to Install on VMware normal, no problem appear.

pbatard, 2017-05-19:
> Please Support Windows XP x64.

[Windows XP is no longer supported by Rufus](https://github.com/pbatard/rufus/wiki/FAQ#Windows_XP_NO_LONGER_SUPPORTED)

closed 2017-05-19 by pbatard

blackcrack, 2017-05-19:
and Reactos ? \*bg\*
would be bad if it's work not in reactos..

pbatard, 2017-05-19:
Well, have *you* tried???

blackcrack, 2017-05-19:
well Reactos it should later 100% XP compatieble,
if you drop the support, then it is not working in Reactos, buuuutttt
moooooooment \*vbg\*
oeehhhmmm.. wow.. 
https://picload.org/view/rirpcrlw/screen00022.jpg.html
https://picload.org/thumbnail/rirpcrlw/screen00022.jpg
https://picload.org/image/rirpcrlw/screen00022.jpg
https://picload.org/infonail/rirpcrlw/screen00022.jpg

beeeeest regards
Blacky

pbatard, 2017-05-19:
> Reactos (...) should (be) 100% XP compatible,

Emphasis on **should**.

Also, the ReactOS team are fully aware that requiring Windows 2003 compatibility [is a bad idea](https://www.reactos.org/pipermail/ros-dev/2015-March/017216.html):

> The current approach of pure Windows 2003 Server SP2 compatibility on user mode side is a dead end. Our target OS version is starting to become a fossil.

So it's up to ReactOS to solve the issue of non Windows 2003 compatibility, __NOT__ Rufus, and as such __I will not do anything to improve ReactOS compatibility if it involves having to support versions of Windows that Microsoft no longer supports__.

Note that this should not be interpreted to mean I don't want to support ReactOS.

For what is worth, I went out of my way in Rufus to add a ReactOS bootloader installation feature, so, I hope that people realize that I'm already doing what I can to support the ReactOS effort. However, as far as I'm concerned, as long as ReactOS haven't sorted the "boot ReactOS from USB on a modern system" issue, and regardless of the effort that its developers have been putting into it, it's only going to remain a toy project, due to being too impractical to test or use.

As such, the amount of effort I am willing to invest in ReactOS support, when I have more pressing things to do, will remain very limited.

Finally, I'll also point out that there is a world of difference between saying _"if you are using Windows XP, then your are on your own as far as Rufus is concerned"_, which is all I have been doing, and dropping support for Windows XP (32-bit) altogether, which I have not done yet.

Last time I checked, Rufus worked fine on Windows XP (32-bit), and I'm not planning to drop XP support until Rufus 3.0, which is still a LONG TIME away. The only thing that I am not planning to do, ever, is add XP 64-bit support, because there wasn't enough demand for this when I still officially supported XP, and it is now an obsolete request.

blackcrack, 2017-05-19:
:)

well, Reactos work together with the wine Project and i guess, they whant ever try to stay by side of the actual System of WinNT and the programmer , like i guess, want not directly hold the lines behind XP
and they want, like i bet, more later, but the first milestone it is easy XP .. the Reactos it is easy a reaction  by one hand because the cloused source Situation and on the other hand a free source  to be in able to make much more what it is possible and what want make the User and  not a Designer pay'ed from Microsoft and hold the Branded Surface what like Billyboy or something.. "we are microsoft..beep...we have own branded system..beeb..and it is Copyrighted ..beeb.. and it is cloused ..beeb .. it is a Factory secret ..beeb.." bahh.. and the 0day exploits it is also branded hea Microsoft .. this is also why it is building Reactos also and not only rebuild a specially system.. so, let us see how it self at all developed :) 
i be in happy i'm curios how it at all developed.. because, if you test the daily build do you see it is more under the graphical desktop as only a xp-replica ;) i have build a reactos-test-packet.iso with unattended vb
 and programming ide silent installer, it is running memorably really pretty well :)
( [filelisting.txt](https://github.com/pbatard/rufus/files/1014734/filelisting.txt) )and so on..
If you test the actual Daily build want you also be memorably :)

best regards
Blacky

pbatard, 2017-05-19:
> well, Reactos work together with the wine Project

However they do have very different goals.

> they whant ever try to stay by side of the actual System of WinNT and the programmer

That's fine and all, but if, by the time they are done, Windows applications have long dropped XP support, then that effort will be wasted.

>  the Reactos it is easy a reaction by one hand because the cloused source Situation

I hope you do realize that you are talking to a Free Software developer, who only happened to create Rufus because he could not stand closed source software... You don't have to try to convince __me__ that Free Software is miles better than closed source.

> if you test the daily build do you see it is more under the graphical desktop as only a xp-replica

Yes, but [they have broken what I consider to be a critical feature](https://www.reactos.org/wiki/LiveUSB), and left it broken for years, which, IMO, is not helping rally people to their cause.

If Rufus can run on a specific OS, then, at the very least, I will expect that platform to be bootable from USB, so that you can use Rufus to create installation media or live media for that platform. Sadly, however, that is not the case for ReactOS, which makes trying to support it in Rufus a bit pointless...

By the way, this was a major issue with XP, because, unlike what is the case for Windows Vista and later, __Microsoft never designed XP to be bootable from USB__. This means that I had to spend about one month coming up with a bootable ISO &rarr; bootable USB conversion process for XP-32 (which isn't guaranteed to work, as it will still fail to boot on some Dell models), which was a major pain in the ass.

Thus, if by following the goal of being XP-like, ReactOS don't plan to sort their boot and run from USB issues, I'm going to consider ReactOS support in Rufus to be a bit pointless.