pbatard/rufus #958: Add transparent Zip support

maffe, 2017-05-21:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [x] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive
  - [ ] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I would like to see transparent unzipping for images (useful for Raspbian, https://www.raspberrypi.org/downloads/raspbian/). If a Zip file is selected as the source image and it contains only a single file, extract that file on the fly while writing on the destination drive (maybe add a checkbox to toggle extraction).

This would make the process easier, faster and less hardware-consuming (no additional temporary space needed on the source drive and less write cycles which might be relevant for SSDs).

Log
---
```
Rufus version: 2.15.1117
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0407
Will use default UI locale 0x0407
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'de-DE'
Found USB 2.0 device 'Generic Ultra HS-SD/MMC USB Device' (0424:4063)
Device eliminated because it appears to contain no media
0 devices found
```

pbatard, 2017-05-21:
This is already in there. Have you __actually__ tried?

The filter on the `DD Image` selection (Since Raspbian is a disk image and not an ISO image) actually has `*.zip`, which should provide some indication that .zip compressed images are supported.

Unless you are experiencing an issue with .zip usage, I'm going to close this issue as _already implemented_.

closed 2017-05-21 by pbatard

maffe, 2017-05-21:
I have tried. The file chooser file type dropdown shows:

- `ISO-Abbild (*.iso)`
- `Alle Dateien (*.*)`

After selecting the zip file, the status bar said 

> Keine Abbild-Datei gewählt

I don’t know what went wrong, maybe the file was still locked by the download manager?

The website and the FAQ don’t mention Zip support.

I tried again and it worked. I thought it would write the Zip file as-is.

Okay, now I see two possible improvements:
- Fix file chooser to display Zip files
- Show the file name of the image inside the Zip file instead of the Zip file name in the status bar. The status bar shows:
> Verwende Abbild: 2017-04-10-raspbian-jessie.zip

pbatard, 2017-05-21:
> The file chooser file type dropdown shows:

You're not using the right mode. You must make sure that you change the dropdown before the small disc button from "ISO Image" to "DD Image". The Raspian download is __NOT__ and ISO image, therefore the filters associated with ISO will not work. Once you use "DD Image", as I mentioned earlier, you will get `*.zip`in the image.

Oh, and before you complain that Rufus should not have two separate modes for DD or ISO images, I do have some vague plan to sort that in version 3.0. But that's not gonna happen for a while.

> After selecting the zip file, the status bar said `Keine Abbild-Datei gewählt`

Clearly this was an issue on your side, and not with Rufus. The image selection code is exceedingly generic and does not behave differently depending on the type of file &rarr; __NOT__ a Rufus issue.

> The website and the FAQ don’t mention Zip support.

But the filters for "DD Image" do, which is what matters. There are tons of Rufus capabilities that are not mentioned on the website, on account of not being the ones that people will look for most.

> * Fix file chooser to display Zip files

It's __ALREADY__ there, if you use the proper selection in the "Create bootable disk using..." dropdown. Raspbian is not an ISO image, so you should not have been using "ISO Image" in the first place. If you had used "DD Image", you would have seen that your `.zip` was listed by default.

> Show the file name of the image inside the Zip file instead of the Zip file name in the status bar.

That's not a valid request, because in 99.9% of cases, this will be pointless, and it would require adding custom code, that would then have to be maintained and checked for regressions, for __exceedingly little benefit__.