pbatard/rufus #959: automake-1.14 missing, but automake-1.15 exist

szg0000, 2017-05-23:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
**There was a similar problem, but not the same. Link: #828** 

- [x] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive
  - [x] Plugging the USB into a different port
  - [x] Running Rufus on a different computer
- [x] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
I tried to compile Rufus on "Debian testing". Automake-1.14 missing, but automake-1.15 exist.

```
szg@debian:~/rufus$ make
 cd . && /bin/bash /home/szg/rufus/missing automake-1.14 --foreign --ignore-deps
/home/szg/rufus/missing: sor: 81: automake-1.14: parancs nem található
WARNING: 'automake-1.14' is missing on your system.
         You should only need it if you modified 'Makefile.am' or
         'configure.ac' or m4 files included by 'configure.ac'.
         The 'automake' program is part of the GNU Automake package:
         <http://www.gnu.org/software/automake>
         It also requires GNU Autoconf, GNU m4 and Perl in order to run:
         <http://www.gnu.org/software/autoconf>
         <http://www.gnu.org/software/m4/>
         <http://www.perl.org/>
Makefile:246: recipe for target 'Makefile.in' failed
make: *** [Makefile.in] Error 1

szg@debian:~/rufus$ ls -l /usr/bin/automake*
lrwxrwxrwx 1 root root     26 ápr   30 09:14 /usr/bin/automake -> /etc/alternatives/automake
-rwxr-xr-x 1 root root 253209 jan   25 19:15 /usr/bin/automake-1.15
szg@debian:~/rufus$
```

pbatard, 2017-05-23:
As it is a __Windows-only__ application, Rufus cannot be compiled on Debian (unless you are using MinGW).

In that case, my answer from #828 still applies:

> You're probably missing autoconf or automake, or they are misconfigured. Also, I hope you do realize that you don't need to recompile Rufus (there are executables readily available), that Rufus is Windows only (you won't be able to recompile a Linux version, if that's what you are attempting to do), and that you need MinGW installed (and properly configured) to be able to build it in the first place. For instance, Linux's default gcc will not work.
>
> Anyway, this is clearly a toolchain issue that I expect a developer to be able to sort out by themselves (and if they don't, they should be able to use google to do so, not have to open an issue with a project they are trying to compile, since this is NOT an error that comes from that project), so I will close this issue.

closed 2017-05-23 by pbatard

szg0000, 2017-05-23:
Thanks. As I see in the error message, rufus needs automake-1.14, but there is a newer version, /usr/bin/automake-1.15.  Is there any special reason to use 1.14 instead of latest version of automake?

pbatard, 2017-05-23:
The default files are the ones that are meant to work with the most recent version of msys2/MinGW on Windows. Again, __Any other configuration than msys2/MinGW on Windows is NOT supported__.

The only (very last) thing I will say is: I expect anyone attempting to compile Rufus (using MinGW) on Linux to also be familiar with checking if a `bootstrap.sh` script exist, and run it if it does.

If you need more help than this, I will kindly ask you to either use the only supported compilation platforms (Visual Studio 2017 or msys2/MinGW __on Windows__) or figure it out on your own.

NSbuilder, 2017-05-25:
> use the only supported compilation platforms (Visual Studio 2017 or msys2/MinGW on Windows)
Well, I didn't know that you should only use the **msys2 MinGW** to compile Rufus.
Perhaps you could point that in the README file.

pbatard, 2017-05-25:
> Well, I didn't know that you should only use the msys2 MinGW to compile Rufus.

You are __minsinterpreting__ what is being said. What I said is that I will only __support__ msys2/MinGW if you have an issue with MinGW. That's very different from saying __you MUST use msys2/MinGW__

As per the README file:

> __Compilation__
> Use either Visual Studio (2017 or later), WDK 7.1 (Windows Driver Kit) or MinGW

__If you know what you are doing__, then using MinGW on Linux does work. But I really don't have the time to educate newbies on how they should get compilation going for MinGW outside of the latest msys2/MinGW on Windows.

NSbuilder, 2017-05-28:
> You are minsinterpreting what is being said. What I said is that I will only support msys2/MinGW if you have an issue with MinGW. That's very different from saying you MUST use msys2/MinGW

You didn't understand me :)
I understood what you said.
I'm saying that you should point in the README file that if someone have an issue with MinGW - you will only support msys2/MinGW ( not that you MUST use it, of course you don't ).
Apparently, I did not explain my words properly.

pbatard, 2017-05-28:
There are many things that I could point out in the README. But then the README would become the FAQ. A README is meant to be short and to the point. Going through very special cases that __very few__ people will encounter is not README material.