pbatard/rufus #961: Rufus has completely deleted all my group policy

Isyndia, 2017-05-29:
[image]
[image]

THANKS.......

Isyndia, 2017-05-29:
Putain...

pbatard, 2017-05-29:
Rufus does alter group policies to __temporarily__ disable autorun. But is it doing so in the secure manner advertised by Microsoft, so that, even if there's a general failure (power failure, etc.) the group policies should be left in a known state and undamaged. At least, that's the promise that Microsoft makes when using the Group Policy APIs that Rufus uses.

On the other hand, Rufus was never designed to be prevented to perform its regular tasks by an AntiVirus solution, so I can't guarantee that a conflict between Rufus and your AntiVirus will not cause trouble. What AntiVirus solution are you using? And have you contacted their support team to find out if they are aware of Group Policy corruption issues, when filtering Group Policy access?

pbatard, 2017-05-29:
Closed the issue by mistake - reopening it.

pbatard, 2017-05-29:
I think [this](https://social.technet.microsoft.com/Forums/windowsserver/en-US/84832e0f-2d84-41a3-aacb-5ad30b1fc84b/forum-faq-domain-administrators-cannot-edit-a-gpo-on-windows-server-2012-or-2012-r2-domain?forum=winserverGP) may help:

1. Go to the following system path:
`%systemroot%\Windows\SYSVOL\domain\Policies\<GUID of GPO that can’t be edited>\User`
2. Check whether there is 0 KB registry.pol file under the path.
3. If yes, delete the file to fix the issue.

Isyndia, 2017-05-29:
It works... but I have lost all of my settings. Sorry for my agressivity, and my English, I'm french.

I have a lot of job to rebuild all my configuration... tant pis.

Merci quand même

pbatard, 2017-05-29:
Je comprends. Et je compatis.

J'aimerais bien pouvoir déterminer s'il y a quelque chose à faire, du côté de Rufus, pour éviter que ça arrive à d'autres personnes. Mais pour celà il faudrait savoir si c'est un problème unique, ou quelque chose qui peut être reproduit systématiquement, et je me doute que vous avez pas trop envie d'essayer de réessayer à nouveau.

Par contre, si possible, j'aimerais bien avoir plus de détails sur l'AntiVirus. D'après les images, c'est ESET Internet Security mais s'agit t'il de la version que l'on peut télécharger à https://www.eset.com/int/home/internet-security/download/ (auquel cas je pense que je vais essayer le Trial pour voir), et si oui, est-ce que ça a été configuré de manière particulière?

Isyndia, 2017-05-29:
C'est bien gentil de parler français ! Il faut dire que je ne maîtrise pas tellement l'anglais... malheureusement.
J'ai peut-être fais des manipulations incorrectes. Bref... c'est peut-être un problème isolé.

Oui il s'agit bien de cet antivirus. Il est assez puissant pour une utilisation quotidienne. La config a été complètement remaniée par mes sois: en l'occurence, le système HIPS (Host Intrusion Prevention System) est activée, l'autodéfense également. Le mode est sur intelligent avec des règles personnalisées selon les applications présentes sur mon système.

pbatard, 2017-05-29:
Merci pour ces précisions. Je vais voir ce que je peux faire pour tested ESET. 

Et bon courage pour restorer la config!

Isyndia, 2017-05-29:
Merci ;)

pbatard, 2017-06-16:
OK, I've been playing with ESET, and set filtering to "Interactive" in the Advanced Setup:

[image]

This allowed me to get the ESET prompt about Rufus accessing the GroupPolicies as per the screenshot below.
However, I haven't seen any issue with the Group Policies afterwards. I was able to access the Local Group Policies without trouble, and the way Rufus is designed is that, because Group Policy access is not critical to its operations, if it can't access them (because of ESET or any other reasons) it will give up after a few seconds, as indicated by the `SetLGP: Killing stuck thread!` message:

[image]

All in all then, it doesn't look to me like, even with ESET's rules in place, Rufus was ever in a position to alter or delete Group Policies so I will dismiss this issue as something that was environmental to OP's machine and external to Rufus.

closed 2017-06-16 by pbatard