pbatard/rufus #962: undetermined error while formatting in rufus

shaan5186, 2017-05-30:
Rufus version: 2.4.757
Windows version: Windows 7 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x0409
Found USB 2.0 device 'Generic Flash Disk USB Device' (0011:7788)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1013, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0xFDC01076
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 7.8 GB (8336146944 bytes)
  Start Sector: 63, Boot: Yes, Recognized: Yes
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'J_CCSA_X64FRE_EN-US_DV5'
  Size: 4083853312 bytes
  Has a >64 chars filename: No
  Has Symlinks: No
  Has a >4GB file: No
  Uses Bootmgr: Yes
  Uses EFI: Yes
  Uses Grub 2: No
  Uses Grub4DOS: No
  Uses isolinux: No
  Uses KolibriOS: No
  Uses ReactOS: No
  Uses WinPE: No
  Uses Install.wim: Yes (version 0.13.1)
Using image: en_windows_10_multiple_editions_x64_dvd_6846432.iso

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
Will use 'L:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Windows 7 Master Boot Record
Volume has an unknown Partition Boot Record
Deleting partitions...
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Error while formatting
Format error: Undetermined error while formatting.
Re-mounted volume as 'L:' after error

Found USB 2.0 device 'Generic Flash Disk USB Device' (0011:7788)
1 device found
No volume information for drive 0x81
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1013, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x01BA8982
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS/exFAT/UDF (0x07)
  Size: 7.8 GB (8335130624 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Found USB 2.0 device 'Generic Flash Disk USB Device' (0011:7788)
1 device found
No volume information for drive 0x81
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1013, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x01BA8982
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS/exFAT/UDF (0x07)
  Size: 7.8 GB (8335130624 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Closed Windows format prompt

pbatard, 2017-05-31:
Didn't follow the checklist. 
Please run a bad blocks check, and also, you probably want to try and see what happens if you use the Windows formatting utility.

Next time, if you don't want your issue to be closed right away, please follow the checklist. It exists for a very good reason.

closed 2017-05-31 by pbatard

shaan5186, 2017-06-01:
sir, i run my usb drive for bad blocks some error were found and fixes. but when i use rufus same error again( undetermined error while formatting) . this error only see in rufus if i use power.iso and other usb bootable tool no error show ..please sir help me

pbatard, 2017-06-01:
> i run my usb drive for bad blocks some error were found

Your USB flash drive is defective

> and fixes

No, Rufus does __NOT__ fix errors. If errors were detected during the bad blocks check then your USB flash drive __IS__ defective, and you will have to use another one. There's nothing Rufus can do to help you.

shaan5186, 2017-06-03:
yes, my pen drive defective , when i use other pen drive error not show ..
thank you sir..