pbatard/rufus #964: Usb unusable after writing in dd mode

encloinc, 2017-06-01:
My usb is unusable after selecting dd mode when the prompt popped up. Whats happening is that windows is detecting my usb as a CDROM drive, which disables my ability to create partitions on the usb through microsoft's diskpart utility. Its giving me the error "No media" for every single thing i tried to do with the usb through the cmd. It turns out the usb is formatted into mbr, but I cant figure out how to reverse it to a normal FAT32 based usb drive. Any ideas on how to fix this?

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [ ] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive
  - [ ] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>

Log
---
```
<FULL LOG>
```

ich0x, 2017-06-02:
What kind of image did you DD?
This sounds like less of an issue for Rufus and a mistake on your part with the image.  The image might of been intend as an ISO image, setting Rufus to prepare it as such.
Depending on the image your dd might be a live linux distro you could boot directly to, and change partitions as needed from it after booting it.

You might be able to recover the usb using a combination of bootsect.exe and diskpart.exe from a recovery console (for win8 thru 10 check these steps, my preferred is holding shift key during the reboot dialog and through out - https://www.tenforums.com/tutorials/2294-boot-advanced-startup-options-windows-10-a.html ).  Or try recovery with a USB stick low-format tool.

My personal opinion I'd suggest using Rufus again with a small live linux distro with some form of parted to recreate the partitions on your "broken" usb.  Something like Porteus distro, which is 300-something megabytes and as easy to use like windows, and it has the gPartEd visual tool that is straightforward to use.  You'd use the ISO option to copy it over.

encloinc, 2017-06-02:
It wont let me use rufus again because it detects a drive, which doesn't show up as a device in rufus,

ich0x, 2017-06-02:
Help, my USB drive is not detected!
https://github.com/pbatard/rufus/wiki/FAQ#Help_my_USB_drive_is_not_detected
Have you checkmarked List USB hard drives

encloinc, 2017-06-02:
yes, it was one of the first things i did

ich0x, 2017-06-02:
Ok, I understand.  That usb should still boot though.
 I guess I'd try to dd or repartition from a linux distro if windows wouldn't let me from a recovery console.
 I still don't think anything is actually wrong with Rufus, just Windows working as it should to our detriment. 
 I also checked the installation procedure for ElementaryOS and can [clearly see the directions using Rufus](https://elementary.io/docs/installation#install-on-windows) in step 3. stating _Select "ISO Image" next to "Create a bootable disk using..."_

encloinc, 2017-06-02:
Yes I know but since it wasnt working with the plain iso file (I got an error) I tried dd mode

ich0x, 2017-06-02:
Your downloaded iso matched SHA256 hash?
48fcf51244c77e18ff68a8456a2e92e2b49270b9b3a1c7a60a88a9d358272c1e

What was on the usb before? What model USB drive is it, I'd prefer the VendorID, ProductID as well - you can use [USBDevView](http://www.nirsoft.net/utils/usb_devices_view.html) to quickly show you other types of useful USB info.  If you have some other linux installed, a copy of the text output that _lsusb -v_ provides is super-useful.

pbatard, 2017-06-02:
As far as I know, USB flash drives being detected as CD-ROM happens when the controller residing on the flash drive cannot access the memory content due to a hardware failure with the flash memory, and reverting to one of its default controller mode.

Unfortunately, if you tried everything you could to repartition your drive and nothing works, there's not much I can do but advise you to purchase a new flash drive. What I can say however is that Rufus had nothing to do with this issue, as this has no access to the internal USB flash drive controller, and what you are observing is typical of a flash drive where the memory failed.

ich0x, 2017-06-02:
I have a last chance texaco fix for that dying readonly usb, but I need the vendor and product IDs and/or the  hardware id linux assigned to it (in a terminal, sudo lsusb -v) or crack open the plastic and provide the details of the usb controller chip inside - and find it's lowlevel format tool from those russian sites. I recovered a 32gb kingston DT111 g3 into a 27gb Generic **Phison PS2251-02**  that is still writeable to an extent.

encloinc, 2017-06-02:
Well, I used a sd card and my camera as a "Usb". Do you want the sd card model or the camera model (Im using the camera as an adapter basically)

ich0x, 2017-06-03:
Um sd card? I've yet to fix one in this way actually. I give up. Other than checking that the writeprotect tab hadn't budged to lock, or destroying the plastic jacket to get to the chip inside and find something on google to lowlevel format it, let's call it catastropic hardware failure.

encloinc, 2017-06-03:
Well it had a nice life.

JonnyTech, 2017-06-03:
If you were running Linux, I would tell you to zero the drive (in diskpart speak it is the "clean all" command) - maybe @pbatard could add that option (possibly as a cheat code)?

pbatard, 2017-06-03:
> maybe @pbatard could add that option (possibly as a cheat code)?

It's already there (<kbd>Alt</kbd>-<kbd>Z</kbd>), and [documented in the FAQ](https://github.com/pbatard/rufus/wiki/FAQ#Power_keysCheat_modes).
However, if the drive isn't listed, that won't work.

JonnyTech, 2017-06-03:
@pbatard, hehe, sorry, my eyes must need testing 8/

pbatard, 2017-06-03:
As far as Rufus is concerned, I don't think there's anything more that can be done, so I will close this issue.

closed 2017-06-03 by pbatard