pbatard/rufus #965: 2.12 works well but 2.15 have an issue. Can not complete the task.

iMocW, 2017-06-09:
Checklist
---------
- [1 ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [1 ] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [1 ] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [1 ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive
  - [ ] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>

Log
---
```
<FULL LOG>
Rufus version: 2.12.1054 (Portable)
Windows version: Windows 10 64-bit (Build 14393)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0804
Will use default UI locale 0x0804
NOTE: The Chinese Simplified (简体中文) translation requires an update, but the current translator hasn't submitted one. Because of this, some messages will only be displayed in English.
If you think you can help update this translation, please e-mail the author of this application
Found USB 3.0 device 'SanDisk ExtremePro USB Device' (0781:5588)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 15566, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x002D2FDF
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 119.2 GB (128034627584 bytes)
  Start Sector: 2048, Boot: Yes

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE2 for write access
Will use 'G:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Windows 7 Master Boot Record
Drive has a FAT32 FreeDOS Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 2048 sectors
Deleting partitions...
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (Large FAT32)...
Opened drive \\?\Volume{10338917-ff2f-11e5-9bc8-b053fd4f8d6f} for write access
I/O boundary checks disabled
Size : 119.2 GB 250067632 sectors
Cluster size 32768 bytes, 512 Bytes Per Sector
Volume ID is 1ef1:3827
32 Reserved Sectors, 30519 Sectors per FAT, 2 FATs
3906352 Total clusters
3906351 Free Clusters
Clearing out 61134 sectors for reserved sectors, FATs and root cluster...
Initializing reserved sectors and FATs...
FAT #0 sector at address: 32
FAT #1 sector at address: 30551
[... 121 lines cut ...]
FAT #0 sector at address: 32
FAT #1 sector at address: 30551
Writing partition boot record...
Using FreeDOS FAT32 partition boot record
Confirmed new volume has a primary FAT32 boot sector
Setting primary FAT32 boot sector for boot...
Confirmed new volume has a secondary FAT32 boot sector
Setting secondary FAT32 boot sector for boot...
Setting Label (This may take while)...
Format completed.
Writing master boot record...
Drive has a Zeroed Master Boot Record
Partition is already FAT32 LBA...
Set bootable USB partition as 0x80
Using Windows 7 MBR
Found volume GUID \\?\Volume{10338917-ff2f-11e5-9bc8-b053fd4f8d6f}\
Waiting for access on \\?\Volume{10338917-ff2f-11e5-9bc8-b053fd4f8d6f} [\Device\HarddiskVolume9]...
Warning: Could not obtain exclusive rights. Retrying with write sharing enabled...
WARNING: The following process(es) or service(s) are accessing \Device\HarddiskVolume9:
o 'C:\Windows\explorer.exe' (pid: 7612, access: r)

```
2.12 works well.
2.15 is conflict with explorer.exe : (
Windows 10 1607-14393

pbatard, 2017-06-09:
> 2.12 works well.

That's not correct. 2.12 ignored other processes that could alter the drive while Rufus was writing to it, __which could create many issues that were very difficult to track__ (because you certainly do __NOT__ want to have another application with the drive open while it is being repartitionned and reformatted).

As a result, and to prevent these kind of errors, newer versions of Rufus are a lot stricter with regards to sharing the drive with processes that may alter its content, hence what you are observing.

Also, it is possible that the process which is accessing the drive is one that Rufus was not able to report (i.e. it may be something else than explorer), as there are processes, such as Antivirus or other Security Solutions, that prevent other processes such as Rufus from knowing what drives they are opened.

I therefore have 2 questions:
- What happens if you close (or preferably kill) Explorer while Rufus is running?
- Do you have any security solution that could interfere with drive access?

iMocW, 2017-06-10:
1\Then it works well like 2.12, and there's an about 2 seconds time gap between "Waiting for access on..." and "Opened ... for exclusive write access".
2\I do not have security software installed. Only Windows Defender is running.
3\You may lock the handle in the beginning(before format) until all process finished completely?

pbatard, 2017-06-10:
Thanks for the update.

Then I have to ask: Do you have Explorer opened to any folder of your USB drive when you try to use it with Rufus. If not, then there's definitely an environmental component to your issue as nobody else is reporting something like that. For instance, do you have extensions installed that add additional context menus to Explorer? My guess is that it could be an application plugin (such as one used to mount a non Windows-native file system), that ties into Explorer, and that keeps all drives open...

I would also encourage you to take a look at the list of software that is known to create incompatibilities with Rufus, [at the end of the FAQ](https://github.com/pbatard/rufus/wiki/FAQ#Incompatible_Software).

> You may lock the handle in the beginning(before format) until all process finished completely?

That's what I've been doing from the start. But locking a drive isn't enough in Windows as this actually doesn't give you exclusive access, and we need to relinquish the lock at various stages anyway to keep Windows happy. So what we do in newer versions is wait until we can have exclusive access __for write__ (we don't actually care about read, and the fact that Explorer is being reported as problematic, but for read access only is what makes me think that there is something else besides explorer with write access), as, surely, no properly designed application should keep a write handle open to a Windows drive for an extensive period of time (but unfortunately, some do, like the Paragon ext4 or HFS+ drivers), and having exclusive write access that we can keep until we have copied all the files is the only way to ensure that, when Rufus reports to the user that the operation is complete, the drive content is precisely as intended, with no possibility of another application having altered it. 

And due to the exclusive request, the expectation is that, because they were effectively locked out, any other application that wants to access the drive to (possibly) write to it will have detected that its partitioning/file system/content has changed from the last time they opened the drive, __which is not something we can rely on if we don't request exclusive access__, as the other application, with the ability to use a previously opened write handle, __may then try to write data against a completely obsolete layout of the drive__. But of course, forcing exclusive access, as we do in newer versions, will now tend to expose ill-behaved applications (i.e. the ones that keep a write handle opened to a drive all the time), which can then be tricky for users to identify...

iMocW, 2017-06-10:
I see.
Temporarily regardless of this problem until I reproduce the problem next time.
I will re-open this issue when it happens again.
Next time I will analyze it.

closed 2017-06-10 by iMocW