pbatard/rufus #966: Permalink for download portable version.

Elgin675, 2017-06-13:
Hello :) !

Thank's for this great software ;).
I have a little request, not about the software, but for the site.

At _https://rufus.akeo.ie/downloads/_, exist a permalink for the install version, named "**_rufus.exe_**".
Can you add, please, a permalink for download the latest **portable** version (like **rufusp.exe**, for example) :) ?

Thank's ;) !

Xeon3D, 2017-06-14:
Erm, the file is exactly the same. You can just download rufus.exe and rename it rufusp.exe. It autodetects the P in the filename and changes it's mode from portable or non-portable accordingly.

Elgin675, 2017-06-14:
Hem...

I feel very stupid. Thank's for the answer ;).
PS : I hasn't read the FAQ before post my question... Very sorry, it's my fault :s.

closed 2017-06-14 by Elgin675