pbatard/rufus #967: Submit to the Windows Store

Nirmal4G, 2017-06-14:
Refer [Rufus APPX Package](/UWP-Open-Source-Community/Rufus) here.

Here is [Desktop to AppX Guide](https://docs.microsoft.com/en-us/windows/uwp/porting/desktop-to-uwp-run-desktop-app-converter) to convert Win32/.NET apps to AppX Package.

I know _Rufus_ is simple without having install/uninstall problems that occur in some big applications but It is easy to let the store handle the update process than all by yourself. You can also upload the AppX Package to a http server (like you always do), no need to submit to the store if you prefer that way.

_WinRT_ is maturing fast, in a few years it could very well replace _Win32_. It is easier to install directly from the store and benefit from all its features like _Crash Reporting_, _Feedback Hub_, _Analytics_, _Differential Downloads_, etc...

Please consider the idea!
Thanks

pbatard, 2017-06-14:
> Refer Rufus APPX Package here.

No need, I already publish a __signed__ APPX [here](https://rufus.akeo.ie/testing/).

> and benefit from all its features like Crash Reporting, Feedback Hub, Analytics, Differential Downloads, etc...

...as well as the __HUGE RESTRICTIONS THAT ARE IMPOSED BY MICROSOFT__, such as __no elevation__.

If you install the `.appx`, you'll see that, even though it is signed with a valid Authenticode certificate and was created following [Microsoft's actual Appx packaging guidelines](https://github.com/pbatard/rufus/tree/master/res/appstore) it fails with the following:
[image]

The issue is that Windows Apps are made to run in a crippled & restricted environment, that prevents pretty much everything that Rufus requires, such as the ability to have raw access to all drives and USB devices, and Microsoft is showing absolutely no sign to relinquish that policy. This means that, as an `appx`, Rufus is pretty useless as it requires users always right click and select _More_ &rarr; _Run as administrator_ to be able to run it, which makes it __super inconvenient as an App_. And there's no way Microsoft will accept it into the App Store either, since it can't run as non admin.

> Please consider the idea!

I already considered it. Instead, I think what you haven't considered is the other side of the App Store, as this is a great way for Microsoft to restrict people's freedoms, with regards to what they can do with their computer...

closed 2017-06-14 by pbatard

Nirmal4G, 2017-06-14:
You got me with the **NO ELEVATION** issue, I think in the future this will be lifted as some of the Microsoft's own apps like Desktop App Converter and others alike require system privileges.

> ...as well as the HUGE RESTRICTIONS THAT ARE IMPOSED BY MICROSOFT, such as no elevation.

As they mentioned in the BUILD, In the future is that eventually everything you can do in Win32, .NET (WF, WPF) you should be able to do in UWP!

_As they said UWP is solely created to separate Apps and System itself, so that Apps themselves directly/indirectly could not bloat/corrupt the system._

> I think what you haven't considered is the other side of the App Store, as this is a great way for Microsoft to restrict people's freedoms, with regards to what they can do with their computer...

As for the App store it is a way to merge multiple store ends that Microsoft had, Xbox Games, Windows Live Games, MS Store, Xbox Music, Movies and TV, Mobile/Phone, as well as others. **Store - Totally optional**!

**Windows Nirvana** -> **One OS, One UI/UX, One Store, One App Platform dynamically adaptive to every combination of device family, specifications, dev tools**

Nirmal4G, 2017-06-14:
thanks any way for trying, we'll get there SOON™

I think elevation support is coming with Fall Creators Update!

pbatard, 2017-06-14:
> I think in the future this will be lifted as some of the Microsoft's own apps like Desktop App Converter and others alike require system privileges.

I certainly hope so. But it's not often that entities in a position of power will relinquish some of that power to others. My fear is that Microsoft will simply give its own apps a "special right" to do system stuff, and everybody else will be constrained to the limited capabilities we have now. I'd definitely like to be proven wrong though, so I'll be keeping an eye to the Fall update with keen interest.

> In the future is that eventually everything you can do in Win32, .NET (WF, WPF) you should be able to do in UWP!

The problem here is security. Microsoft also wants to use UWP to update on their app security, meaning that they are going to place serious hurdles with regards to breaking out of the restricted environment they want apps to run it. I could very well see them require developer to acquire a super expensive EV Certificate before they can submit elevated apps to the app store (if they ever allow elevated apps), which will be a huge barrier of entry for most developers...

If they __really__ wanted do have UWP do everything one can do in Win32, they'd have enabled elevation from the get go. After all, they have everything they need to revoke an ill-behaved elevated app as soon as it is found to be ill-behaved. Yet they haven't done that, which makes me think that either there is something that Microsoft isn't telling us, or they are testing the waters to see how much they can get away with...

All this to say: I personally wouldn't take a Microsoft promise at face value.

Still, that doesn't mean I have given up on providing Rufus on Windows S ARM64 systems if possible...

Nirmal4G, 2017-06-14:
There are features like **Console Mode** for UWP Apps, **Console apps support system** like registering _app.exe alias_, _PowerShell alias_, **True Multi Windows**, **Multi-User** apps, **3rd party system** components, **3rd party framework packages** and **3rd party extensions for Windows components**, and a lot more!

We'll learn soon enough. I think next two years with RS3 (1709), RS4 (1803), RS5 (1809) will be interesting!

Years ago, they said the same thing with Win32 at PDC, if history is any indication, UWP will become the next Win32, I'm sure of it. It may not be now, we'll just have wait...

imsai0, 2017-07-02:
thanks @pbatard for your consideration...  Like me, many people are eagerly waiting for rufus on the windows store.. As a well known developer, you can ask ms to provide the Elevation support in UWP. Please do something..

pbatard, 2017-07-02:
@imsai0, thanks!

But I would reverse the request: When it comes to Microsoft it's actually the voice of __users__ who is the strongest, and the only way we're gonna have MS relinquish their _"No evelevated Apps in the Windows Store, Ever!"_ is if Windows __users__ start to vehemently complain about it.

I know for a fact that other developers have complained about it, and gone absolutely nowhere. So I would encourage you to engage with Microsoft and let them know that, as an UWP __user__, you find that their Windows Store is too restrictive and let them know!

imsai0, 2017-07-04:
thanks @pbatard for replying. I will let them know that we want it

scratchfive, 2018-08-26:
It seems that Windows Store apps can be set by the user to always run in an elevated mode which is what TreeSize is doing in their app. This just has to be a setting available to the user and at least _something_ has to be displayed to non elevated users as per [this](https://docs.microsoft.com/en-us/windows/uwp/porting/desktop-to-uwp-prepare).

pbatard, 2018-08-26:
Per your link (2nd bullet point):

> Your app always runs with elevated security privileges. Your app needs to work while running as the interactive user. Users who install your app from the Microsoft Store may not be system administrators, so requiring your app to run elevated means that it won't run correctly for standard users. Apps that require elevation for any part of their functionality won't be accepted in the Microsoft Store.

So, I think I'm already doing everything I can by self-publishing the `.appx` [here](https://rufus.ie/testing/) and letting people know that if they install it they will have to run it as admin.

The _Apps that require elevation for any part of their functionality won't be accepted in the Microsoft Store_ part of the Microsoft documentation is pretty explicit on that subject, and doesn't leave much to interpretation.

Rufus __does__ need elevation to repartition a drive and as such, if Microsoft does follow the guidelines it publishes, it will not be accepted in the Microsoft Store.