pbatard/rufus #968: Rufus is very very slow

mariobalanica, 2017-06-17:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [ ] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive
  - [x] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [x] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Everytime I try to create a Windows 10 bootable USB using Rufus I have to wait a very long time.
My USB drive is a Kingston DataTraveler 100 G3 8GB and I have an ASUS K53SV laptop.
I tried using a different USB stick and the problem persists. 
[image]

Log
---
```
<FULL LOG>
```
No errors.

pbatard, 2017-06-17:
No log &rarr; closed. The log would have indicated what speed your USB is actually using. Anyway, this is a hardware issue. Using a decent USB 3.0 flash drive on an USB 3.0 port, you can create a Windows bootable drive in less than 3 mins. If that is not your case, then your hardware or environment is the porblem.

closed 2017-06-17 by pbatard

mariobalanica, 2017-06-17:
Ok, i have edited the post.
It finished in 48 minutes.

terminalstderr, 2018-04-02:
I see similar poor performance using Kingston DataTraveler 3.0 USB Device (16GB). I've been copying ISO from an SSD to this thumb drive for 20 minutes so far and Rufus reports approx 20% complete. The performance of this device is abysmal.

http://usb.userbenchmark.com/Compare/Kingston-DataTraveler-100-G3-USB-30-16GB-vs-Group-/1521vs10

> With a sequential write speed of 9.9 MB/s and a random write of 0.046 MB/s This drive is actually slower at writing than a lot of USB 2.0 drives

pbatard, 2018-07-27:
There's no difference in the manner Rufus writes files between GPT and MBR. Once the partitions have been created, it's exactly the same. And nobody else is reporting this issue, so I'm afraid that your assumption that using MBR rather than GPT when creating a Windows 10 bootable USB is going to be slower is wrong, and that your problem has to come from your environment.

If this is a GPT vs MBR issue, then surely you should be able to demonstrate that you have the same issue when using a different USB Flash Drive as well as a different computer. Unless you can do that, I'm afraid you will have to look elsewhere for the reason why your flash drive creation is so slow. Does your log say anything about that?

vnluc, 2018-07-27:
@pbatard , thank you for your response. I don't know yet the part that write the files how it work. Now it has running for 2 hour 24min and not yet finish. 

As per what you comment maybe I will test using same Laptop / PC and writing to MBR format to see how it differ. I will share log of GPT writing once it finish. Also this log not having time stamp at each entry. If it has time on each file write it would be better.

pbatard, 2018-07-27:
> If it has time on each file write it would be better.

Rufus is not a benchmarking tool, so that's not going to happen.

lawson2705, 2018-08-08:
I also have some problems. Well I don't know if it's a problem or it's normal. I am attempting to burn Parrot OS ( Linux ) To my USB. Here is the log so far. 40 mins in. A long way from complete. 20% in perhaps...

Rufus version: 3.1.1320 (Portable)
Windows version: Windows 7 SP1 32 bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0409
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'Kingston DataTraveler 3.0 USB Device' (0951:1666)
1 device found
Disk type: Removable, Disk size: 7.9GB, Sector size: 512 bytes
Cylinders: 956, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or Unpartitioned
Partition 0:
  Type: FAT32 (0x00)
  Size: 7.3 GB (7864320000 bytes)
  Start Sector: 0, Boot: No
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.03/20171017 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'ParrotSec'
  Size: 3.6 GB (Projected)
  Has a >64 chars filename
  Uses: Syslinux/Isolinux v6.03
  Uses: EFI
  Note: This ISO uses symbolic links, which will not be replicated due to file system limitations.
  Because of this, some features from this image may not work...
Using image: Parrot-security-4.1_i386.iso (3.6 GB)
Will reuse 'ldlinux.sys' and 'ldlinux.bss' from './rufus_files/rufus_files/syslinux-6.03/20171017/' for Syslinux installation

Format operation started
Requesting disk access...
Opened \\.\PHYSICALDRIVE1 for exclusive write access
Requesting lock...
Will use 'E:' as volume mountpoint
I/O boundary checks disabled
Requesting lock...
Analyzing existing boot records...
Drive has an unknown Master Boot Record
Volume has an unknown Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (FAT32)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Writing master boot record...
Drive has a Zeroed Master Boot Record
Set bootable USB partition as 0x80
Using Syslinux MBR
Found volume GUID \\?\Volume{8f7212bf-9a37-11e8-b45f-00262243f6ba}\
Installing Syslinux 6.03...
Opened \\?\Volume{8f7212bf-9a37-11e8-b45f-00262243f6ba} for shared write access
Using existing './rufus_files/syslinux-6.03/20171017/ldlinux.sys' ✓
Using existing './rufus_files/syslinux-6.03/20171017/ldlinux.bss' ✓
Successfully wrote 'ldlinux.sys'
Successfully wrote Syslinux boot record
Successfully remounted \\?\Volume{8f7212bf-9a37-11e8-b45f-00262243f6ba}\ on E:
[... 547 lines cut ...]
Extracting: E:\install\vmlinuz (3.8 MB)
Extracting: E:\isolinux\advanced.cfg (214 bytes)
Extracting: E:\isolinux\boot.cat (2 KB)
Extracting: E:\isolinux\hdt.c32 (161.1 KB)
Extracting: E:\isolinux\install.cfg (679 bytes)
Extracting: E:\isolinux\isolinux.bin (38 KB)
Extracting: E:\isolinux\isolinux.cfg (59 bytes)
Extracting: E:\isolinux\ldlinux.c32 (113.1 KB)
Extracting: E:\isolinux\libcom32.c32 (176.3 KB)
Extracting: E:\isolinux\libgpl.c32 (64.8 KB)
Extracting: E:\isolinux\libmenu.c32 (23.4 KB)
Extracting: E:\isolinux\libutil.c32 (22.5 KB)
Extracting: E:\isolinux\live.cfg (1.9 KB)
Extracting: E:\isolinux\live.de.cfg (2.2 KB)
Extracting: E:\isolinux\live.es.cfg (2.4 KB)
Extracting: E:\isolinux\live.fr.cfg (2.3 KB)
Extracting: E:\isolinux\live.it.cfg (2.3 KB)
Extracting: E:\isolinux\menu.cfg (591 bytes)
Extracting: E:\isolinux\splash.png (36.9 KB)
Extracting: E:\isolinux\stdmenu.cfg (484 bytes)
Extracting: E:\isolinux\vesamenu.c32 (26.1 KB)
Extracting: E:\live\filesystem.packages (92.5 KB)
Extracting: E:\live\filesystem.packages-remove (1 bytes)
Extracting: E:\live\filesystem.size (12 bytes)
Extracting: E:\live\filesystem.squashfs (3.3 GB)

pbatard, 2018-08-08:
Again, whatever slowdown you are experiencing has nothing to do with Rufus. Rufus is always writing data at the maximum speed it can, and if it had anything to do with the application, then everybody would experience slow speed, which isn't the case.

There is absolutely nothing in the code that even remotely could affect the write speed between one device or another, especially during the file copy phase, which is exceedingly straightforward.

In other words, whatever slowdown you are experiencing comes either from the hardware you use or your environment, and there is no "magic switch", that Rufus could activate in the code, that will suddenly make your drive go faster.

vnluc, 2018-08-08:
Back to my issue, it was slow on Windows 7. When I change to new Laptop that is clean install, the write operation is faster and finish in 10min. So I think slow may be because of Antivirus software or the speed of USB port.

hrvylein, 2018-11-29:
Sorry to reopen but I am encountering this issue too. On my new laptop the Windows ISO from MSDN took around 9 minutes to be copied to the Usb 2.0 Stick with GPT/UEFI settings. On my gaming rag it wouldn't copy 300mb after 50 minutes.
Microsoft media creation, Unetbootin, ISO2USB took the same amount whereas Rufus wasn't able to complete the task on the same machine.

Maybe there is sth. going on with legacy->uefi->legacy conversion going on?

pbatard, 2018-11-29:
Try to disable Defender or whatever Antivirus you are using. Defender has been especially hit and miss lately with regards to slowing down file copy operations. And it seems that, depending of the application being used, it will be more or less active. For instance I've seen a massive speed bump when creating a Windows To Go drive for ARM after disabling Defender.

Obviously, there's not much I can do in Rufus if whatever AntiVirus is active has decided it wants to do a thorough scan of whatever is being copied over to a flash drive. I would therefore encourage you to:
- Go through a full power cycle (full shutdown and then power up)
- Disable the Antivirus
- Try the same operation again
To see if you are still seeing slow speeds.

Also, if you want decent speeds, you certainly should not be using USB 2.0 devices in the first place, but instead switch to USB 3.0 ones with a decent write speed (of course plugged to a USB 3.0 port).

denniske, 2019-03-18:
For me it took 20 minutes for copying files to the USB device. I accept that it is due to the the USB device being slow.

What I would suggest is having a more fine grained progress indication at least when copying the large files. (like copying abc.data 45% in the status bar).

The problem in my case was that I read that rufus takes only 4 minutes. So when after 5 minutes the main progress bar did not move for two minutes I assumed that the copying had failed somehow and restarted rufus. This could be avoided.

pbatard, 2019-03-18:
> What I would suggest is having a more fine grained progress indication at least when copying the large files. (like copying abc.data 45% in the status bar).

The problem with that is that we can't have this kind of fine grained data unless we drop the memory caching that Windows uses, as, when you are copying a large file, Windows will usually copy as much as it can to memory (e.g. 1 GB) and report that this amount of data has been copied over, and then, over the next few tens of second actually copy that data to the USB while it makes the process that launch the copy wait. That's why you may have noticed that, when Rufus copies something like Windows's `install.wim` to USB, it seems to go very fast at the beginning and then get stuck for a while, before the progress bar starts to move again.

Unless we completely disable the cache (which would make all operations much slower), what you are asking is not something.

You really have to be mindful that, nowadays, loads of copy operations are cached, which is really what you want when moving data to a slow peripheral like a USB drive, which completely removes the ability to provide the kind of __accurate__ copy reports people think the OS can provide, unless you completely disable the mechanisms that are there to improve copy speed. So, no, I'm afraid accurate copy reports cannot be provided unless you either:
1. Make everything __much slower__ by disabling caching
2. Add some complex snooping of what Windows is actually doing with its cache, which of course Microsoft doesn't document and which trying to reverse engineer doesn't strike me as that good a use of a developer's time...

Also note that what I am talking about applies to ISO file copy mode, where Windows caching can be extremely aggressive, as opposed to DD mode, where progress is usually a bit more accurate.

thearjuna, 2019-04-05:
Same issue here. Rufus take 10-15 mins to write my iso on Windows 10. Same operation takes 3 minutes in both Etcher and Imagewriter.

pbatard, 2019-04-05:
> Rufus take 10-15 mins to write my iso on Windows 10. Same operation takes 3 minutes in both Etcher and Imagewriter.

Well, considering that, if you are using an __offcial__ Microsoft Windows 10 ISO, Etcher and Imagewriter will he hopeless to write them (Etcher and Imagewriter only work with ISOHybrid images in DD mode, and Windows ISO are __NOT__ ISO Hybrid), I think you may have to elaborate on that statement.

Where exactly did you obtain these ISOs? Were they official (clearly they weren't)? And, if you happened to come across an __UNOFFICIAL__ Windows 10 ISOHybrid image, what mode did you use to write them in Rufus: DD or ISO?

bad-jesus, 2020-02-11:
I just want to try and help, and mention I am having the same issue. 
The ISO I am using at the moment is opnsense_VGA_64bit
Rufus took over a full day to write, and it was corrupt. (I redownloaded the image a few times)

Using Win32DiskImager, USBImageTool and a few others, it worked. 

This image specifically overwrites the settings in Rufus to GPT, but it needs to be MBR to work (I don't think this is a Rufus issue, this is an image issue, but I am mentioning in case Rufus can overwrite these settings? I realize it is probably not that simple so I am merely mentioning it)

I tried about 6 other image writers, all wrote to the same USB drive in seconds. 

Rufus, was days.

None of the images worked at first,  because they all wrote GPT but I used PartMan to change to MBR and it worked. 

In regards to speed only, Rufus was the only option tried that took more than seconds to write, and it was over a day. Sorry, I didn't keep the log but I had the log open and I remember correctly it just sat with the last line of something like 

"Writing Compressed Image....."

I don't remember it adding any more detail to the log for the entire day.

pbatard, 2020-02-11:
@bad-jesus, at the very least you should provide the name of the image you were writing.

The fact that you were writing a compressed image means it's a completely different issue from OP, since you were writing in DD mode and he was writing in ISO mode.

Recently (these will be available in Rufus 3.9) some improvements were added to speed up the decompression speed, but one thing I will point out is that "on the fly" decompression of compressed images is provided as a courtesy feature as there are lots of compression formats, and using the fastest library for all of these would both require a lot of development time as well as greatly increase the size of the application. So the decompression library we use is geared towards simplicity and small size rather than performance, which means that I'm not entirely surprise that it may be slow for some specific cases. But of course, without knowing precisely what compressed image you use, it's really hard to provide anything but speculation on this matter...

bad-jesus, 2020-02-11:
> @bad-jesus, at the very least you should provide the name of the image you were writing.
>
> The fact that you were writing a compressed image means it's a completely different issue from OP, since you [...]

Lol, just trying to help man. I did give you the name of the image. The file extension was : .img.bz2 as are every image format for opnsense. 
I am not a pro with this, I gave you the info I had, and tried to explain the best I could. You get more flies with honey, man. 

Reading through this thread and others, you mostly seem to just deny any issues on here as user error, so I decided to comment to help show that there are a lot of people having this issue. The OP's error seemed identical to mine. If not, you can Google my specific use case, you will see dozens of threads with dozens of people experiencing the same issue. There are dozens of posts on Reddit, as well as the OpnSense and PFSense groups, alone. 

If it is just the image, and not Rufus, cool. Anecdotally, the problem lies with Rufus as NO OTHER program had this same issue with speed. Every other program I tried wrote in SECONDS not DAYS using the same image file. 

I'm not looking for support, I am simply trying to give feedback to help. I got my image to write with one of the other programs and then I changed it to MBR after the fact. If my information isn't helpful, cool. Just ignore it, I am just trying to help. I could have simply not given any information, but I went out of my way to provide what I hoped was helpful information. 

The way the image loads, I can't even select DD mode. All features are greyed out. So there aren't a lot of things I can give you for information aside from what has already been provided.

pbatard, 2020-02-11:
I was asking for the complete name of the image, not bits and pieces that I need to piece together. And you certainly did __NOT__ give me the extension of your image until the message above.

Your expectation is that developers should be grateful for incomplete data, and just fill the blank themselves (_"It's easy, just go to the PFSense website and you'll figure it out!"_) but that's not how it works. You are busy and so am I. I do monitor posts on Reddit (If you do, you'll see that I'm trying to help people there on [very regular basis](https://old.reddit.com/user/_Akeo_/)) and while I am aware of some PFsense users seemingly having issues booting, I don't recall PFsense users specifically complaining about Rufus writing speed. Or maybe we [should try a google search then](https://www.google.com/search?client=firefox-b-d&q=rufus+speed+pfsense)?

But the nice thing is, you don't have to take it to my word: You can do the [search yourself](https://old.reddit.com/search?q=rufus+speed+pfsense&restrict_sr=&sort=new&t=all).

So, yes, I feel I am entitled to indicate that:
- No, OP error is not the same as yours.
- No, you didn't provide enough data for me to properly investigate it
- No you are not helping your case by adding claims that can easily be debunked.

Now, if you are offended by someone simply telling you that you could have done a better job reporting the issue, in the hope that this kind of information might be helpful to you for future reports, and want to erroneously reduce it down to these "damn entitled developers being all arrogant and denying user claims" so be it.

NekoDm, 2020-03-10:
Ok I have the same issues too, from what I found is, it was the rufus it self, on MBR partition rufus doing fine from copying the boost file from ISO around 5-6 minute in usb 2.0 and 2-3 minute in usb 3.0, and the strange thing is,when switching to GTP partition, rufus struggling so badly when trying to copy the boost file on ISO to usb for some reason, yes I even try different usb stick, and other machine with clean windows installed, and the caught I found is it was the "file system" fault , you just need to change the system file type from FAT32 to NTFS and cluster size is default 4092 bytes and proplem will be gone LOL

pbatard, 2020-03-10:
Most likely this is a block/sector alignment issue and a problem with the internal design of the drive being used.

For instance, say the drive is designed internally (at the flash level) to work with 4KB blocks but you picked a cluster size of 1KB. What happens then is that if you have a series of 1KB files to be written, and the flash drive controller is poorly designed, you're going to end up with the controller issuing a lot of rewrites to the same blocks of data, which is slow, whereas if you use a cluster size that is larger than the internal flash block size, then you'll be able to queue read and writes in parallel (since they don't access the same blocks of flash).

The end result is that you're going to lose more free space (3 KB for each 1 KB file) but writes are going to be much faster.

Note that this is a very simplified view of what __might__ happen as there are plenty of other factors that can have an impact (such as the type of flash memory, the presence of an internal cache, the presence of defective sectors, which the controller will try to work around transparently but which may incur delays, etc.). And of course the partition's alignment plays a part too, though, as long as you don't use _"Fixes for old BIOSes"_ this shouldn't be an issue.

Most of the time, Windows (and therefore Rufus) tries to queue sequential writes so that the controller from your drive should be able to optimize writes for speed. But of course that only works if the controller is smart enough and the internal characteristics of the drive can favour such writes. In some cases, through no fault of the application, you may run afoul of internal limitations and experience very slow write speeds, in which case you may indeed want to switch file systems or cluster size, since Rufus or Windows have no insight whatsoever as to what might work best (so Rufus cannot really select the one that will be faster).

zvonka, 2020-07-08:
Sorry to reopen this but I believe i have found something that may be useful to anyone experiencing these slowdowns, which I was myself seeing as well yesterday with a number of different flash drives, burning win7 as GPT onto a 2.0  drive, taking 2-3 hours per write.

Process hacker application showed that rufus was burning at 1.5mbps roughly, and right clicking it and going to I/O priority it was set to Low, and the process priority was set to low as well, not sure if by design or by windows settings**, but changing both process priority and I/O priority to high brought the write speeds up to about 120mbps on a 16gb transcend USB2.0 drive**

zvonka, 2020-07-08:
One last addition, changing file systems or cluster sizes is not an option for certain types of installations so I dont see this as being a rufus issue instead either, something within windows that organizes the way data is fed into it to go to the drive because rufus was clearly being held back in my case by windows I/O scheduling implementation

pbatard, 2020-07-09:
I agree that I/O simply doesn't get slow on its own, i.e. that if the system is otherwise idle, you're not going to see a jump in I/O throughput simply by increasing a process' priority. In other words, unless something else is actually starving the system, Windows will always write to disk as fast as it can, for a processes that is using normal priority. Especially, if nothing else is accessing the disk, you're not going to see a noticeable jump in I/O bandwidth simply because you changed a process priority from normal to high or real-time. If you don't want to take my word for it, this can easily be verified by picking a process with a high level of disk access _on an otherwise idle system_, and altering its priority level to see if it improves its throughput.

So priority doesn't really matter for most of Rufus' usage scenarios, where there shouldn't be much of anything else taxing the system I/O. And that is the reason why I haven't really bothered fiddling with priority so far...

This being said, there can of course be situations where you are trying to run Rufus concurrently to other I/O intensive processes, in which case you may want a little more "oompf".

So what I'm planning to do for the next version is set the formatting and apply WIM image threads (these are the threads that are created after you press START and that handle the bulk of disk I/O) with an "above normal" priority by default, as it may help read and write operations get a little more prioritized by Windows.

I'm also thinking that I'll probably add a persistent cheat mode (probably <kbd>Alt</kbd>-<kbd>+</kbd>/<kbd>Alt</kbd>-<kbd>-</kbd>) allowing users to increase or decrease the default priority for these thread. I am not sure if I will enable doing that in real-time though (meaning that the default priority level will only be alterable when Rufus is idle) as this presents some challenges, especially for the apply WIM image thread...

KillCore-0, 2020-07-21:
Same issue. I opened Task Manager and Rufus is barely writing anything. It's writing about 3MB every 20 seconds. I am setting up a usb drive with Raspberry Pi OS and the their official Imager tool is 10 times faster!

pbatard, 2020-07-22:
@KillCore-0, I'm going to need to see a log of how you create the drive, as, once again, my main issue is that I can't replicate these findings, and, because of this, I just can't act on a report of "Rufus is slow!" that doesn't include a comprehensive log of how exactly you created your drive in Rufus.

HwangLiu, 2021-03-19:
Looks like it is Kingston Data Traveler issue, i have 2 of those (each 32 gb) and both are writing the ISO with incredibly slow speeds. Tried both USB devices on different PCs. Came back home, hoping it is done, but it is 8 hours into copying and only 43% complete.

dimkir, 2021-05-26:
Kingston Data Traveller here too 64GB. However I think(unfortunately I didn't record first run time) my first Rufus run was fast but subsequent have been very slow. About 60minutes to prepare(format) the disk and then about 10 minutes  to copy Ubuntu files onto it (i don't have problem with copying speed, but formatting speed was very slow).

I do indeed believe that there's something with my environment, but the question is : what is it? or how to determine what it is? 

[image]
 
```
Rufus x86 v3.14.1788
Windows version: Windows 10 64-bit (Build 19042.985)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0809 (en-GB)
Will use default UI locale 0x0809
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'Kingston DataTraveler 3.0 USB Device' (0951:1666)
NOTE: This device is a USB 3.0 device operating at lower speed...
No logical drive found (unpartitioned?)
Found USB 2.0 device 'WD Elements 1023 USB Device' (1058:1023)
Device eliminated because it was detected as a Hard Drive (score 13 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
Found UAS (USB 3.0) device 'External USB3.0 SCSI Disk Device' (152D:0578)
Device eliminated because it was detected as a Hard Drive (score 23 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
Found UAS (USB 3.0) device 'External USB3.0 SCSI Disk Device' (152D:0578)
Device eliminated because it was detected as a Hard Drive (score 13 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
1 device found
No volume information for drive 0x85
Disk type: Removable, Disk size: 64 GB, Sector size: 512 bytes
Cylinders: 7544, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x1E39497B
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 2.8 GB (3002098176 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2:
  Type: GNU/Linux (0x83)
  Size: 55.0 GB (59055768576 bytes)
  Start Sector: 5865521, Boot: No
Closed Windows format prompt
Found USB 2.0 device 'Kingston DataTraveler 3.0 USB Device' (0951:1666)
NOTE: This device is a USB 3.0 device operating at lower speed...
Found USB 2.0 device 'WD Elements 1023 USB Device' (1058:1023)
Device eliminated because it was detected as a Hard Drive (score 13 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
Found UAS (USB 3.0) device 'External USB3.0 SCSI Disk Device' (152D:0578)
Device eliminated because it was detected as a Hard Drive (score 23 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
Found UAS (USB 3.0) device 'External USB3.0 SCSI Disk Device' (152D:0578)
Device eliminated because it was detected as a Hard Drive (score 13 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
1 device found
No volume information for drive 0x85
Disk type: Removable, Disk size: 64 GB, Sector size: 512 bytes
Cylinders: 7544, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x1E39497B
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 2.8 GB (3002098176 bytes)
[... 151 lines cut ...]
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
Found UAS (USB 3.0) device 'External USB3.0 SCSI Disk Device' (152D:0578)
Device eliminated because it was detected as a Hard Drive (score 23 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
Found UAS (USB 3.0) device 'External USB3.0 SCSI Disk Device' (152D:0578)
Device eliminated because it was detected as a Hard Drive (score 13 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
1 device found
Disk type: Removable, Disk size: 64 GB, Sector size: 512 bytes
Cylinders: 7544, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x1E717F12
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 2.8 GB (3002098176 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2:
  Type: GNU/Linux (0x83)
  Size: 55.0 GB (59055768576 bytes)
  Start Sector: 5865521, Boot: No

```

pbatard, 2021-05-26:
@dimkir, here is your problem:

```
Found USB 2.0 device 'Kingston DataTraveler 3.0 USB Device' (0951:1666)
NOTE: This device is a USB 3.0 device operating at lower speed...
```

This means that, for whatever reason (and this has nothing to do with Rufus, which has no control whatsoever on port speeds &mdash; port speed is set entirely by Windows and the hardware), your USB 3.0 device is actually running as USB 2.0 speeds, so, of course, it is going to be slow.

You are going to have to figure out what it is on your system that makes USB 3.0 devices be handled by Windows as USB 2.0 ones, because, until you can get Windows to set your device to use full USB 3.0 speeds, Rufus, and every other application accessing your device, will always be slow to read or write to it.

And again, this is __not__ something Rufus can fix for you. It's a pure Windows/hardware issue.