pbatard/rufus #969: Rufus was denied inside the log. :(

DylanAD, 2017-06-19:


DylanAD, 2017-06-19:
I'm new to Rufus, so I'll put the troubles down here
Rufus version: 2.15.1117
Windows version: Windows 10 64-bit (Build 14393)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0409
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'PNY USB 2.0 FD USB Device' (154B:007A)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 992, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x20694E02
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 7.6 GB (8165261312 bytes)
  Start Sector: 2048, Boot: Yes
Checking for Rufus updates...
Checking release channel...
No new release version found.

Format operation started
Requesting disk access...
Opened \\.\PHYSICALDRIVE1 for exclusive write access
Requesting lock...
Will use 'E:' as volume mountpoint
Waiting for access on \\?\Volume{ce90317a-4bb1-11e7-8a8d-4ccc6a08bd78} [\Device\HarddiskVolume4]...
Warning: Could not obtain exclusive rights. Retrying with write sharing enabled...
WARNING: The following process(es) or service(s) are accessing \Device\HarddiskVolume4:
o '\\?\Volume{ce90317a-4bb1-11e7-8a8d-4ccc6a08bd78}\setup.exe' (pid: 15352, access: x)
o '\\?\Volume{ce90317a-4bb1-11e7-8a8d-4ccc6a08bd78}\sources\setupprep.exe' (pid: 12724, access: rx)
You should try to close these applications before attempting to reformat the drive.
I/O boundary checks disabled
Requesting lock...
Could not lock access to \\?\Volume{ce90317a-4bb1-11e7-8a8d-4ccc6a08bd78}: [0x00000005] Access is denied.
Could not lock volume
Re-mounted volume as 'E:' after error

Found USB 2.0 device 'PNY USB 2.0 FD USB Device' (154B:007A)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 992, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x20694E02
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 7.6 GB (8165261312 bytes)
  Start Sector: 2048, Boot: Yes
Found USB 2.0 device 'PNY USB 2.0 FD USB Device' (154B:007A)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 992, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x20694E02
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 7.6 GB (8165261312 bytes)
  Start Sector: 2048, Boot: Yes

Format operation started
Requesting disk access...
Opened \\.\PHYSICALDRIVE1 for exclusive write access
Requesting lock...
Will use 'E:' as volume mountpoint
Waiting for access on \\?\Volume{ce90317a-4bb1-11e7-8a8d-4ccc6a08bd78} [\Device\HarddiskVolume4]...
Warning: Could not obtain exclusive rights. Retrying with write sharing enabled...
WARNING: The following process(es) or service(s) are accessing \Device\HarddiskVolume4:
o '\\?\Volume{ce90317a-4bb1-11e7-8a8d-4ccc6a08bd78}\setup.exe' (pid: 15352, access: x)
o '\\?\Volume{ce90317a-4bb1-11e7-8a8d-4ccc6a08bd78}\sources\setupprep.exe' (pid: 12724, access: rx)
You should try to close these applications before attempting to reformat the drive.
I/O boundary checks disabled
Requesting lock...
Could not lock access to \\?\Volume{ce90317a-4bb1-11e7-8a8d-4ccc6a08bd78}: [0x00000005] Access is denied.
Could not lock volume
Re-mounted volume as 'E:' after error

Found USB 2.0 device 'PNY USB 2.0 FD USB Device' (154B:007A)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 992, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x20694E02
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 7.6 GB (8165261312 bytes)
  Start Sector: 2048, Boot: Yes

Format operation started
Requesting disk access...
Opened \\.\PHYSICALDRIVE1 for exclusive write access
Requesting lock...
Will use 'E:' as volume mountpoint
Waiting for access on \\?\Volume{ce90317a-4bb1-11e7-8a8d-4ccc6a08bd78} [\Device\HarddiskVolume4]...
Warning: Could not obtain exclusive rights. Retrying with write sharing enabled...
WARNING: The following process(es) or service(s) are accessing \Device\HarddiskVolume4:
o '\\?\Volume{ce90317a-4bb1-11e7-8a8d-4ccc6a08bd78}\setup.exe' (pid: 15352, access: x)
o '\\?\Volume{ce90317a-4bb1-11e7-8a8d-4ccc6a08bd78}\sources\setupprep.exe' (pid: 12724, access: rx)
You should try to close these applications before attempting to reformat the drive.
I/O boundary checks disabled
Requesting lock...
Could not lock access to \\?\Volume{ce90317a-4bb1-11e7-8a8d-4ccc6a08bd78}: [0x00000005] Access is denied.
Could not lock volume
Re-mounted volume as 'E:' after error

Found USB 2.0 device 'PNY USB 2.0 FD USB Device' (154B:007A)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 992, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x20694E02
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 7.6 GB (8165261312 bytes)
  Start Sector: 2048, Boot: Yes

pbatard, 2017-06-19:
> I'm new to Rufus

That doesn't prevent you from following the checklist you know...

Anyway, as per the log, it looks like the following applications are accessing your drive:
```
WARNING: The following process(es) or service(s) are accessing \Device\HarddiskVolume4:
o '\\?\Volume{ce90317a-4bb1-11e7-8a8d-4ccc6a08bd78}\setup.exe' (pid: 15352, access: x)
o '\\?\Volume{ce90317a-4bb1-11e7-8a8d-4ccc6a08bd78}\sources\setupprep.exe' (pid: 12724, access: rx)
You should try to close these applications before attempting to reformat the drive.
```

You'll have to figure out what they are and close them.

pbatard, 2017-06-19:
Ah, sorry, it looks like you followed the checklist but forgot to close a tag so that it would display. Let me see if I can fix it.

pbatard, 2017-06-19:
For the record, you can download something like [Process Hacker](https://wj32.org/processhacker/nightly.php) and try to kill the processes with the relevant pid (`15352`, `12724`). Be mindful however that killing processes can have unintended consequences.

If I were to venture a guess, it would look to me like you might have started to run `setup.exe` and `sources\setupprep.exe` from the USB you are trying to format. Of course, if you start a process from the USB and then attempt to format it, it will not work!

pbatard, 2017-06-21:
I'm going to assume that OP realized that he shouldn't have tried to launch the `setup.exe` that resided on the USB before trying to format it with Rufus and close this issue.

closed 2017-06-21 by pbatard