pbatard/rufus #970: Progress bar frozen while Creating Windows On-The-Go - does it work nonetheless?

frawag, 2017-06-19:
Checklist
---------
- [X] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [X] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [X] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [X] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [X] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [ ] I also tried one or more of the following:
  - [X] Using a different USB drive
  - [X] Plugging the USB into a different port
  - [X] Running Rufus on a different computer
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I wanted to use Rufus to make a Windows-To-Go installation on an external drive. This started as expected, unfortunately the progress froze at 3.2% mentioning "Applying Windows Image" in the log. I waited for 1.5 hours, but nothing happened.

Tried to close Rufus, but couldn't restart it, but got the message "Rufus already running" (something similar) - which I could also see in the Task Manager. Rufus couldn't be closed from there, with permission denied message. Only a reboot would allow me to run Rufus again.

Was the transfer continuing in the background and only the progress bar froze? Or did the transfer stop? In the Task Manager I could not see traffic to the destination drive any longer.

Many thanks!

Log
---
```
Rufus version: 2.15.1117 (Portable)
Windows version: Windows 10 64-bit (Build 15063)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0409
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 3.0 device 'Samsung SSD 850 EVO 500G USB Device' (174C:55AA)
Device eliminated because it was detected as a Hard Drive (score 11 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
0 devices found
Found USB 3.0 device 'Samsung SSD 850 EVO 500G USB Device' (174C:55AA)
1 device found
Disk type: FIXED, Sector Size: 512 bytes
Cylinders: 60801, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: GPT, NB Partitions: 3
Disk GUID: {7F3D5275-DC45-4F08-85A9-5BAA106D6414}
Max parts: 128, Start Offset: 17408, Usable = 500107827712 bytes
Partition 1:
  Type: {E3C9E316-0B5C-4DB8-817D-F92DF00215AE}
  Name: 'Microsoft reserved partition'
  ID: {BB8CA6DE-655C-4384-82E1-6CC064B98D58}
  Size: 128 MB (134217728 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {E266EB16-69BA-4E6E-9432-99D4E4F6D0A5}
  Size: 465.5 GB (499867683840 bytes)
  Start Sector: 264192, Attributes: 0x0000000000000000
Partition 3:
  Type: {C12A7328-F81F-11D2-BA4B-00A0C93EC93B}
  Name: 'EFI system partition'
  ID: {FFF267DD-AB9C-4D5E-97D6-8C4D5C35D807}
  Size: 100 MB (104864256 bytes)
  Start Sector: 976568262, Attributes: 0x0000000000000000
1 device found
Disk type: FIXED, Sector Size: 512 bytes
Cylinders: 60801, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: GPT, NB Partitions: 3
Disk GUID: {7F3D5275-DC45-4F08-85A9-5BAA106D6414}
Max parts: 128, Start Offset: 17408, Usable = 500107827712 bytes
Partition 1:
  Type: {E3C9E316-0B5C-4DB8-817D-F92DF00215AE}
  Name: 'Microsoft reserved partition'
  ID: {BB8CA6DE-655C-4384-82E1-6CC064B98D58}
  Size: 128 MB (134217728 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {E266EB16-69BA-4E6E-9432-99D4E4F6D0A5}
  Size: 465.5 GB (499867683840 bytes)
  Start Sector: 264192, Attributes: 0x0000000000000000
Partition 3:
  Type: {C12A7328-F81F-11D2-BA4B-00A0C93EC93B}
  Name: 'EFI system partition'
  ID: {FFF267DD-AB9C-4D5E-97D6-8C4D5C35D807}
  Size: 100 MB (104864256 bytes)
  Start Sector: 976568262, Attributes: 0x0000000000000000
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'CENA_X64FREV_EN-US_DV5'
  Size: 4.0 GB (Projected)
  Uses: EFI
  Uses: Bootmgr
  Uses: Install.wim (version 0.13.1)
Using image: en_windows_10_enterprise_version_1703_updated_march_2017_x64_dvd_10189290.iso
WIM extraction method(s) supported: wimgapi.dll
WIM apply method supported: wimgapi.dll
Opening: \\.\CDROM1\sources\install.wim:[0] (API)
Extracting: C:\Users\Frank\AppData\Local\Temp\RufD2B.tmp (From [1].xml)
Closing: \\.\CDROM1\sources\install.wim
Will use 'Windows 10 Enterprise' (Build: 15063, Index 1) for Windows To Go

Format operation started
Requesting disk access...
Opened \\.\PHYSICALDRIVE1 for exclusive write access
Requesting lock...
Will use 'F:' as volume mountpoint
I/O boundary checks disabled
Requesting lock...
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Volume has an unknown Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (GPT)...
Adding MSR partition
Adding extra partition
Reserved 3251 tracks (100 MB) for extra partition
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Found volume GUID \\?\Volume{7482b5ce-1c22-47ad-a7c2-da02b2fc7863}\
\\?\Volume{7482b5ce-1c22-47ad-a7c2-da02b2fc7863}\ was already mounted as F:\
Successfully remounted \\?\Volume{7482b5ce-1c22-47ad-a7c2-da02b2fc7863}\ on F:
Applying Windows image...
Windows To Go mode selected
Mounted ISO as '\\.\CDROM1'
Opening: \\.\CDROM1\sources\install.wim:[1]
Applying Windows image...
```

pbatard, 2017-06-19:
Hmm, you're using an USB 3.0 device which should have a decent enough speed (though a non negligible part of it has to do with the Windows image application process, which is tied to how fast your machine is), so it would look to me like you encountered a hardware issue during the process, which froze it.

Unfortunately, Rufus needs to use the Windows WIM extraction API to create the Windows To Go drive, which are pretty dreadful at reporting proper speed or errors during the process, so chances are that, as that API froze, it never returned an error to Rufus. And trying to stop the process through the WIM API is also touch and go in my experience.

At this stage, I would reboot the PC altogether (as this should terminate the various underlying processes) and retry. I would also make sure that your flash drive is plugged directly to an USB 3.0 port (and not through an extension cable). A bad blocks check may also help in case this is a flash memory issue.

frawag, 2017-06-20:
Thanks for the suggestions!

I tried did reboot multiple times, and started the process again. The same freezing happened every time, although at different percentages - it got as far as 50% or so but eventually did stop.

Bad sector check (1 pass) on the SSD did not reveal any problems.

I noticed, that in the Task Manager the "Active time" for the drive goes to 100% when the speed goes to 0. That's where both numbers then stay for ever. So, probably it is the problem on the windows side as you mentioned.

Now I Ordered a new cable to connect the drive directly to the USB-C port instead of going through a USB-C to USB-A hub - will let you know if that changes anything once it is here.

pbatard, 2017-06-20:
> The same freezing happened every time, although at different percentages

This too would seem to point to a hardware or environment issue.

> Bad sector check (1 pass) on the SSD did not reveal any problems.

Okay. Then it could be an issue with extension cables, as USB 3.0 can be very sensitive to this.

> Now I Ordered a new cable to connect the drive directly to the USB-C port instead of going through a USB-C to USB-A hub

Sounds good. I've also just applied an improvement, which will be in the next version of Rufus, to be more verbose about the files being applied during WIM extraction, as I can understand that staring at the percentage, with no other indication of whether files are actually being extracted, is not ideal...

pbatard, 2017-07-03:
Any update on your new cable? Thanks.

frawag, 2017-07-03:
Hi Pete,
sorry I didn't come back to you yet.
It seems to have been the adapter/hub I used indeed. With the cable that allowed me to connect the drive directly everything worked flawlessly. I even ran it several times to be sure it's stable now - which it is.
Many thanks for your help!

pbatard, 2017-07-03:
Glad to hear you solved it - I will close the issue then.

closed 2017-07-03 by pbatard