pbatard/rufus #971: Rufus wont detect my external SSD via SATA adapter

superxeon, 2017-06-21:
log:

Rufus version: 2.15.1117
Windows version: Windows 10 64-bit (Build 15063)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0409
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
0 devices found
0 devices found

END

List USB hardrives is selected. The volume is volume F and is one partition.

pbatard, 2017-06-21:
Rufus is an __USB__ utility. SATA is not supported.

pbatard, 2017-06-21:
This being said, and now that the issue is closed, I'll point out that you __may__ be able to get Rufus to access your drive by using the __UNDOCUMENTED__ <kbd>Ctrl</kbd>-<kbd>Alt</kbd>-<kbd>F</kbd> cheat mode, provided that your SATA drive is seen as Removable. But this is a completely unsupported mode, and you're 100% on your own if you attempt to use it (and end up formatting the wrong drive by mistake for any reason).

closed 2017-06-21 by pbatard

superxeon, 2017-06-21:
How do I get it seen as removable?

pbatard, 2017-06-21:
BIOS settings (if you're lucky).