pbatard/rufus #973: [Suggestion] Add please a sound notification and some blink on the taskbar when all progress is done

drakulaboy, 2017-06-28:
As the title says, it would be great some features like this, while i'm waiting and Rufus is in background, i would love to be notified when is done. Thank you.

pbatard, 2017-06-28:
I'll think about it, but I'm a bit two minds about this. I know that quite a few people don't like sound notifications, and I don't want to have to add a customization page to a simple utility like Rufus (_Sound notifications on/off_). I need to consider it further before I decide on whether I want to add such a feature...

drakulaboy, 2017-06-28:
imo this option don't need another customization page, add it after Format Options label and it will be fine. Cheers.

pbatard, 2017-06-28:
> add it after Format Options label and it will be fine. 

I beg to differ. This can hardly be considered a major option (at best it is a "cosmetic enhancement"), therefore it should not feature on the main UI. Else, there are plenty of other options, that are currently available as a cheat mode, that I would feature in the UI.

Please try to consider that, while you may see it as a very important feature, others may have quite a different view, and that I am trying to design Rufus with the majority of users in mind, rather than what a minority may find useful.

drakulaboy, 2017-06-29:
oh, okay then, i have Visual Studio 2017 installed, maybe i can do it for myself, will give it a try.

wbsdty331, 2017-07-03:
I think play a notice music is good.

commit 92d046e6 (2017-07-04): [ui] produce a sound and flash taskbar icon on success/error

* Closes #973
closed 2017-07-04 by pbatard