pbatard/rufus #976: Add support for MS-DOS, in Windows 10 upgrades, by using the "Windows.old" folder

neutralin0, 2017-07-01:
Rufus doesn't support MS-DOS, in Windows 10, because the file **diskcopy.dll** has been removed from this operating system.

But, for systems that have been upgraded to Windows 10, the file may still be available in the "**Windows.old**" folder. More specifically, the file may still exist in this location:
**C:\Windows.old\Windows\System32\diskcopy.dll**

In Windows 10, Rufus should try to find **diskcopy.dll** in the aforementioned location.
If the file exists, then Rufus should use it to provide MS-DOS support.

pbatard, 2017-07-01:
Not gonna happen.

Unless you can demonstrate that you have found an issue with FreeDOS compatibility, that the FreeDOS developers are unwilling to address, I'm not going to do anything to help the very limited number of people who still seem to prefer a __closed source & no longer maintained__ version of DOS over the __Open Source & actively maintained__ one. Plus, I doubt this will benefit many people besides yourself (because you need to have performed a 7 or 8 to 10 upgrade __AND__ think you have a very good reason to prefer MS-DOS over FreeDOS, which drastically reduces the target population), and I prefer to spend my time on features that, instead, will benefit a majority of people.

closed 2017-07-01 by pbatard

neutralin0, 2017-07-02:
I've read your previous comments on this subject, so I had a feeling you would respond like that.
I'll try to explain myself in a more detailed way.

> Unless you can demonstrate that you have found an issue with FreeDOS compatibility, that the FreeDOS developers are unwilling to address,

Herein lies the problem.
I want to update the BIOS. If I use FreeDOS and the process fails, because there was some issue with MS-DOS compatibility, I risk bricking the computer.
Then, the proof of incompatibility, with MS-DOS, will be a bricked computer.
Needless to say, this outcome helps no one.

> I'm not going to do anything to help the very limited number of people who still seem to prefer a closed source & no longer maintained version of DOS over the Open Source & actively maintained one.

Please understand that I do not **_want_** to use MS-DOS.
I **_need_** to use MS-DOS, to ensure that the BIOS update process is successful and doesn't brick the computer.

This is what the FreeDOS homepage currently says:
_"Any program that works on MS-DOS **should** also run on FreeDOS."_

The word "should" is crucial. It means there's no guarantee that FreeDOS is 100% compatible with MS-DOS.

Should I just experiment with FreeDOS, when performing BIOS updates?
Should I simply hope the computer doesn't get bricked, because of some obscure FreeDOS incompatibility with MS-DOS?
I'm sorry, but I can't take those risks.

> Plus, I doubt this will benefit many people besides yourself (because you need to have performed a 7 or 8 to 10 upgrade AND think you have a very good reason to prefer MS-DOS over FreeDOS, which [...]

Again, performing BIOS/firmware updates is the single greatest reason to still prefer MS-DOS over FreeDOS.
This preference is all about necessity, not about opinion.

The reasons why I posted the "Windows.old" scenario, are the following:

- It should be easy to implement
- Requires no UI changes whatsoever
- It's 100% legal

Ideally, Rufus should have support for external MS-DOS files, provided by the user.
But you made yourself clear, in another issue (#870), that this scenario is sketchy, from a legal point of view, and therefore you will not support it.

When I stumbled upon the "Windows.old" scenario, I realized this was a fully legal way of having MS-DOS support in Windows 10.
Since you refuse to support user provided MS-DOS files, this would be a completely legal way around that.

pbatard, 2017-07-02:
I don't address "might be" scenarios.

As I stated before, unless you can produce a FreeDOS incompatibility that the FreeDOS developers are unwilling to address, I'm not going to add the feature you want. As someone who has been subscribed to the FreeDOS mailing lists for many years, I can tell you that your "FreeDOS might brick my computer during a BIOS upgrade" is bullshit. I have never seen a single instance of someone reporting such an occurrence, and I'm not going to address your __irrational__ belief that, if you use FreeDOS, you're going to be the first person to.

You'll have to find a better reason to try to justify what remains purely a __personal preference__.
Or you can do what the "I want to be able to provide my own version of `diskcopy.dll`" guy did and develop your own custom version of Rufus, if using MS-DOS over FreeDOS in Windows 10 is that important to you.

Plus, one thing you may want to consider, if you want to bring the legal argument, is that if I were to add what you request, then it wouldn't be long before people started copying versions of the `diskcopy.dll`, that they are not entitled to use with Windows 10, into the specific `Windows.old` DLL (since I'm not gonna add additional checks to find out if the version they have is really from an upgrade) and there again we have a legal liability on our hands.

Have a nice day.

JonnyTech, 2017-07-02:
As a computer engineer and FOSS advocate, I have used FreeDOS to flash many thousands of systems without issue. When booting ensure that you do not load any drivers or memory managers, just the plain core environment. If you do find an incompatibly then the devs and I would be keen to know.

neutralin0, 2017-07-03:
> As I stated before, unless you can produce a FreeDOS incompatibility that the FreeDOS developers are unwilling to address, I'm not going to add the feature you want. As someone who has been [...]

Would you mind telling me how can someone report a bricked computer to the FreeDOS devs?
What exactly does one tell them?

Because when the computer dies, there are no more logs, no errors messages, no nothing. Just dead silence.

FreeDOS might be the culprit. Or perhaps the flashing program. Or perhaps a faulty, buggy BIOS file. And so on...

When BIOS updates fail and computers get bricked, people go straight to the manufacturers, for support, because they immediately think it's a hardware problem (which by now it really is).
They don't go to FreeDOS or AMI or Phoenix. That would be ridiculous. You need physical access to the machine in order to diagnose and fix it.

That's why you don't see reports of bricked computers, in the FreeDOS bug report list, and you never will.
All these reports have gone to the manufacturers.

Can you guess where I'll go if I end up using FreeDOS and it bricks my computer?
That's right, I'll go straight to the manufacturer.

>You'll have to find a better reason to try to justify what remains purely a personal preference.
Or you can do what the "I want to be able to provide my own version of diskcopy.dll" guy did and develop your own custom version of Rufus, if using MS-DOS over FreeDOS in Windows 10 is that important to you.

Again, please understand that this is **_not_** about personal preference.
I couldn't care less about MS-DOS.

But if I want to reduce the probability of a failed BIOS update, to a minimum, then MS-DOS is really the only choice I have. This is not an opinion, it is a matter of fact.

Rufus is a great program. It's very useful and user friendly.
The request I made was to try and make Rufus even more useful.

As I pointed out earlier, this request is easy to implement, requires no UI changes whatsoever and is 100% legal (see below for more on this).
The only obstacle seems to be your inflexibility.

But thanks anyway. Your work is very appreciated and Rufus is a real gem.
I will continue to use Rufus and I'll keep recommending it to others.

I'll simply use another tool for this specific job.

> Plus, one thing you may want to consider, if you want to bring the legal argument, is that if I were to add what you request, then it wouldn't be long before people started copying versions of the [...]

Supporting the "Windows.old" scenario is completely legal on the Rufus side.

Can people work around this fully legal scenario and use it in a not so legal way?
Of course they can, but Rufus can't be held responsible for that.

If you try to blame Rufus for something like that, then it gets silly pretty fast. For example:

Can operating systems be used for illegal activity? Of course they can.
Should Microsoft, Apple and all Linux devs be held responsible for that?

Can people use Rufus with illegally acquired Windows ISO files? Of course they can.
Should the author of Rufus be held responsible for that?

Can kitchen knives be used for criminal activity? Of course they can.
Should the manufacturers be held responsible for all criminal activity, where kitchen knives were used?

Pretty silly, as you can see.

The other scenario, where users must provide the files, is different. I do agree with you, that this one is a bit sketchy. In this case, Rufus would be actively inciting and encouraging people to provide specific files, that are not legally available on their own.

But in the "Windows.old" scenario, there's no inciting of this kind whatsoever.
The files are provided by Microsoft itself. Therefore, it's completely legal.

> Have a nice day.

I think you have just destroyed my sarcasm detector.

Anyway, this discussion has become much longer than I expected and it seems to me that nothing useful will be achieved.

Thank you for listening.
Sorry for the long posts and for taking up your time.

Cheers!

pbatard, 2017-07-03:
> This is not an opinion, it is a matter of fact.

This is where you are wrong. You are clearly of the __opinion__ that using FreeDOS instead of MS-DOS might brick your computer, despite not being able to provide any proof that this can or has ever be the case (Because don't you think that manufacturers would engage with the FreeDOS developers if they got such report from their users? Or how much do you need to distort reality, so that they wouldn't do that, to fit your scenario?).

Let me therefore make one last statement for all the people who would like to see Rufus offer an MS-DOS option in Windows 10: __This ship has sailed. If you want MS-DOS support in Windows 10, then first ask Microsoft to reinstate it in a DLL. Until then you are wasting your time.__

Sybaris1, 2018-02-10:
Hi,
I was trying to update the BIOS on a Zotac machine yesterday and ran into a problem with FreeDOS.

After making a bootable USB stick with Rufus (latest version as of January 2018), and having to select "FreeDOS" because I am running Windows 10 on all my machines, I proceeded to boot the Zotac machine with the new stick.

It booted up fine, but when I ran the batch.bat file to start the BIOS updating process I got a "Not enough memory available" message.

Looking around the web I found someone who had solved this by re-making the USB stick on a Windows 7 machine and selecting MS-DOS instead.

I did not have this option so I figured I would re-make my USB stick but this time unticking "Quick Format". Success, this worked. No more memory message.