pbatard/rufus #977: [Suggestion] Add Shortcut Keys for GUI

wbsdty331, 2017-07-03:
Hi pbatard,
As the title says, I hope rufus could support shortcut Keys, For Example:

```
<Ctrl> + <S>: Start
<Ctrl> + <C>: Close
<Ctrl> + <A>: About...
<Ctrl> + <L>: Log
```

Thank you.

pbatard, 2017-07-03:
I don't think I want to add this. Rufus already supports navigation for vision impaired people, and the amount of people who will need this "because they happened to use Rufus on a PC with a broken mouse" is too limited to make it that helpful. It's a bit like a request to "map Rufus controls to XBox 360 joystick buttons, in case someone prefers to use a joystick": Yeah, it might be helpful __to some__, but it'll be mostly pointless to the vast majority and therefore not worth it. If you have a broken input device and want to use Rufus, please fix your device first...

Also see [here](https://github.com/pbatard/rufus/wiki/FAQ#Why_cant_you_just_add_the_feature_I_want_I_mean_youre_a_developer__it_shouldnt_be_that_hard).

closed 2017-07-03 by pbatard

mk2soldier, 2018-03-28:
Firstly I would like to thank you pbatard for developing Rufus, it's an amazing "little" piece of software, I enjoy using it!

Having said this I second wbsdty331 suggestion but maybe it would be less hassle to enable an "Alt + <letter>" style shortcuts to reach the most used buttons, for example: Alt+U to select the USB device, Alt+I to select an image and so on..

I think that it isn't much about the presence or absence of a mouse but more of a speed improvement for those people who use Rufus on a regular basis (muscle memory)

I've read your FAQ and I think I understand what you wrote there, developing software is very much time consuming..
I'll try to figure out a way to accomplish this and maybe submit a patch for review as you said in the FAQ :)

Thank you again for this software!

pbatard, 2018-03-28:
Okay, I'll keep that in mind that other people might find it useful.

But please understand that this improvement, if I ever got around to do it, will always be very low priority for me, as having to handle 2 modes of generating UI actions is always a massive headache (for instance, if you have that then you can't rely on disabled buttons to prevent conflicting user input), and will require A LOT of work just to satisfy a very limited number of people. As such, it's exceedingly unlikely that, if I decide to implement this, you will see it happen before 2 or 3 years at best (since that's the worth of higher priority features in my current backlog).