pbatard/rufus #979: ER: Please inrease the size of EFI partition to 200MB at least

kirbyzhou, 2017-07-12:
Issue description
-----------------

rufus create GPT disk with 100MB hidden EFI partition now.
It is acceptable for Windows, but macOS requires at least 200MB EFI partition, otherwise its 'Disk Utility.app" refused to operate the disk. such as format to HFS.

So please inrease the size of EFI partition to 200MB at least.

Thanks very much.

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [ ] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

kirbyzhou, 2017-07-12:
I think the code is here:

rufus/src/drive.c
--- drive.c~	2017-07-12 08:46:33.000000000 +0800
+++ drive.c	2017-07-12 08:46:48.000000000 +0800
@@ -1141,9 +1141,9 @@
 		if (extra_partitions & XP_EFI) {
 			// The size of the EFI partition depends on the minimum size we're able to format in FAT32,
 			// which in turn depends on the cluster size used, which in turn depends on the disk sector size.
 			if (SelectedDrive.SectorSize <= 1024)
-				ms_efi_size = 100*MB;
+				ms_efi_size = 200*MB;
 			else if (SelectedDrive.SectorSize <= 4096)
 				ms_efi_size = 300*MB;
 			else
 				ms_efi_size = 1200*MB;	// That'll teach you to have a nonstandard disk!

pbatard, 2017-07-12:
Because you chose not to provide a log, I fail to see the kind of scenario where this would actually matter (What exactly are you trying to achieve here? What ISO are you using? How large is your drive?).

Granted, you may have found an annoyance that you believe warrants an alteration, but you have to realize that, if I were to arbitrarily increase the size of the ESP, I can envision that quite a few more people, who, for instance, are creating a Windows installer against a 4GB drive, and therefore are running close to hitting a size limit, may legitimately be a lot more annoyed than you are by the sudden doubling of the ESP if it means that they are now going to run out of space for no other reason than keep a very limited number of people happy.

I'm afraid that making life easier for Mac users, who also happen to use Rufus (which is already a very reduced set of population) falls second to ensuring that space is not being wasted for the vast majority of users, who will either never plug a flash drive created by Rufus that has an ESP (also a limited scenario), or reformat their drive (and remove the ESP) before doing so, since Rufus is mostly targeted at one-of installations.

Again, because you did not provide a log, which would have helped me deduce the kind of scenario where this might actually matter, I don't think I want to alter the current code.

pbatard, 2017-07-12:
As an addendum, as you can see from the comment, you'll see that the size of the ESP was __deliberately chosen__ to be as small as physically possible when FAT32 limitations are taken into consideration:
```
// The size of the EFI partition depends on the minimum size we're able to format in FAT32
```
This should indicate that unless you can provide a scenario that __many people__ are likely to run into, I will be exceedingly reluctant to increase it.

kirbyzhou, 2017-07-12:
Thanks for your comments.
I can provide a scenario, but I do not know whether it is for many people.
My scenario is to install 'Windows to Go' and a apple HFS partition into the same external usb gpt drive to get a portable working environment.

HFS can only be formatted under macOS, so my steps are:
1. run rufus to create a 'Windows to Go' gtp drive.
2. resize the NTFS main partition under Windows.
3. create a dummy partition to hold the space under Windows.
4. reboot to macOS
5. Try to format "dummy partition" to HFS wit "Disk Utility.app"

Step 5 failed due to the size of EFI partition smaller than 200MB.

pbatard, 2017-07-12:
Okay.

I don't really think Rufus is the right tool to do what you want to achieve, because first of all, if you resize the partition that Rufus creates, then it's not something that's supported, and I also have [no plans to add generic multiple partition support](https://github.com/pbatard/rufus/wiki/FAQ#Do_you_plan_to_add_multiple_partition_support). So, it is unlikely that many people would be using a Windows To Go drive created by Rufus in any other manner than with the default partitions, and I'm afraid your scenario is a bit too unique to justify a patch.

Furthermore, I think it would be a lot more beneficial if you contacted Apple to ask them to fix what appears to be a bug from their Disk Utility application, than ask every Windows application that may format a drive in a manner that Disk Utility will be unhappy about, to add a work around.

As per the [Microsoft Documentation](https://technet.microsoft.com/en-us/library/hh824839.aspx#DiskPartitionRules) I pointed out, it is expected that drives created on Windows may have an ESP that is as small as 100MB, as this is what Microsoft indicates should be the smallest size. Surely, it is in the interest of Apple to have the MacOS Disk Utility compatible with drives created according to Microsoft's recommendations, and you will be helping a lot more users (who may be creating ESP's on Windows using a different utility than Rufus, and who, in order to save space, may be using the smallest size they can) if you manage to get Apple to fix their Disk Utility than if I were to simply raise the minimum size to 200 MB.

Have you tried contacting Apple to see what they have to say about this issue with their Disk Utility?

kirbyzhou, 2017-07-12:
I talked with apple. the response is below:
The EFI/ESP partition is standard on a GPT disk. This size of 200MB is standard for a MAC, and should not be changed.

pbatard, 2017-07-12:
If that's how they answered, they clearly didn't understand your request.

Please explain to them again that you are __not__ trying to change an EFI System Partition's size, but simply trying to access a drive, that was partitioned in Windows according to Microsoft's guidelines, and that contains an ESP, that was created on a Windows system, and that has a size of 100 MB (which is compliant to Microsoft's recommendations).

If Apple genuinely cares about its users, then its software, such as Disk Utility, should be able to work with drives that were partitioned in Windows, and therefore __may__ contain an ESP that is different from Apple's standard guidelines. As opposed to what the support representative appears to have assumed, this is __not__ a request to be able to change the default 200MB of the ESP on Mac, but instead a request to fix interoperability with __UEFI-compliant__ Windows-partitioned drives.

pbatard, 2017-07-20:
Since the ball is really in Apple's camp on this one (either that or you should use a different partitioning tool on MacOS), as any partition tool that refuses to work with a specs compliant 100 MB ESP should be fixed by its manufacturer, I will close this issue.

closed 2017-07-20 by pbatard

amahlaka, 2018-06-18:
i just noticed that i cannot update windows 10 as "efi partition is too small"
i created windows 10 to go to 1tb hard drive and it appears that efi partiton was created to the end of the disk space, so it cannot be extended
100mb efi is too small for latest windows 10 feature update to install

pbatard, 2018-06-18:
Do you have the error message from Windows? I don't mind extending the size of the EFI partition if this is a real issue, but I'd like to see some good corroboration that this is indeed the case, with an actual error message.

My issue is that Rufus is 100% following Microsoft's recommendations as per https://docs.microsoft.com/en-us/windows-hardware/manufacture/desktop/configure-uefigpt-based-hard-drive-partitions which clearly states that a 100 MB EFI partition should be fine (since this is the minimum size).

If that is not the case, then I would prefer if you reported this issue to Microsoft, so that they can update their official documentation, which in turn we will follow, because, as far as we are concerned, what Microsoft states (in an official document that was last updated about 1 year ago) has a lot more weight than a one-of user report.

Moreover, since it Microsoft's business to document whether a 100 MB ESP is large enough or not for a feature update, if it turns out that this is not the case, then Microsoft should be your first line of contact here, so that, if confirmed, your report will also help people who haven't been using Rufus to create their Windows installation, and just followed the official documentation.

It may sound like I'm deflecting the issue here, but my problem is that, if the __official__ documentation about the minimum size is incorrect, then I literally have no idea what actual minimum size I should use in Rufus. The last thing I want to go through is something like _"Let's try 200MB..."_, _"Oh no wait, that's still too small, so try 256MB"_, _"Agh! Too small again. 400MB, maybe?!?"_, with absolutely no __official__ idea as to what the proper minimum size should be (and of course, the other thing I don't want is allocate way too much space for the ESP and get Rufus users complain that it's taking too much space).

In other words, according to their documentation, 100MB is what Microsoft states a Windows-dedicated ESP can be set to, with the expectation that it is not going to create any problems. If that is not the case, then Microsoft need to be alerted that their documentation is not correct, so that applications like Rufus, that depend on said documentation, can follow suit and update their ESP size requirements.

amahlaka, 2018-06-19:
When perfoming update, it gives error 0xc1900200
System doesn't meet minimum requirements, on system that has 8 gigs of ram,
600 gigs of freee space, intel i7,.gtx1080, and a another harddrive with
all latest updates
Multiple forum posts state that a efi partition beijg less than 250-400 can
cause this

pbatard, 2018-06-19:
Fine, I'll set the minimum ESP size to 300MB (but no higher!) in Rufus 3.1 which I'm planning to release today.

But if anybody ever complains about this, I'll point them here...

pbatard, 2018-06-19:
Also, someone from these forums should __REALLY__ report this issue to Microsoft and ask them to validate the issue and update their official documentation if that is the problem.